evcc-io/evcc #11330: gelegentlich fehlerhafte Initialisierung von tariff im UI

VolkerK62, 2023-12-29:
Nach einem Neustart von evcc zeigt sich seit kurzem danach folgendes Bild.
Der Preis sollte eigentlich CO2 anzeigen und anklickbar sein.
Im Log ist kein Fehler, keine Warnung.
`evcc tariff` zeigt keinen Fehler.

[image]

Nach einem erneuten Neustart ist meist wieder alles so, wie es sein sollte.
Funktionale Einschränkung ist, dass das Fenster "smartes Netzladen" nicht erreichbar ist.
Es ist nicht immer reproduzierbar.

 api/state:
`{"result":{"auth":{"vehicles":{}},"aux":[{"power":0}],"auxPower":0,"battery":[{"power":-45.493626661485344,"soc":40,"capacity":7.5,"controllable":false}],"batteryCapacity":7.5,"batteryConfigured":true,"batteryDischargeControl":false,"batteryEnergy":0,"batteryMode":"unknown","batteryPower":-45.493626661485344,"batterySoc":40,"bufferSoc":100,"bufferStartSoc":0,"currency":"EUR","greenShareHome":0,"greenShareLoadpoints":0,"gridConfigured":true,"gridCurrents":[0.93,1.27,1.15],"gridEnergy":3856.2469999999994,"gridPower":593.4091721232999,"gridPowers":[171.84,217.71,207.22],"homePower":547.9155454618145,"loadpoints":[{"chargeCurrent":0,"chargeCurrents":[0,0,0],"chargeDuration":0,"chargePower":0,"chargeRemainingEnergy":24200,"chargeTotalImport":1343.7,"chargeVoltages":[232.6,0,0],"chargedEnergy":0,"chargerFeatureHeating":false,"chargerFeatureIntegratedDevice":false,"chargerIcon":null,"charging":false,"connected":true,"connectedDuration":0,"disableThreshold":800,"effectiveLimitSoc":100,"effectiveMaxCurrent":20,"effectiveMinCurrent":6,"effectivePlanSoc":0,"effectivePlanTime":"0001-01-01T00:00:00Z","effectivePriority":0,"enableThreshold":-500,"enabled":false,"guardAction":"inactive","guardRemaining":0,"limitEnergy":0,"limitSoc":100,"maxCurrent":20,"minCurrent":6,"mode":"pv","phaseAction":"inactive","phaseRemaining":0,"phasesActive":1,"phasesConfigured":null,"phasesEnabled":1,"planEnergy":0,"planProjectedStart":"0001-01-01T00:00:00Z","planTime":"0001-01-01T00:00:00Z","priority":0,"pvAction":"inactive","pvRemaining":0,"sessionCo2PerKWh":null,"sessionEnergy":0,"sessionPrice":null,"sessionPricePerKWh":null,"sessionSolarPercentage":0,"title":"Carport","vehicleCapacity":72.6,"vehicleClimaterActive":null,"vehicleDetectionActive":false,"vehicleFeatureOffline":false,"vehicleIcon":"car","vehicleIdentity":"\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","vehicleName":"auto","vehicleOdometer":0,"vehiclePresent":true,"vehicleRange":285,"vehicleSoc":70,"vehicleTargetSoc":100,"vehicleTitle":"(T)Raumschiff"},{"chargeCurrent":0,"chargeDuration":0,"chargePower":0,"chargeRemainingEnergy":90.27777777777779,"chargeTotalImport":1.2533833333333335,"chargedEnergy":0,"chargerFeatureHeating":false,"chargerFeatureIntegratedDevice":true,"chargerIcon":"bike","charging":false,"connected":false,"connectedDuration":9223372036854775807,"disableThreshold":0,"effectiveLimitSoc":100,"effectiveMaxCurrent":0.75,"effectiveMinCurrent":0.75,"effectivePlanSoc":0,"effectivePlanTime":"0001-01-01T00:00:00Z","effectivePriority":5,"enableThreshold":0,"enabled":false,"guardAction":"inactive","guardRemaining":0,"limitEnergy":0,"limitSoc":0,"maxCurrent":0.75,"minCurrent":0.75,"mode":"off","phaseAction":"inactive","phaseRemaining":0,"phasesActive":1,"phasesConfigured":null,"phasesEnabled":1,"planEnergy":0,"planProjectedStart":"0001-01-01T00:00:00Z","planTime":"0001-01-01T00:00:00Z","priority":5,"pvAction":"inactive","pvRemaining":0,"sessionCo2PerKWh":null,"sessionEnergy":0,"sessionPrice":null,"sessionPricePerKWh":null,"sessionSolarPercentage":0,"title":"Garage","vehicleCapacity":0.625,"vehicleClimaterActive":null,"vehicleDetectionActive":false,"vehicleFeatureOffline":false,"vehicleIcon":"bike","vehicleName":"bike","vehicleOdometer":0,"vehiclePresent":true,"vehicleRange":0,"vehicleSoc":87,"vehicleTargetSoc":0,"vehicleTitle":"E-Bikes"},{"chargeCurrent":0,"chargeDuration":0,"chargePower":0,"chargeRemainingEnergy":833.3333333333335,"chargeTotalImport":0,"chargedEnergy":0,"chargerFeatureHeating":true,"chargerFeatureIntegratedDevice":true,"chargerIcon":"waterheater","charging":false,"connected":false,"connectedDuration":9223372036854775807,"disableThreshold":0,"effectiveLimitSoc":60,"effectiveMaxCurrent":3,"effectiveMinCurrent":2,"effectivePlanSoc":0,"effectivePlanTime":"0001-01-01T00:00:00Z","effectivePriority":10,"enableThreshold":0,"enabled":false,"guardAction":"inactive","guardRemaining":0,"limitEnergy":0,"limitSoc":0,"maxCurrent":3,"minCurrent":2,"mode":"off","phaseAction":"inactive","phaseRemaining":0,"phasesActive":1,"phasesConfigured":null,"phasesEnabled":1,"planEnergy":0,"planProjectedStart":"0001-01-01T00:00:00Z","planTime":"0001-01-01T00:00:00Z","priority":10,"pvAction":"inactive","pvRemaining":0,"sessionCo2PerKWh":null,"sessionEnergy":0,"sessionPrice":null,"sessionPricePerKWh":null,"sessionSolarPercentage":0,"title":"WP SG+","vehicleCapacity":15,"vehicleClimaterActive":null,"vehicleDetectionActive":false,"vehicleFeatureOffline":false,"vehicleIcon":"waterheater","vehicleName":"WasserSG","vehicleOdometer":0,"vehiclePresent":true,"vehicleRange":0,"vehicleSoc":55,"vehicleTargetSoc":0,"vehicleTitle":"WasserSG"},{"chargeCurrent":16,"chargeDuration":0,"chargePower":0,"chargeRemainingEnergy":6666.666666666668,"chargeTotalImport":938.9631999999999,"chargedEnergy":0,"chargerFeatureHeating":true,"chargerFeatureIntegratedDevice":true,"chargerIcon":"generic","charging":false,"connected":true,"connectedDuration":0,"disableThreshold":0,"effectiveLimitSoc":52,"effectiveMaxCurrent":16,"effectiveMinCurrent":6,"effectivePlanSoc":0,"effectivePlanTime":"0001-01-01T00:00:00Z","effectivePriority":0,"enableThreshold":0,"enabled":true,"guardAction":"inactive","guardRemaining":0,"limitEnergy":0,"limitSoc":52,"maxCurrent":16,"minCurrent":6,"mode":"now","phaseAction":"inactive","phaseRemaining":0,"phasesActive":1,"phasesConfigured":null,"phasesEnabled":1,"planEnergy":0,"planProjectedStart":"0001-01-01T00:00:00Z","planTime":"0001-01-01T00:00:00Z","priority":0,"pvAction":"inactive","pvRemaining":0,"sessionCo2PerKWh":null,"sessionEnergy":0,"sessionPrice":null,"sessionPricePerKWh":null,"sessionSolarPercentage":0,"title":"WP -Anzeige-","vehicleCapacity":0,"vehicleClimaterActive":null,"vehicleDetectionActive":false,"vehicleFeatureOffline":false,"vehicleIcon":"","vehicleName":"","vehicleOdometer":0,"vehiclePresent":true,"vehicleRange":0,"vehicleSoc":12,"vehicleTargetSoc":0,"vehicleTitle":"WasserWP"},{"chargeCurrent":16,"chargeDuration":0,"chargePower":0,"chargeTotalImport":26.363466666666667,"chargedEnergy":0,"chargerFeatureHeating":false,"chargerFeatureIntegratedDevice":true,"chargerIcon":"generic","charging":false,"connected":true,"connectedDuration":0,"disableThreshold":0,"effectiveLimitSoc":100,"effectiveMaxCurrent":16,"effectiveMinCurrent":6,"effectivePlanSoc":0,"effectivePlanTime":"0001-01-01T00:00:00Z","effectivePriority":0,"enableThreshold":0,"enabled":true,"guardAction":"inactive","guardRemaining":0,"limitEnergy":0,"limitSoc":0,"maxCurrent":16,"minCurrent":6,"mode":"now","phaseAction":"inactive","phaseRemaining":0,"phasesActive":1,"phasesConfigured":null,"phasesEnabled":1,"planEnergy":0,"planProjectedStart":"0001-01-01T00:00:00Z","planTime":"0001-01-01T00:00:00Z","priority":0,"pvAction":"inactive","pvRemaining":0,"sessionCo2PerKWh":null,"sessionEnergy":0,"sessionPrice":null,"sessionPricePerKWh":null,"sessionSolarPercentage":0,"title":"Waschen -Anzeige-","vehicleCapacity":0,"vehicleClimaterActive":null,"vehicleDetectionActive":false,"vehicleFeatureOffline":true,"vehicleIcon":"generic","vehicleName":"Waschen","vehicleOdometer":0,"vehiclePresent":true,"vehicleRange":0,"vehicleSoc":0,"vehicleTargetSoc":0,"vehicleTitle":"Waschmaschine"},{"chargeCurrent":6,"chargePower":0,"chargerFeatureHeating":false,"chargerFeatureIntegratedDevice":true,"chargerIcon":"generic","disableThreshold":0,"effectiveLimitSoc":100,"effectiveMaxCurrent":16,"effectiveMinCurrent":6,"effectivePlanSoc":0,"effectivePlanTime":"0001-01-01T00:00:00Z","effectivePriority":0,"enableThreshold":0,"guardAction":"inactive","guardRemaining":0,"limitEnergy":0,"limitSoc":0,"maxCurrent":16,"minCurrent":6,"mode":"now","phaseAction":"inactive","phaseRemaining":0,"phasesActive":1,"phasesConfigured":null,"phasesEnabled":1,"planEnergy":0,"planTime":"0001-01-01T00:00:00Z","priority":0,"pvAction":"inactive","pvRemaining":0,"title":"Trocknen -Anzeige-","vehicleCapacity":0,"vehicleClimaterActive":null,"vehicleDetectionActive":false,"vehicleFeatureOffline":true,"vehicleIcon":"generic","vehicleName":"Tr [... cut ...] 5451879742047}},"tariffCo2":57,"tariffCo2Home":57,"tariffCo2Loadpoints":57,"tariffFeedIn":0.1095,"tariffGrid":0.354,"tariffPriceHome":0.354,"tariffPriceLoadpoints":0.354,"vehicles":{"Trocknen":{"title":"Trockner"},"Waschen":{"title":"Waschmaschine"},"WasserSG":{"title":"WasserSG","limitSoc":60},"WasserWP":{"title":"WasserWP"},"auto":{"title":"(T)Raumschiff","limitSoc":100},"bike":{"title":"E-Bikes","minSoc":87,"limitSoc":100},"offline":{"title":"ohne Bluelink"}},"version":"0.123.2 (170e2cad)"}}`

VolkerK62, 2023-12-29:
Ich bin damit wohl nicht alleine
https://github.com/evcc-io/evcc/discussions/10978#discussioncomment-7974494

mdkeil, 2023-12-29:
Kann ich bestätigen.. hatte ich heute auch einmal.. bin dann auf die nightly gewechselt und smartes Laden war wieder verfügbar. (Also die Auswahl des kWh-Preises) Denke ein Neustart hätte auch gereicht.

Balujuhu, 2023-12-29:
Hatte ich heute morgen auch, Neustart hat es dann behoben.

0x3d13f, 2023-12-29:
Hier sieht es etwas anders aus. Ich habe die Anzeige mit Preis bzw. CO2-Wert beim Netzbezug gar nicht (mehr), nicht die fehlende Verlinkung. Was sind die Voraussetzungen dafür, dass da etwas angezeigt wird?

0x3d13f, 2023-12-29:
Fehler gefunden: es stand grünstromindex mit `planner:` statt `co2:` drin.

naltatis, 2023-12-30:
Der Dialog hängt vom `smartCostType` ab. Die publishen den an zwei Stellen:

https://github.com/evcc-io/evcc/blob/master/core/site.go#L830

https://github.com/evcc-io/evcc/blob/master/core/site.go#L833

Der Wert muss `co2`, `dynamicprice` oder `priceforecast` sein, damit wir den Dialog anbieten ([siehe hier](https://github.com/evcc-io/evcc/blob/master/assets/js/components/Energyflow/Energyflow.vue#L221)). Hier scheint dann beim Publishen etwas schief zu laufen.

naltatis, 2023-12-30:
@VolkerK62 bei dir sieht man auch, dass `smartCostLimit: null` ist. Verstehe noch nicht ganz wann/wie das passieren kann.

\\cc @andig

VolkerK62, 2023-12-30:
@naltatis 
>dass `smartCostLimit: null` ist.

du meinst bestimmt `smartCostType: null` (nicht Limit)

andig, 2023-12-30:
Wir raten schon wieder. Gibts eine Config dazu? Sonst würde ich zu machen…

VolkerK62, 2023-12-30:
Config, bitte schön ;-)

```
Konfiguration (/etc/evcc.yaml):

network:
  schema: http
  host: evcc.local
  port: 7070

interval: 5s

sponsortoken: *****

telemetry: true

log: debug
levels:
  #site: trace
  #lp-1: info
  #lp-2: info
  #lp-3: info
  #lp-4: info
  #daheimladen-mb: trace
  #hyundai: trace
  #main: trace
  #cache: trace
  #db: trace
  #http: trace
  #influx: trace
  #mqtt: trace
  #proxy-5200: trace
  #gsi: trace
  

########################################################################
meters:
- name: grid
  type: movingaverage
  decay: 0.33
  meter:
    type: template
    template: shelly-3em
    usage: grid
    host: 192.168.178.28

- name: pv
  type: movingaverage
  decay: 0.33
  meter:
    type: template
    template: solarlog
    usage: pv
    host: 192.168.178.23
    port: 502

- name: battery
  type: movingaverage
  decay: 0.33
  meter:
    type: custom
    power:
      ##Shelly EM Stromzange 2
      source: http
      uri: http://192.168.178.21/emeter/1
      jq: if .pf==0 then .power else .power / (.pf | fabs) end
    soc:
      source: mqtt
      topic: Haus/Bat_SoC
      timeout: 60m
[... 270 lines cut ...]
    stop:
      title: Stop {{.title}}
      msg: |
        {{ if eq .title "Carport" }} Stop um {{printf `(%02d:%02d)` now.Hour now.Minute}} {{- end }}
        {{ if eq .title "Garage" }} Stop um {{printf `(%02d:%02d)` now.Hour now.Minute}} {{- end }} 
        {{ if eq .title "WP -Anzeige-" }} Stop um {{printf `(%02d:%02d)` now.Hour now.Minute}} {{- end }} 
        

######################################################################
influx:
  url: http://localhost:8086
  database: evcc

mqtt:
  broker: localhost:1883
  topic: evcc

modbusproxy:
- port: 5200
  uri: 192.168.178.26:502

#####################################################################

Version: `0.123.2 (170e2cad)`
```

VolkerK62, 2023-12-31:
Hab mir heute vormittag das Nightly geholt.
Ich sags ja nur ungern, aber das oben beschriebene Verhalten besteht weiterhin.
Es ist nicht sooooo tragisch, weil es nach dem zweiten Neustart bisher immer wieder ok war.

naltatis, 2023-12-31:
> Ich sags ja nur ungern, aber das oben beschriebene Verhalten besteht weiterhin.

Schade :}

andig, 2024-01-01:
...das ist ja super nervig. Irgendeine race condition, aber was raced da?

@VolkerK62 kann ich irgendwie auf Deine Installation drauf und da ein bisschen was am Code für mehr Logging verändern? SSH?

bilbolodz, 2024-01-01:
I've similar problem @andig I can arrange ssh access to check it.

andig, 2024-01-01:
@VolkerK62 @bilbolodz please take a look at your logfile in case of error: is there any log line like

    failed configuring co2 tariff

VolkerK62, 2024-01-01:
Hab es gerade nochmal ausprobiert.
Neustart, danach war kein co2 verfügbar
Ein paar Minuten später nochmal Neustart und co2 war wieder da
Im Log kann ich nichts auffälliges entdecken.
[start.log](https://github.com/evcc-io/evcc/files/13805266/start.log)

VolkerK62, 2024-01-01:
@andig   Ich hatte doch vor kurzem dieses "outdated"  Problem bei tariff  (da haben wir am 23.12. auf Slack geschrieben).
Dann hast du irgendetwas eingebaut.
Gefühlt tritt es seitdem auf.

andig, 2024-01-01:
Der Fehler scheint am co2 Tarif zu hängen, also GSI. Ich würde raten, dass die Initialisierung nicht klappt, evcc dann aber wunschgemäß weiter macht anstatt auf die Nase zu fallen. Das würde ich dann aber im Log als ERROR erwarten. Ich finde leider keinen Weg, das zu reproduzieren :/

bilbolodz, 2024-01-01:
@andig no errors in log file

andig, 2024-01-01:
@VolkerK62 setz doch mal bitte gsi auf trace, dann nochmal ein Log vom Fehler.

VolkerK62, 2024-01-01:
Werd ich machen, aber `tariff` bringt keinen Fehler und wie im Eingangspost zu sehen, co2 ist da (Tooltip)

andig, 2024-01-01:
Ja- aber anscheinend nicht zum Zeitpunkt wo der smart cost type ermittelt wird. Warum auch immer- kann egtl. nicht sein (tm).

VolkerK62, 2024-01-01:
es ist wie verhext. 
Seit ich `gsi: trace` eingeschaltet habe, kommt der Fehler nicht mehr. Ich lass das mal drin, dann hab ich es fürs nächste mal.

andig, 2024-01-01:
@bilbolodz can you confirm

    smartCostType: null

in case of error? Does 

    gsi: trace

give any useful log on error?

bilbolodz, 2024-01-01:
I don't have nay "smartCostType" entries in my config. Should I add it? If so in which section? I looked in documentation but I have found nothing for such keyword.

andig, 2024-01-01:
In case of error, check `/api/state?jq:.smartCostType`. Also, please share your tariff config.

andig, 2024-01-01:
> Keine Ahnung, ob es damit etwas zu tun hat

Eigentlich nicht, denn wir setzen das nur einmal zum Start. Hier ist auch nix unvollständig, nur das Log wird nach 4k abgeschnitten.

mdkeil, 2024-01-01:
Bei mir ist es ja auch mal aufgetreten, habe aber keinen CO2-Tarif konfiguriert.

andig, 2024-01-01:
> Bei mir ist es ja auch mal aufgetreten

Was konkret? Was sagt /api/state in dem Moment?

mdkeil, 2024-01-01:
Konkret, dass kein SmartCostLimit via UI einstellbar ist, da bei Netzbezug kein weiteres Untermenü verfügbar ist.. hatte ich aber schon vor 3 Tagen geschrieben.. seitdem ist es aber auch nicht mehr aufgetreten. Unter Tarifs habe ich nur Tibber konfiguriert.

naltatis, 2024-01-01:
Ich hatte den Effekt (`smartCostType=null`) heute leider auch mehrmals. Sowohl beim lokalen Testen (fixed Tarif mit Zones), aber auch mit Tibber als Grid Tarif. Irgendwas ist da leider krumm (geworden?). Nach nem Neustart ists idR. ok.

andig, 2024-01-01:
@naltatis wir haben ja zusammen rein geschaut. Kann egtl. nur passieren, wenn einer der Tarife nil ist- das scheint aber nicht der Fall zu sein. Ich hab noch einen PR mit ein bisschen Logging, den könnten wir ins nightly nehmen.

andig, 2024-01-01:
> Konkret, dass kein SmartCostLimit via UI einstellbar ist, da bei Netzbezug kein weiteres Untermenü verfügbar ist.. hatte ich aber schon vor 3 Tagen geschrieben.. seitdem ist es aber auch nicht mehr aufgetreten. Unter Tarifs habe ich nur Tibber konfiguriert.

@mdkeil was sagt /api/state in dem Moment?

mdkeil, 2024-01-01:
Ich liefere den state, wenn es nochmal auftritt.

andig, 2024-01-01:
> Ich hatte den Effekt (`smartCostType=null`) heute leider auch mehrmals. Sowohl beim lokalen Testen (fixed Grid Tarif mit Zones und demo.yaml mit co2 grünstromindex), aber auch mit Tibber als Grid Tarif. 

Ich habs bei ca 50 Versuchen 1x gesehen, das reicht leider nicht fürs Debugging. Wenns irgendwas gibt das häufiger zu provozieren bin ich für Tipps dankbar.

mdkeil, 2024-01-02:
`wenn einer der Tarife nil ist- das scheint aber nicht der Fall zu sein.
`
Aus dem Bauch heraus würde ich sagen, dass es bei mir in dem Zeitraum aufgetreten ist, wo Tibber API/Server Probleme hatte und zumindest teilweise die Abfragen nicht sauber bei evcc-start gelaufen sind. Aber ohne /api/state bleibt es nur eine Vermutung.

naltatis, 2024-01-02:
Hier ein kleines Shellscript zum Prüfen. Wie gesagt, bei mir passiert das auch im Fixed Tariff `./tests/plan.evcc.yaml`. Also ohne Netzwerkabfrage und co.

Bei 10 Iterationen ist bei mir "meist" einer dabei der `null` liefert.

```sh
#!/bin/bash

for i in {1..10}; do
  echo "--- Iteration $i start ---"
  ./evcc --config tests/plan.evcc.yaml &
  pid=$!
  sleep 2
  echo ">>>>>"
  curl http://localhost:7070/api/state?jq=.smartCostType
  echo "<<<<<"
  kill $pid
  echo "--- Iteration $i complete ---"
done

echo "All iterations completed."
```

```
--- Iteration 4 start ---
[main  ] INFO 2024/01/02 08:25:34 evcc 0.123.3 (7ac032818)
[main  ] INFO 2024/01/02 08:25:34 using config file: tests/plan.evcc.yaml
[main  ] INFO 2024/01/02 08:25:34 starting ui and api at :7070
[db    ] INFO 2024/01/02 08:25:34 using sqlite database: /Users/michael/.evcc/evcc.db
[lp-1  ] WARN 2024/01/02 08:25:34 phases not configured, assuming 3p
[lp-2  ] WARN 2024/01/02 08:25:34 phases not configured, assuming 3p
[site  ] WARN 2024/01/02 08:25:34 vehicle 'Vehicle no SoC no Capacity' does not support automatic detection
[site  ] WARN 2024/01/02 08:25:34 vehicle 'Vehicle no SoC with Capacity' does not support automatic detection
[site  ] WARN 2024/01/02 08:25:34 vehicle 'Vehicle with SoC no Capacity' does not support automatic detection
[site  ] WARN 2024/01/02 08:25:34 vehicle 'Vehicle with SoC with Capacity' does not support automatic detection
[site  ] WARN 2024/01/02 08:25:34 vehicle 'Vehicle with SoC with Massive Capacity' does not support automatic detection
[site  ] WARN 2024/01/02 08:25:34 vehicle 'Mein Testfahrzeug' does not support automatic detection
[site  ] INFO 2024/01/02 08:25:34 site config:
[site  ] INFO 2024/01/02 08:25:34   meters:      grid ✓ pv ✗ battery ✗
[site  ] INFO 2024/01/02 08:25:34     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/01/02 08:25:34   vehicles:
[site  ] INFO 2024/01/02 08:25:34     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2024/01/02 08:25:34     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2024/01/02 08:25:34     vehicle 3: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2024/01/02 08:25:34     vehicle 4: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2024/01/02 08:25:34     vehicle 5: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2024/01/02 08:25:34     vehicle 6: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/01/02 08:25:34 loadpoint 1:
[lp-1  ] INFO 2024/01/02 08:25:34   mode:        pv
[lp-1  ] INFO 2024/01/02 08:25:34   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/01/02 08:25:34   meters:      charge ✗
[lp-2  ] INFO 2024/01/02 08:25:34 loadpoint 2:
[lp-2  ] INFO 2024/01/02 08:25:34   mode:        pv
[lp-2  ] INFO 2024/01/02 08:25:34   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-2  ] INFO 2024/01/02 08:25:34   meters:      charge ✗
[site  ] ERROR 2024/01/02 08:25:34 battery not configured
[site  ] WARN 2024/01/02 08:25:34 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[lp-1  ] INFO 2024/01/02 08:25:34 car connected
[lp-2  ] INFO 2024/01/02 08:25:34 car connected
>>>>>
null
<<<<<
--- Iteration 4 complete ---
--- Iteration 5 start ---
[main  ] INFO 2024/01/02 08:25:36 evcc 0.123.3 (7ac032818)
[main  ] INFO 2024/01/02 08:25:36 using config file: tests/plan.evcc.yaml
[main  ] INFO 2024/01/02 08:25:36 starting ui and api at :7070
[db    ] INFO 2024/01/02 08:25:36 using sqlite database: /Users/michael/.evcc/evcc.db
```

StefanSchoof, 2024-01-02:
Kann man das mit `git bisect run` nutzen, um raus zu finden mit welchen commit das Problem entstanden ist?

StefanSchoof, 2024-01-02:
Ich habe das mal gemacht und komme auf https://github.com/evcc-io/evcc/commit/55df2b502cfcab800f1730ae89cec756b75757af

Habe dafür
```
git bisect start master 6bcfb04f057420e0d8864acf2acf04f136c0d95d
git bisect run ../evcc_tariff.sh
```

mit 

```bash
#!/bin/bash
make install-ui
make

for i in {1..50}; do
  echo "--- Iteration $i start ---"
  ./evcc --config ../plan.evcc.yaml &
  pid=$!
  sleep 2
  echo ">>>>>"
  res=$(curl http://localhost:7070/api/state?jq=.smartCostType)
  echo $res
  if [[ "$res" == "null" ]]; then
    kill $pid
    exit 1
  fi
  echo "<<<<<"
  kill $pid
  echo "--- Iteration $i complete ---"
done

echo "All iterations completed."
```

`plan.evcc.yaml` ist aus dem master. Da ein bisschen Zufall dabei ist, habe ich auf 50 erhöht und das zweimal ausgeführt.

All-Ex-U, 2024-01-02:
Das Problem tritt bei mir gerade auch auf:
[image]

Hier /api/state:

{
   "result":{
      "auth":{
         "vehicles":{
            
         }
      },
      "batteryConfigured":false,
      "batteryDischargeControl":false,
      "batteryMode":"unknown",
      "bufferSoc":0,
      "bufferStartSoc":0,
      "currency":"EUR",
      "greenShareHome":0.06182795698924731,
      "greenShareLoadpoints":0,
      "gridConfigured":true,
      "gridEnergy":4742.741400000001,
      "gridPower":349,
      "homePower":372,
      "loadpoints":[
         {
            "chargeCurrent":0,
            "chargeCurrents":[
               0,
               0,
               0
            ],
            "chargeDuration":0,
            "chargePower":0,
            "chargeTotalImport":913.088,
            "chargeVoltages":[
               232.3000030517578,
               0,
               0
            ],
            "chargedEnergy":0,
            "chargerFeatureHeating":false,
            "chargerFeatureIntegratedDevice":false,
            "chargerIcon":null,
            "charging":false,
            "connected":false,
            "connectedDuration":9223372036854775807,
            "disableThreshold":2730,
            "effectiveLimitSoc":80,
            "effectiveMaxCurrent":16,
            "effectiveMinCurrent":12,
            "effectivePlanSoc":0,
            "effectivePlanTime":"0001-01-01T00:00:00Z",
            "effectivePriority":0,
            "enableThreshold":-40,
            "enabled":false,
            "guardAction":"inactive",
            "guardRemaining":0,
            "limitEnergy":0,
            "limitSoc":0,
            "maxCurrent":16,
            "minCurrent":12,
            "mode":"pv",
            "phaseAction":"inactive",
            "phaseRemaining":0,
            "phasesActive":1,
            "phasesConfigured":null,
            "phasesEnabled":1,
            "planEnergy":0,
            "planProjectedStart":"0001-01-01T00:00:00Z",
            "planTime":"0001-01-01T00:00:00Z",
            "priority":0,
            "pvAction":"inactive",
            "pvRemaining":0,
            "sessionCo2PerKWh":null,
            "sessionEnergy":0,
            "sessionPrice":null,
            "sessionPricePerKWh":null,
            "sessionSolarPercentage":0,
            "title":"Wallbox",
            "vehicleCapacity":77,
            "vehicleClimaterActive":null,
            "vehicleDetectionActive":false,
            "vehicleFeatureOffline":false,
            "vehicleIcon":"car",
            "vehicleName":"ID4",
            "vehicleOdometer":4991,
            "vehiclePresent":true,
            "vehicleRange":0,
            "vehicleSoc":0,
            "vehicleTargetSoc":0,
            "vehicleTitle":"ID4"
         }
      ],
      "prioritySoc":0,
      "pv":[
         {
            "power":0,
            "energy":13.530000000000001
         },
         {
            "power":18
         },
         {
            "power":5,
            "energy":733.992055600754
         }
      ],
      "pvConfigured":true,
      "pvEnergy":747.522055600754,
      "pvPower":23,
      "residualPower":0,
      "siteTitle":"Zuhause",
      "smartCostActive":false,
      "smartCostLimit":0,
      "smartCostType":null,
      "sponsor":"All-Ex-U",
      "sponsorTokenExpires":0,
      "statistics":{
         "30d":{
            "avgCo2":196.03950316021044,
            "avgPrice":0.36699045533777874,
            "chargedKWh":349.8340000000004,
            "solarPercentage":2.240155743799127
         },
         "365d":{
            "avgCo2":164.38214214198246,
            "avgPrice":0.3711432069344261,
            "chargedKWh":890.5040000000007,
            "solarPercentage":1.1339352865139831
         },
         "total":{
            "avgCo2":164.38214214198246,
            "avgPrice":0.3711432069344261,
            "chargedKWh":890.5040000000007,
            "solarPercentage":1.1339352865139831
         }
      },
      "tariffCo2":170,
      "tariffCo2Home":159.48924731182797,
      "tariffCo2Loadpoints":170,
      "tariffFeedIn":0,
      "tariffGrid":0.3754,
      "tariffPriceHome":0.3521897849462366,
      "tariffPriceLoadpoints":0.3754,
      "vehicles":{
         "ID4":{
            "title":"ID4",
            "minSoc":45,
            "limitSoc":80
         }
      },
      "version":"0.123.3"
   }
}

StefanSchoof, 2024-01-02:
Meintest du im loadpoint.publish? Habe da habe ich nur ein `lp` und kein `site`

```
	select {
	case lp.uiChan <- p:
	default:
		go func() {
			lp.log.INFO.Printf("async publish: %s %v", key, val)
			lp.uiChan <- p
		}()
	}
```

```
[main  ] INFO 2024/01/02 11:23:25 evcc 0.123.3 (f517a8f3)
[main  ] INFO 2024/01/02 11:23:25 using config file: ../plan.evcc.yaml
[main  ] INFO 2024/01/02 11:23:25 starting ui and api at :7070
[db    ] INFO 2024/01/02 11:23:25 using sqlite database: /home/stefan/.evcc/evcc.db
[lp-1  ] WARN 2024/01/02 11:23:25 phases not configured, assuming 3p
[lp-2  ] WARN 2024/01/02 11:23:25 phases not configured, assuming 3p
[site  ] WARN 2024/01/02 11:23:25 vehicle 'Vehicle no SoC no Capacity' does not support automatic detection
[site  ] WARN 2024/01/02 11:23:25 vehicle 'Vehicle no SoC with Capacity' does not support automatic detection
[site  ] WARN 2024/01/02 11:23:25 vehicle 'Vehicle with SoC no Capacity' does not support automatic detection
[site  ] WARN 2024/01/02 11:23:25 vehicle 'Vehicle with SoC with Capacity' does not support automatic detection
[site  ] WARN 2024/01/02 11:23:25 vehicle 'Vehicle with SoC with Massive Capacity' does not support automatic detection
[site  ] INFO 2024/01/02 11:23:25 site config:
[site  ] INFO 2024/01/02 11:23:25   meters:      grid ✓ pv ✗ battery ✗
[site  ] INFO 2024/01/02 11:23:25     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/01/02 11:23:25   vehicles:
[site  ] INFO 2024/01/02 11:23:25     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2024/01/02 11:23:25     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2024/01/02 11:23:25     vehicle 3: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2024/01/02 11:23:25     vehicle 4: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2024/01/02 11:23:25     vehicle 5: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/01/02 11:23:25 loadpoint 1:
[lp-1  ] INFO 2024/01/02 11:23:25   mode:        off
[lp-1  ] INFO 2024/01/02 11:23:25   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/01/02 11:23:25   meters:      charge ✗
[lp-2  ] INFO 2024/01/02 11:23:25 loadpoint 2:
[lp-2  ] INFO 2024/01/02 11:23:25   mode:        off
[lp-2  ] INFO 2024/01/02 11:23:25   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-2  ] INFO 2024/01/02 11:23:25   meters:      charge ✗
[lp-2  ] INFO 2024/01/02 11:23:25 async publish: limitEnergy 0
[lp-2  ] INFO 2024/01/02 11:23:25 async publish: minCurrent 6
[lp-1  ] INFO 2024/01/02 11:23:25 async publish: phasesActive 3
[lp-1  ] INFO 2024/01/02 11:23:25 async publish: phaseAction inactive
[lp-2  ] INFO 2024/01/02 11:23:25 async publish: chargerFeatureHeating false
[lp-2  ] INFO 2024/01/02 11:23:25 async publish: maxCurrent 16
[lp-2  ] INFO 2024/01/02 11:23:25 async publish: chargerIcon 
[lp-2  ] INFO 2024/01/02 11:23:25 async publish: enableThreshold 0
[lp-2  ] INFO 2024/01/02 11:23:25 async publish: vehicleTitle 
[lp-2  ] INFO 2024/01/02 11:23:25 async publish: vehicleIcon 
[lp-2  ] INFO 2024/01/02 11:23:25 async publish: chargerFeatureIntegratedDevice false
[lp-2  ] INFO 2024/01/02 11:23:25 async publish: planEnergy 0
[lp-2  ] INFO 2024/01/02 11:23:25 async publish: vehicleName 
[lp-2  ] INFO 2024/01/02 11:23:25 async publish: phaseAction inactive
[lp-2  ] INFO 2024/01/02 11:23:25 async publish: limitSoc 0
[lp-2  ] INFO 2024/01/02 11:23:25 async publish: vehicleCapacity 0
[lp-2  ] INFO 2024/01/02 11:23:25 async publish: vehiclePresent false
[lp-2  ] INFO 2024/01/02 11:23:25 async publish: disableThreshold 0
[lp-2  ] INFO 2024/01/02 11:23:25 async publish: phasesActive 3
[lp-2  ] INFO 2024/01/02 11:23:25 async publish: vehicleOdometer 0
[lp-1  ] INFO 2024/01/02 11:23:25 async publish: phaseRemaining 0s
[lp-1  ] INFO 2024/01/02 11:23:25 async publish: planEnergy 0
[lp-1  ] INFO 2024/01/02 11:23:25 async publish: pvAction inactive
[lp-1  ] INFO 2024/01/02 11:23:25 async publish: pvRemaining 0s
[lp-1  ] INFO 2024/01/02 11:23:25 async publish: limitEnergy 0
[lp-1  ] INFO 2024/01/02 11:23:25 async publish: limitSoc 0
[lp-1  ] INFO 2024/01/02 11:23:25 async publish: title Loadpoint
[lp-2  ] INFO 2024/01/02 11:23:25 async publish: title Loadpoint with SoC
[... 411 lines cut ...]
[lp-2  ] INFO 2024/01/02 11:23:26 async publish: planTime 0001-01-01 00:00:00 +0000 UTC
[lp-2  ] INFO 2024/01/02 11:23:26 async publish: mode off
[lp-2  ] INFO 2024/01/02 11:23:26 async publish: planEnergy 0
[lp-2  ] INFO 2024/01/02 11:23:26 async publish: planProjectedStart 0001-01-01 00:00:00 +0000 UTC
[lp-1  ] INFO 2024/01/02 11:23:27 async publish: sessionPricePerKWh <nil>
[lp-1  ] INFO 2024/01/02 11:23:27 async publish: sessionPrice <nil>
[lp-1  ] INFO 2024/01/02 11:23:27 async publish: effectivePriority 0
[lp-1  ] INFO 2024/01/02 11:23:27 async publish: sessionCo2PerKWh <nil>
[lp-1  ] INFO 2024/01/02 11:23:27 async publish: enabled false
[lp-1  ] INFO 2024/01/02 11:23:27 async publish: chargedEnergy 0
[lp-1  ] INFO 2024/01/02 11:23:27 async publish: charging false
[lp-1  ] INFO 2024/01/02 11:23:27 async publish: planEnergy 0
[lp-1  ] INFO 2024/01/02 11:23:27 async publish: mode off
[lp-1  ] INFO 2024/01/02 11:23:27 async publish: effectiveMinCurrent 6
[lp-1  ] INFO 2024/01/02 11:23:27 async publish: sessionSolarPercentage 0
[lp-1  ] INFO 2024/01/02 11:23:27 async publish: chargeDuration 0s
[lp-1  ] INFO 2024/01/02 11:23:27 async publish: connected true
[... cut ...] urrent 16
[lp-1  ] INFO 2024/01/02 11:23:27 async publish: effectivePlanSoc 0
[lp-1  ] INFO 2024/01/02 11:23:27 async publish: effectivePlanTime 0001-01-01 00:00:00 +0000 UTC
[lp-1  ] INFO 2024/01/02 11:23:27 async publish: effectiveLimitSoc 100
[lp-1  ] INFO 2024/01/02 11:23:27 async publish: planProjectedStart 0001-01-01 00:00:00 +0000 UTC
[lp-1  ] INFO 2024/01/02 11:23:27 async publish: sessionEnergy 0
[lp-1  ] INFO 2024/01/02 11:23:27 async publish: planTime 0001-01-01 00:00:00 +0000 UTC
```

closed 2024-01-02 by andig