evcc-io/evcc #11331: cannot create charger 'connectiq'

vfricke, 2023-12-29:
### Describe the bug

Nach update auf 0.123.2 bekomme ich folgende Fehlermeldung nachdem ich den Key uri geändert habe:

cannot create charger 'Parkplatz#9'
cannot create charger 'connectiq'
1 error(s) decoding: * '' has invalid keys: host, port

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
evcc.yaml: 
network:  
  schema: http  
  host:   0.0.0.0  
  port:   7070  
  
interval: 25s # control cycle interval  
  
log: info  
  
tariffs:  
  currency: EUR # (default EUR)  
  grid:  
    # static grid price  
    type: fixed  
    price: 0.38 # [currency]/kWh  
  
  feedin:  
    # rate for feeding excess (pv) energy to the grid  
    type: fixed  
    price: 0.133 # [currency]/kWh  
      
meters:  
- type: template  
  template: sma-energy-meter  
  usage: grid   
  host: 192.168.178.28  # IP-Adresse oder Hostname  
  name: grid1  
- type: template  
  template: sma-inverter  
  usage: pv   
  host: 192.168.178.26  # IP-Adresse oder Hostname  
  password: *****  
  name: pv2  
  
chargers:  
- type: connectiq  
  host: 192.168.178.31  
  port: 4444   
  name: Parkplatz#9  
    
- type: connectiq  
  host: 192.168.178.37  
  port: 4444  
  name: Schuppen#4  
    
loadpoints:  
- title: Schuppen  
  charger: Schuppen#4  
  mode: pv  
  mincurrent: 6  
  maxcurrent: 16  
  phases: 1  
    
- title: Parkplatz  
  charger: Parkplatz#9  
  mode: now  
  mincurrent: 6  
  maxcurrent: 16  
  phases: 3  
    
site:  
  title: Zuhause  
  meters:  
    grid: grid1  
    pvs:  
    - pv2
```

### Log details

```text
[main  ] INFO 2023/12/29 18:02:39 evcc 0.123.2
[main  ] INFO 2023/12/29 18:02:39 using config file: /etc/evcc.yaml
[main  ] INFO 2023/12/29 18:02:39 starting ui and api at :7070
[db    ] INFO 2023/12/29 18:02:42 using sqlite database: /home/pi/.evcc/evcc.db
[main  ] FATAL 2023/12/29 18:02:45 cannot create charger 'Schuppen#4': cannot create charger 'connectiq': 1 error(s) decoding:

* '' has invalid keys: host, port
[main  ] FATAL 2023/12/29 18:02:45 will attempt restart in: 5m0s
```

### What type of operating system are you running?

Linux

### Version

 evcc version 0.123.2

VolkerK62, 2023-12-29:
Mit `uri` war nur der `network` Teil betroffen.

vfricke, 2023-12-29:
Yepp - Das wars. Danke!

vfricke, 2023-12-29:
Super Support.

closed 2023-12-29 by vfricke