evcc-io/evcc #11333: Fronius Wattpilot "out of sync" after switching operation

eilmi, 2023-12-29:
### Describe the bug

After nearly every switching operation (Off -> Fast, Fast -> Solar, ....) evcc reports, that the charger is "out of sync"
most of the time it's "charger out of sync: expected disabled, got enabled"

strangely there aren't any signs of this in the output of the docker container at all. (as log output below shows) 

Problem also persists when using an interval >60s and disabling auto phase switching. 

Already tried resetting the wallbox with no luck. WiFi Accespoint within 2m of the Wattpilot so there shouldn't be any connection issues.

After switching back to version 0.194.4 the problem disappeared completely. 

### Steps to reproduce

1. install latest firmware of evcc
2. connect Tesla Model Y to Fronius Wattpilot
3. change charging mode and wait for a few seconds

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: XXX

interval: 30s # control cycle interval

meters:
- type: template
  template: fronius-solarapi-v1 
  usage: grid  
  host: 192.168.1.X 
  name: grid1
- type: template
  template: fronius-solarapi-v1 
  usage: pv  
  host: 192.168.1.X  
  name: pv2

chargers:
- type: template
  template: fronius-wattpilot 
  host: 192.168.1.X  
  password: xxx
  name: wallbox4

vehicles:
- type: template
  template: tesla 
  title: xxx  
  icon: car  
  accessToken: xxx
  refreshToken: xxx
  capacity: 77  
  phases: 3  
  mode: now  
  minCurrent: 5  
  maxCurrent: 16  
  name: ev3

loadpoints:
- title: Garage
  charger: wallbox4
  mode: now
  phases: 0
  mincurrent: 6
  maxcurrent: 16
  guardduration: 2

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pv:
    - pv2

tariffs:
  grid:

    type: awattar
    region: at # optional, choose at for Austria
    charges: 0.10 # optional, additional charges per kWh
    tax: 0 # optional, additional tax (0.1 for 10%)

mqtt:
  broker: 192.168.1.X
  topic: evcc
  user: evcc
  password: xxx
```

### Log details

```text
[lp-1  ] DEBUG 2023/12/28 19:05:20 charge power: 0W
[site  ] DEBUG 2023/12/28 19:05:20 pv power: 0W
[site  ] DEBUG 2023/12/28 19:05:21 grid power: 302W
[site  ] DEBUG 2023/12/28 19:05:21 site power: 302W
[lp-1  ] DEBUG 2023/12/28 19:05:21 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/12/28 19:05:21 charger status: B
[lp-1  ] DEBUG 2023/12/28 19:05:21 pv charge current: 0A = 0A + -1.31A (302W @ 1p)
[site  ] DEBUG 2023/12/28 19:05:50 ----
[lp-1  ] DEBUG 2023/12/28 19:05:50 charge power: 0W
[site  ] DEBUG 2023/12/28 19:05:50 pv power: 0W
[site  ] DEBUG 2023/12/28 19:05:50 grid power: 302W
[site  ] DEBUG 2023/12/28 19:05:50 site power: 302W
[lp-1  ] DEBUG 2023/12/28 19:05:50 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/12/28 19:05:50 charger status: B
[lp-1  ] DEBUG 2023/12/28 19:05:50 pv charge current: 0A = 0A + -1.31A (302W @ 1p)
[site  ] DEBUG 2023/12/28 19:06:20 ----
[lp-1  ] DEBUG 2023/12/28 19:06:20 charge power: 0W
[site  ] DEBUG 2023/12/28 19:06:20 pv power: 0W
[site  ] DEBUG 2023/12/28 19:06:20 grid power: 297W
[site  ] DEBUG 2023/12/28 19:06:20 site power: 297W
[lp-1  ] DEBUG 2023/12/28 19:06:20 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/12/28 19:06:20 charger status: B
[lp-1  ] DEBUG 2023/12/28 19:06:20 pv charge current: 0A = 0A + -1.29A (297W @ 1p)
[site  ] DEBUG 2023/12/28 19:06:50 ----
[lp-1  ] DEBUG 2023/12/28 19:06:50 charge power: 0W
[site  ] DEBUG 2023/12/28 19:06:50 pv power: 0W
[site  ] DEBUG 2023/12/28 19:06:50 grid power: 300W
[site  ] DEBUG 2023/12/28 19:06:50 site power: 300W
[lp-1  ] DEBUG 2023/12/28 19:06:50 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/12/28 19:06:50 charger status: B
[lp-1  ] DEBUG 2023/12/28 19:06:50 pv charge current: 0A = 0A + -1.31A (300W @ 1p)
```

### What type of operating system are you running?

Linux

### Version

0.121.0

eilmi, 2023-12-29:
of course i meant version 0.119.4 which works without any problems
sorry!

andig, 2023-12-29:
Das Log passt nicht zur Fehlerbeschreibung

closed 2023-12-29 by andig

eilmi, 2023-12-29:
Ernsthaft?

wie bereits geschrieben, listet das Log in keiner einzelnen Zeile, dass Problem auf, alles was kommt ist ein Orangenes Dreieck in der UI, das die Fehlermeldung von sich gibt, während im Log weiterhin so getan wird, als würde der charger nachwievor Daten übermitteln, wobei die halt nicht mehr der Wahrheit entsprechen, anstelle wird der letzte Valide wert halt auf ewig wiederholt....

andig, 2023-12-29:
Wir brauchen ein Log vom Fehler, nicht irgendeins

eilmi, 2023-12-30:
gesamtes Logfile:

Creating network "evcc_default" with the default driver
Creating evcc ... 
Creating evcc ... done
Attaching to evcc
[36mevcc    |[0m [main  ] INFO 2023/12/30 08:08:03 evcc 0.123.2
[36mevcc    |[0m [main  ] INFO 2023/12/30 08:08:03 using config file: /etc/evcc.yaml
[36mevcc    |[0m [main  ] INFO 2023/12/30 08:08:03 starting ui and api at :7070
[36mevcc    |[0m [db    ] INFO 2023/12/30 08:08:03 using sqlite database: /root/.evcc/evcc.db
[36mevcc    |[0m [mqtt  ] INFO 2023/12/30 08:08:03 connecting evcc-1213725396 at tcp://172.17.0.1:1883
[36mevcc    |[0m [mqtt  ] DEBUG 2023/12/30 08:08:03 tcp://172.17.0.1:1883 connected
[36mevcc    |[0m [site  ] INFO 2023/12/30 08:08:10 site config:
[36mevcc    |[0m [site  ] INFO 2023/12/30 08:08:10   meters:      grid ✓ pv ✓ battery ✗
[36mevcc    |[0m [site  ] INFO 2023/12/30 08:08:10     grid:      power ✓ energy ✗ currents ✗
[36mevcc    |[0m [site  ] INFO 2023/12/30 08:08:10     pv 1:      power ✓ energy ✗ currents ✗
[36mevcc    |[0m [site  ] INFO 2023/12/30 08:08:10   vehicles:
[36mevcc    |[0m [site  ] INFO 2023/12/30 08:08:10     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[36mevcc    |[0m [lp-1  ] INFO 2023/12/30 08:08:10 loadpoint 1:
[36mevcc    |[0m [lp-1  ] INFO 2023/12/30 08:08:10   mode:        pv
[36mevcc    |[0m [lp-1  ] INFO 2023/12/30 08:08:10   charger:     power ✓ energy ✗ currents ✓ phases ✓ wakeup ✗
[36mevcc    |[0m [lp-1  ] INFO 2023/12/30 08:08:10   meters:      charge ✓
[36mevcc    |[0m [lp-1  ] INFO 2023/12/30 08:08:10     charge:    power ✓ energy ✗ currents ✓
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:10 phase timer inactive
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:10 pv timer inactive
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:10 guard timer inactive
[36mevcc    |[0m [site  ] DEBUG 2023/12/30 08:08:10 ----
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:10 charge power: 0W
[36mevcc    |[0m [site  ] DEBUG 2023/12/30 08:08:10 pv power: 93W
[36mevcc    |[0m [site  ] DEBUG 2023/12/30 08:08:10 grid power: 506W
[36mevcc    |[0m [site  ] DEBUG 2023/12/30 08:08:10 site power: 506W
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:10 charge currents: [0 0 0]A
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:10 charger status: B
[36mevcc    |[0m [lp-1  ] INFO 2023/12/30 08:08:10 car connected
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:10 vehicle api refresh
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:10 pv timer elapse
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:10 pv timer inactive
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:10 charger: guard elapse
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:10 guard timer inactive
[36mevcc    |[0m [main  ] ERROR 2023/12/30 08:08:10 vehicle status: asleep
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:10 available power -506W < 4140W min 3p threshold
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:10 phase scale1p in 0s
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:10 switched phases: 1p
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:10 phase timer inactive
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:10 pv charge current: 0A = 0A + -2.2A (506W @ 1p)
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:10 pv enable timer reset
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:10 pv timer inactive
[36mevcc    |[0m [lp-1  ] INFO 2023/12/30 08:08:22 vehicle updated: unknown -> Tessy
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:22 set charge mode: now
[36mevcc    |[0m [site  ] DEBUG 2023/12/30 08:08:22 ----
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:22 charge power: 0W
[36mevcc    |[0m [site  ] DEBUG 2023/12/30 08:08:23 pv power: 95W
[36mevcc    |[0m [site  ] DEBUG 2023/12/30 08:08:23 grid power: 505W
[36mevcc    |[0m [site  ] DEBUG 2023/12/30 08:08:23 site power: 505W
[36mevcc    |[0m [lp-1  ] ERROR 2023/12/30 08:08:23 vehicle odometer: asleep
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:23 charge currents: [0 0 0]A
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:23 charger status: B
[36mevcc    |[0m [lp-1  ] ERROR 2023/12/30 08:08:23 vehicle soc: asleep
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:23 set ev3 plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:23 switched phases: 3p
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:23 max charge current: 16A
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:23 charger enable
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:23 wake-up timer: start
[36mevcc    |[0m [site  ] DEBUG 2023/12/30 08:08:40 ----
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:40 charge power: 0W
[36mevcc    |[0m [site  ] DEBUG 2023/12/30 08:08:40 pv power: 98W
[36mevcc    |[0m [site  ] DEBUG 2023/12/30 08:08:40 grid power: 390W
[36mevcc    |[0m [site  ] DEBUG 2023/12/30 08:08:40 site power: 390W
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:40 charge currents: [0 0 0]A
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:08:40 charger status: B
[... 362 lines cut ...]
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:17:40 charge power: 10759W
[36mevcc    |[0m [site  ] DEBUG 2023/12/30 08:17:41 pv power: 159W
[36mevcc    |[0m [site  ] DEBUG 2023/12/30 08:17:41 grid power: 344W
[36mevcc    |[0m [site  ] DEBUG 2023/12/30 08:17:41 site power: 344W
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:17:41 charge currents: [16 16 16]A
[... cut ...]
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:18:11 detected active phases: 3p
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:18:11 charger status: C
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:18:11 soc estimated: 68.10% (vehicle: 68.00%)
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:18:11 vehicle soc: 68%
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:18:11 vehicle soc limit: 80%
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:18:11 vehicle range: 347km
[36mevcc    |[0m [lp-1  ] DEBUG 2023/12/30 08:18:11 set ev3 plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT

andig, 2023-12-30:
> wie bereits geschrieben, listet das Log in keiner einzelnen Zeile, dass Problem auf

Jetzt bin ich verwirrt- wie passt das zusammen?

eilmi, 2023-12-30:
Ok ja stimmt, ein einziges Mal wird es reported, sorry!!

ab dieser Zeile stimmen dann die "charger power: 10759W" Einträge halt nicht mehr, in Wirklichkeit hat das Auto aufgehört zu laden und es müssten 0W sein

eilmi, 2023-12-30:
[image]

Auch die In und Out Werte passen wenn der charger "out of sync" geht dann absolut nicht mehr zusammen

andig, 2023-12-30:
> ab dieser Zeile stimmen dann die "charger power: 10759W" Einträge halt nicht mehr, in Wirklichkeit hat das Auto aufgehört zu laden und es müssten 0W sein

Die Leistung kommt von der Wallbox. Scheint also, als würde noch geladen.

@mabunixda evtl. irgendein Implementierungsfehler?

mabunixda, 2023-12-31:
Ja es könnte das Caching der Werte da genau diesen Fehler hervorrufen.Schau ich mir nach den Feiertagen an

mabunixda, 2024-01-16:
Ich habe nun mit Caching-Timeouts versucht das anders zu implementieren - aber das leert zu viele Daten dann. 
Ganz klar ist mir nicht, warum das passiert. Es kann nur - wieder einmal - ein Netzwerkproblem im Grunde sein, da ja der Wattpilot eine ziemliche schlechte WLan Implementierung hat.

@eilmi tritt das bei dir regelmäßig auf? Dann könnten wir sonst mal probieren das Loglevel der API Implementierung raufzudrehen um mehr Infos rauszubekommen.

eilmi, 2024-01-16:
Ist leider die Anlage vom Nachbarn, werd's mir anschauen, sobald ich nicht mehr rund um die Uhr für die Uni lerne....

Aber ja, war regelmäßig, quasi jede einzelne Schalthandlung ist es danach "out of sync" gegangen, und war nurnoch durch einen Neustart des docker-containers zu beheben

eilmi, 2024-01-16:
Netzwerkprobleme kann ich (soweit der Wattpilot von Fronius halbwegs einen stabilen Wifi-chip hat) übrigens nahezu ausschließen, Wifi-AccessPoint und Steckdose sind vllt. maximal 2m voneinander entfernt....

außerdem würd das ja nicht erklären, wiesos mit der alten Version seit 2 Wochen ohne jegliche Probleme funktioniert oder?

Ist zwar nicht mein Auto, aber ich kann mir sicher mal ein paar Nachmittage organisieren, in denen ich mit verschiedensten Konfigurationen herumspielen kann (hab keine Erfahrung mit Go, aber an meinen Linux und co. Kenntnissen solls nicht scheitern ;-)

bitte einfach sagen, wie ich mich bestmöglich an der Behebung des Fehlers beteiligen kann!

(Hardware is übrigens ein Raspberry Pi 3 und eben evcc im docker-container)

eilmi, 2024-01-16:
habs allerdings auch auf deutlich potenterer x86 Hardware getestet und bin zu selbigem Ergebnis gekommen (würd daher den Raspberry als Bottleneck eher ausschließen, dessen Wifi is ja bekanntlich instabil genug, dass er sowieso per Kabel im Netz hängt)

mabunixda, 2024-01-17:
Ok gut für die Info! 
Was ist die alte Version, und was die aktuelle?

eilmi, 2024-01-17:
0.119.4 vs 0.123.2

mabunixda, 2024-01-17:
@eilmi Stoppt das Auto oder sollte evcc stoppen den Ladevorgang?

Spannend ist ja, dass es hier keine Änderungen in der wattpilot Bibliothek gab in der Kommunikation/Logik.

eilmi, 2024-01-17:
evcc stoppt den Ladevorgang, entweder aufgrund von PV-Änderungen oder wegen Interaktion mit der Weboberfläche

naja im ReleaseNote von  v0.119.5 gabs: "Wattpilot: make identify return trx instead of carId (https://github.com/evcc-io/evcc/pull/9712)" 
und in v0.120.0:  Wattpilot: fix nil on trx (https://github.com/evcc-io/evcc/pull/9830)

außerdem wurde die Wattpilot zwischen den beiden Releases von 1.5.0 auf 1.6.2 geupdated, nicht?

eilmi, 2024-01-17:
ich hab mir auch schon gedacht ich zieh mir den Source-Code und änder des wieder zurück auf 1.5.0 in nem aktuellen Release von evcc und probier ob's dann funktioniert, da ist halt die Frage kann das überhaupt noch funktionieren mit der alten Wattpilot-library in nem aktuellen evcc-release und hatte leider nichtmehr die Zeit es auszuprobieren :-/

mabunixda, 2024-01-18:
> evcc stoppt den Ladevorgang, entweder aufgrund von PV-Änderungen oder wegen Interaktion mit der Weboberfläche
> 
> naja im ReleaseNote von v0.119.5 gabs: "Wattpilot: make identify return trx instead of carId (#9712)" und in v0.120.0: Wattpilot: fix nil on trx (#9830)
> 
> außerdem wurde die Wattpilot-library zwischen den beiden Releases von 1.5.0 auf 1.6.2 geupdated, nicht? (entnommen von go.sum)

Ja aber diese Änderungen waren rein darauf, wie/welche Authentifizierungsinformationen vom Auto bzw Dongle ausgelesen werden. 
Die Kommunikationsänderung war mit 1.5.0 schon - und die ist ja in der Version von dir schon drinnen....

Prinzipiell hat der Wattpilot nur eine Information zum Laden, ob es erlaubt ist oder nicht. Ob das Auto dann zu dem Zeitpunkt auch lädt - oder evcc erst in 2h tarifbasiert zu Laden beginnt, ist dem Wattpilot egal.

Ein Grund, warum manchmal Out of Sync auftaucht, ist einfach erkärt, da die Kommunnikation per Websocket asynchron ist. Die Bibliothek hat, wenn evcc das Laden beendet, noch immer den Wert gespeichert, dass Laden aktiv ist - und das so lange, bis Wattpilot dann nach x Sekunden meldet, dass das Laden beendet ist/wurde. 
Sollte die Kommunkation hier Probleme haben, kann sein dass es "besser" ist, dass der Wert eben erst so aktualisiert wird, wie es implementiert ist. Es kann aber natürlich auch anders herum sein, dass Wattpilot das Laden beendet hat, das Update aber am Weg verloren ging - oder "nur" lange braucht.

eilmi, 2024-01-20:
Sorry da kann ich jetzt nicht zu 100% folgen.....
Zwischen meinen zwei Versionen (0.119.4 -> 0.123.2) wurden also keine Änderungen an der Implementierung der Kommunikation durchgenommen?

Also wurde auch früher die Antwort via Websocket abgewartet, bevor in evcc der Ladestopp signalisiert wurde?

Dann wird uns eh nichts anderes übrigbleiben als ne aktuelle Version zudebuggen oder?