evcc-io/evcc #11335: Laden Start über MQTT aus EVCC in HomeAssistant?

MarioHAGH, 2023-12-30:
Hallo,
ich habe folgendes Problem. Ich hätte gerne eine Statusmeldung bei Ladung Start aus evcc über MQTT.

Warum: Ich möchte sobald die Ladung startet (autm. über den dyn. Tarif) eine Meldung zum Home Assistant haben um meinen Hausakku zu "trennen" bzw. die Entladestrom auf 0 setzen, damit das Auto nicht den Hausakku leer zieht.

Würde über eine Meldung via MQTT gut in eine Automation zu machen sein. 

Vielleicht hat ja auch einer eine bessere Idee. Bekomme es aktuell nicht hin in der yaml. 

Danke und Grüße
Mario

VolkerK62, 2023-12-30:
Geht doch, die nötigen INformationen kommen über MQTT.
Kommt darauf an, wie die Ladung startet.
Geht das über den "Plan" `evcc/loadpoints/1/planActive` oder über "smartes Netzladen" `evcc/site/smartCostActive`?

andig, 2023-12-30:
…und in jedem Falle zeigt `charging` an, dass geladen wird.

closed 2023-12-30 by andig

MarioHAGH, 2023-12-30:
Okay das klingt schonmal super. Wie und wo genau muss ich es in der yaml einfügen? Ist es unter Massage oder Event? Werde da aus der Doku irgendwie nicht richtig schlau.