evcc-io/evcc #11337: Battery Control : SMA Sunny Island / Sunny Boy Storage

premultiply, 2023-12-30:
Betriebsbereichsabgrenzung mit relativen Leistungswerten (Prozentual mit zwei Nachkommastellen).

Im SI-Handbuch (funktioniert genauso beim SBS und vermutlich auch beim STP SE) beschreibt SMA dies bildlich so:
[image]

Die Bezeichnungen "Minimale Wirkleistung" und "Maximale Wirkleistung" sind hier übrigens maximal verwirrend.
Das versteht beim Lesen erstmal kein Mensch obwohl die Funktion dahinter genau das bietet was wir suchen.

Übersetzt für uns heisst das:
```
Modbus ID: 3

normal:    Register 44039  int32 := 10000;  Register 44041  int32 := -10000
hold:      Register 44039  int32 := 0;      Register 44041  int32 := -10000
charge:    Register 44039  int32 := -10000; Register 44041  int32 := -10000
```

Lustiger Effekt: Der WR macht hier bei Vorgabenänderungen sehr, sehr softe (=langsame) Übergänge.
Hier muss man auch ggf. durchaus etwas warten bis man einen Effekt sieht! Bis der Zielzustand erreicht wird kann es im Extremfall (zumindest mit den sonstigen Defaulteinstellungen meines Testsystems) ein paar Minuten dauern.

Diese Beeinflussung taucht im WR-Status als "Externe Wirkleistungsvorgabe 2" auf und bietet nochmal gesonderte Konfigurationsoptionen - auch für den Fallback. Auch hier ist per Default ein langes Timeouts (10 Minuten) bei mir konfiguriert gewesen dass ich nun auf 2 Minuten reduziert habe.

War hier übrigens auch ausreichend nur das Register 44039 alleine (**regelmäßig**) zu schreiben, denn 44041 bleibt in unserem Fall immer auf dem Default-Wert (-100%).

Auch diese (WO-)Register sind für diesen Zweck vorgesehen und landen nicht im Flash!

@andig Ref: https://github.com/evcc-io/evcc/discussions/10805#discussioncomment-7676975

_Originally posted by @premultiply in https://github.com/evcc-io/evcc/discussions/10805#discussioncomment-7685939_

andig, 2023-12-30:
Duplikat zu https://github.com/evcc-io/evcc/issues/10912, oder ist das hier etwas Anderes?

premultiply, 2023-12-31:
Funktionell sehr ähnlich aber völlig andere Implementierung über proprietäre "SMA-Register" nötig.

andig, 2023-12-31:
Wie oft muss das geschrieben werden? Wenns einen Tester gibt kann ich gerne PR machen. Die Leistungsangabe ist prozentual mit Faktor 100?

premultiply, 2023-12-31:
Nur innerhalb des Timeouts. Bei mir war das Default auf 10 Minuten.
Einfach in jedem Interval neu schreiben dürfte das einfachste sein.
Wir können das gerne direkt mit meinem SBS testen.

andig, 2023-12-31:
Brauchen wir ein neues Template oder könnte das in ein vorhandenes mit rein?

premultiply, 2023-12-31:
Beim Hybrid kann es das vorhandene sein.

Ich muss mal nachlesen und testen in wie weit auch die anderen Parameter vom Hybrid zu SBS und SI passen.
Im dümmsten Fall brauchen die je neue und eigene Modbus-Templates wenn man die Batteriesteuerung nutzen will.

Über Speedwire geht das ja leider nicht bzw. ist nicht dokumentiert oder bekannt wie.

premultiply, 2024-01-01:
Soweit ich jetzt stichprobenartig die Dokumente durchgeschaut habe scheinen die Register bei allen Modellen identisch zu sein.
Ich gucke mal ob ich das bestehende Hybrid-Template auf den vollen Funktionsumfang erweitern und lokal mit allen SMA-Geräten benutzen kann.

Nokius, 2024-01-04:
@andig kann SBS 2.5 und 3.7 testen. Meldet euch gerne.

closed 2024-01-07 by andig