evcc-io/evcc #11338: Hyundai seit Update auf 123.2. immer offline. Per App aber problemlos erreichbar

Matwas82, 2023-12-30:
### Describe the bug

Leider erscheint mein Hyundai Ioniq seit dem letzten Update immer offline. Warum hab ich bisher absolut nicht rausfinden können, die Diskussionen hier zu den ID3 usw hab ich alle durchgelesen, aber keine mögliche Lösung gefunden.

Bis zu diesem Update hat unser System immer komplett stabil funktioniert. Auto war auch immer verfügbar.

### Steps to reproduce

1. Version 0.123.2 installieren
2. Linux Ubuntu als System
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: 132a2d32ec1833b91dc0ed7836b0fd4a0ee58c8cb2a10b37990da322922b4871

interval: 10s # control cycle interval

sponsortoken: *****************

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- name: pv2
  type: custom
  power:
    source: http
    uri: http://pvserver:*******@*******
    regex: '(?s)aktuell.*?(\d+)<'
- name: grid1
  type: custom
  power: #power reading (W)
    source: http
    uri: http://192.168.178.107/cm?cmnd=Status%208
    jq: .StatusSNS.SML.curr_w
  

chargers:
- type: template
  template: alphatec 
  id: 1
  baudrate: 9600  
  comset: "8N1"
  host: 192.168.178.103  
  port: 502  
  modbus: rs485tcpip  
  name: wallbox4

vehicles:
- type: template
  template: hyundai 
  title: Ioniq  
  user: ***************
  password: ************
  capacity: 38  
  phases: 1  
  icon: car  
  cache: 15m  
  language: en  
  mode: minpv  
  minSoc: 20  
  targetSoc: 90  
  minCurrent: 6  
  maxCurrent: 16  
  name: ev3

loadpoints:
- title: Garage
  charger: wallbox4
  vehicle: ev3
  mode: minpv
  phases: 3
  mincurrent: 6
  maxcurrent: 16

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.2940 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.129 # [currency]/kWh
    
site:
  title: Daheim
  meters:
    grid: grid1
    pv:
    - pv2
```

### Log details

```text
matthias@matthias-desktop:~$ evcc --log debug
[main  ] INFO 2023/12/30 09:07:23 evcc 0.123.2
[main  ] INFO 2023/12/30 09:07:23 using config file: /etc/evcc.yaml
[main  ] INFO 2023/12/30 09:07:23 starting ui and api at :7070
[db    ] INFO 2023/12/30 09:07:24 using sqlite database: /home/matthias/.evcc/evcc.db
[main  ] ERROR 2023/12/30 09:07:24 creating vehicle ev3 failed: cannot create vehicle 'template': invalid key: targetsoc
[site  ] INFO 2023/12/30 09:07:24 site config:
[site  ] INFO 2023/12/30 09:07:24   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/12/30 09:07:24     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/12/30 09:07:24     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/12/30 09:07:24   vehicles:
[site  ] INFO 2023/12/30 09:07:24     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/12/30 09:07:24 loadpoint 1:
[lp-1  ] INFO 2023/12/30 09:07:24   mode:        minpv
[lp-1  ] INFO 2023/12/30 09:07:24   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/12/30 09:07:24   meters:      charge ✗
[lp-1  ] DEBUG 2023/12/30 09:07:24 phase timer inactive
[lp-1  ] DEBUG 2023/12/30 09:07:24 pv timer inactive
[lp-1  ] DEBUG 2023/12/30 09:07:24 guard timer inactive
[lp-1  ] INFO 2023/12/30 09:07:24 vehicle updated: unknown -> Ioniq (offline)
[site  ] WARN 2023/12/30 09:07:24 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2023/12/30 09:07:24 ----
[lp-1  ] DEBUG 2023/12/30 09:07:24 charge power: 0W
[site  ] DEBUG 2023/12/30 09:07:24 pv power: 57W
[site  ] DEBUG 2023/12/30 09:07:24 grid power: 272W
[site  ] DEBUG 2023/12/30 09:07:24 site power: 272W
[lp-1  ] DEBUG 2023/12/30 09:07:24 charger status: A
[lp-1  ] INFO 2023/12/30 09:07:24 car disconnected
[lp-1  ] DEBUG 2023/12/30 09:07:24 set session soc limit: 0
[lp-1  ] DEBUG 2023/12/30 09:07:24 set session energy limit: 0
[site  ] DEBUG 2023/12/30 09:07:34 ----
[lp-1  ] DEBUG 2023/12/30 09:07:34 charge power: 0W
[site  ] DEBUG 2023/12/30 09:07:34 pv power: 56W
[site  ] DEBUG 2023/12/30 09:07:34 grid power: 271W
[site  ] DEBUG 2023/12/30 09:07:34 site power: 271W
[lp-1  ] DEBUG 2023/12/30 09:07:34 charger status: A
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.123.2

Matwas82, 2023-12-30:
Hier noch kurz der Auszug aus dem Debug car:

'''matthias@matthias-desktop:~$ sudo evcc -l debug vehicle
[sudo] password for matthias: 
[main  ] INFO 2023/12/30 09:12:06 evcc 0.123.2
[main  ] INFO 2023/12/30 09:12:06 using config file: /etc/evcc.yaml
[db    ] INFO 2023/12/30 09:12:07 using sqlite database: /root/.evcc/evcc.db
[main  ] ERROR 2023/12/30 09:12:07 creating vehicle ev3 failed: cannot create vehicle 'template': invalid key: targetsoc
.Soc:     vehicle not available: cannot create vehicle 'template': invalid key: targetsoc
Capacity: 38.0kWh
Features: [Offline]'''

VolkerK62, 2023-12-30:
Steht doch alles da:  `invalid key: targetsoc`
und `minsoc` auch gleich wegwerfen

Beide Werte werden jetzt im UI eingestellt.
Bei angestecktem Fahrzeug: Loadpoint unten Mitte "Plan". Dann Tab "Ankunft" auswählen.

Matwas82, 2023-12-30:
Danke, das ging ja schnell. Dann hab ich das wohl bei der neuen Version irgendwie überlesen.

closed 2023-12-30 by Matwas82