evcc-io/evcc #11343: BMW i3 100% Ladung stoppt bei 99%, kein Aufwärmen mehr über Wallbox möglich

puzich, 2023-12-30:
### Describe the bug

Bei unserem BMW i3 kann man kein Ladeziel wie zB 80% einstellen. Einmal angesteckt lädt dieser auf 100%. Bei uns ist dies zumindest in den Wintermonaten auch nötig wegen der Reichweite. Im EVCC ist die Ladegrenze auf 100% gestellt und bisher (und fragt mich bitte nicht, bis zu welcher EVCC Version das "bisher" galt) hat das Auto auch ohne Probleme auf 100% geladen. Seit einiger Zeit wird aber bei 99% abgebrochen, da EVCC anscheinend über die BMW API 100% gemeldet bekommt. Die BMW App und das Auto zeigen aber nur 99% Ladung an und einen Abbruch beim Ladevorgang. Wenn das Auto nun auch noch vorklimatisiert wird, dann zieht es nicht den Strom aus der Wallbox (was eigentlich der Gedanke war) sondern aus dem Akku. 

### Steps to reproduce

wie oben im Text beschrieben

### Configuration details

```yaml
<details><summary>Konfiguration (/etc/evcc.yaml)</summary>

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

#log: debug
log: info
#levels:
#  cache: error
##  modbus: trace
##log.easee: debug

# unique installation id
plant: *****
sponsortoken: *****
#interval: 30s # control cycle interval
#interval: 10s # control cycle interval
interval: 15s # control cycle interval

site:
  title: Home
  meters:
    grid: my_grid
    pv: my_pv
    battery: my_battery
  ##prioritySoc: 50
#  residualPower: 350
  residualPower: 100
  ##bufferSoc: 80
  ##batteryDischargeControl: true

meters:
  - name: my_grid
    type: template
    template: sofarsolar-g3
    usage: grid
    modbus: rs485serial
    id: 1
    device: /dev/ttyUSB0
    baudrate: 9600 # Pruefe Geraeteeinstellungen, typische Werte sind 9600, 19200, 38400, 57600, 115200
    comset: "8N1" # Kommunikationsparameter fuer den Adapter
    delay: 1s # Optional
  - name: my_pv
    type: template
    template: sofarsolar-g3
    usage: pv
    modbus: rs485serial
    id: 1
    device: /dev/ttyUSB0
    baudrate: 9600 # Pruefe Geraeteeinstellungen, typische Werte sind 9600, 19200, 38400, 57600, 115200
    comset: "8N1" # Kommunikationsparameter fuer den Adapter
    delay: 1s # Optional
  - name: my_battery
    type: template
    template: sofarsolar-g3
    usage: battery
    modbus: rs485serial
    id: 1
    device: /dev/ttyUSB0
    baudrate: 9600 # Pruefe Geraeteeinstellungen, typische Werte sind 9600, 19200, 38400, 57600, 115200
    comset: "8N1" # Kommunikationsparameter fuer den Adapter
    delay: 1s # Optional

chargers:
- name: wallbox1
  type: template
  template: easee
  user: *****
  password: *****
  charger: EHPW8R46
- name: wallbox2
  type: template
  template: easee
  user: *****
  password: *****
  charger: EHLC2AVR

vehicles:
- type: template
  template: bmw
  title: BMW i3
  user: *****
  password: *****
  vin: *****
#  minSoc: 21
#  targetSoc: 100
#  targetSoc: 85
  priority: 100
  name: bmw_i3
  capacity: 28
- type: template
  template: tesla
  title: Tesla ECTO-1
  accessToken: *****
  refreshToken: *****
#  minSoc: 25
#  targetSoc: 90
  priority: 10
  name: tesla_ecto1
  capacity: 78

loadpoints:
  - title: Carport links # display name for UI
    charger: wallbox1 # Wallbox Referenz
    vehicle: bmw_i3 # Referenz auf Standardfahrzeug
    mode: now # charge mode (off, now, minpv, pv)
    #resetOnDisconnect: true
    priority: 100
  - title: Carport rechts # display name for UI
    charger: wallbox2 # Wallbox Referenz
    vehicle: tesla_ecto1 # Referenz auf Standardfahrzeug
    mode: now # charge mode (off, now, minpv, pv)
    #resetOnDisconnect: true
    priority: 1
#    enable: # pv mode enable behavior
#      delay: 1m # threshold must be exceeded for this long
#      threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
#    disable: # pv mode disable behavior
#      delay: 5m # threshold must be exceeded for this long
#      threshold: 0 # maximum import power (W)
#    guardDuration: 5m # switch charger contactor not more often than this (default 5m)

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.279 # [currency]/kWh
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.082 # [currency]/kWh
  co2:
    type: grünstromindex
    zip: 21244

</details>
Version: `0.123.1`
```

### Log details

```text
Debug wird noch erstellt, bei dem nächsten Ladevorgang
```

### What type of operating system are you running?

Linux

### Version

0.123.1

andig, 2023-12-30:
Bitte Logfile, dann machen wir hier wieder auf.

Hofyyy, 2023-12-30:
Hi, der Fehler ist bekannt. Es geht im Ui darum wann das letzte Update aus der Cloud gezogen wird. Das Auto stoppt die Ladung weil voll. Die Cloud zeigt noch 99%. Dann hört der regelkreis auf.

andig, 2023-12-30:
> Man bräuchte ein letztes Cloudupdate nach einem Zeitpunkt X.

OP schreibt, dass auch die BMW App 99% anzeigt. Auch dann hat das Fahrzeug die Ladung beendet, die Frage ginge dann wohl an BMW wie das sein kann...

puzich, 2023-12-30:
Ich habe mal das Log angehangen. Der Ladepunkt an dem das Auto hängt und auch das Auto ist auf Debug Level gestellt. Wie zu sehen ist, meldet die API 100% SoC, das Auto hat aber "nur" 99,5%. 

Die 0,5% weniger im Akku machen jetzt kein wirkliches Reichweitenproblem, sondern eher, dass das Auto auch den Strom zum vorheizen nicht mehr aus der Wallbox nimmt, sondern nun aus dem Akku, was bei Abfahrt dann 5-8% ausmachen kann.

[image]
[image]

andig, 2023-12-30:
> Wie zu sehen ist, meldet die API 100% SoC

Kann ich nicht sehen, aber spielt auch keine Rolle. Das **Auto** beendet den Ladevorgang, hat also nix mit evcc zu tun...

andig, 2023-12-30:
Ich muss mich bei Dir entschuldigen.  Ich bin im Logfile über diese Zeile gestolpert:

> [lp-1  ] DEBUG 2023/12/30 13:29:52 limitSoc reached: 100.0% > 100%

Früher war das so:

	return lp.Soc.target > 0 &&
		lp.Soc.target < 100 &&
		lp.vehicleSoc >= float64(lp.Soc.target)

Heute ist das so:

	limit := lp.effectiveLimitSoc()
	return limit > 0 && lp.vehicleSoc >= float64(limit)

Wir haben also die Bedingung, *NICHT* bei 100% abzuschalten verloren und genau da dürfte auch der Fehler liegen. Fix folgt.

puzich, 2023-12-30:
Hach, ich bin voll verknallt in dieses Projekt. Vielen vielen Dank für deinen Support!

Hofyyy, 2023-12-30:
Cooler fehler. Da habe ich wohl zu ungenau gelesen :-9

closed 2023-12-30 by andig

andig, 2023-12-30:
Da kommste auch nicht drauf…