evcc-io/evcc #11349: Evcc not starting, after update

MenosChris, 2023-12-30:
### Describe the bug

App does not start anymore in home assist after last update. What can I do? 
Please have look in my evcc.yaml

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
I have not changed the evcc.yaml
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/12/30 15:52:36 evcc 0.123.2
[main  ] INFO 2023/12/30 15:52:36 using config file: /config/evcc.yaml
[main  ] INFO 2023/12/30 15:52:36 starting ui and api at :7070
[main  ] FATAL 2023/12/30 15:52:37 failed configuring loadpoints: failed configuring loadpoint: 1 error(s) decoding:

* '' has invalid keys: resetondisconnect
[main  ] FATAL 2023/12/30 15:52:37 will attempt restart in: 5m0s
```

### What type of operating system are you running?

Linux

### Version

0.123.2

MenosChris, 2023-12-30:
It is an home assistant add on and it worked for many months until now.

StefanSchoof, 2023-12-30:
Please read the release notes: https://github.com/evcc-io/evcc/releases/tag/0.123.0

closed 2023-12-30 by andig