evcc-io/evcc #11352: Out > In (Ausgangsleistung größer als Eingangsleistung) in Visualisierung

g-mocken, 2023-12-30:
Während die Regelung usw. gut funktioniert, habe ich einen komischen Offset in der Visualisierung: es geht manchmal scheinbar mehr Leistung ins Auto, als PV und Netz gemeinsam liefern, und der Hausverbrauch steht dabei auf null.

Das Problem scheint daher zu rühren, dass die Ladeleistung mangels separatem Zähler wohl aus der Wallbox-Stromvorgabe errechnet wird, das Auto aber sich real deutlich weniger gönnt als es dürfte (warum auch immer!).

Perfekt erfassen kann man das natürlich nicht ohne einen weiteren Zähler, aber wenn die geratene Ladeleistung zumindest auf die Summe PV+Bezug gedeckelt wäre, wäre sie weniger falsch und die Summen würden stimmen.

(Wenn beim Fahrzeug ein Ladestrom-Offset oder -Korrekturfaktor möglich wäre, wäre das natürlich noch besser, allerdings müsste man da natürlich erstmal das eigene Auto genau vermessen, was nicht unbedingt der Mühe wert ist)

Anbei noch meine config und der Screenshot (darin ist übrigens auch etwas merkwürdig, dass "In" in kW und "Out" in W angezeigt wird, außerdem ist 1,1 kW +0,5 kW nicht 1,5 kW).

```
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info 
levels:
  cache: error

# unique installation id
plant: *****

interval: 10s # control cycle interval

meters:
- type: template
  template: sma-energy-meter 
  usage: grid  
  host: smaXXX.fritz.box  
  name: grid1
- type: template
  template: sma-inverter 
  usage: pv  
  host: smaYYY.fritz.box  
  password: *****
  name: pv2

chargers:
- type: template
  template: phoenix-em-eth 
  id: 180  
  host: walli.fritz.box 
  port: 502  
  modbus: tcpip  
  name: wallbox3

loadpoints:
- title: Garage
  charger: wallbox3
  phases: 3 
  mincurrent: 6
  maxcurrent: 16
  soc:
    estimate: true
  enable:
      delay: 2m
      threshold: -1400
  disable:
      delay: 10m
      threshold: 1000 
  guardduration: 5m

site:
- title: Anlage
  meters:
    grid: grid1
    pv:
    - pv2
  residualPower: 0 
  
vehicles:    
- name: zoe1ph
  title: Renault ZOE (1-phasig)
  type: renault
  user: *****
  password: *****
  vin: *****
  capacity: 22
  phases: 1
  onIdentify:
    mincurrent: 8
```

[image] 

Alles nichts dramatisches, aber vielleicht kann sich das ja mal einer kurz angucken, vielleicht mache ich ja doch noch was grundlegend falsch in meiner Konfiguration ...?

andig, 2023-12-30:
Was sagt /api/state in dem Moment?

g-mocken, 2023-12-30:
Werde bei nächster Gelegenheit (braucht ja etwas Sonne dazu und evtl. auch das spezielle Auto) nachsehen.

Das war übrigens noch mit der 0.123.0 oder 0.123.1 (aber definitiv nicht älter und aus dem Release-Zweig), habe gerade auf 0.123.2 aktualisiert.

VolkerK62, 2023-12-30:
>außerdem ist 1,1 kW +0,5 kW nicht 1,5 kW).

doch ;-)
1,06 (gerundet angezeigt 1,1) + 0,46 (gerundet angezeigt 0,5) = 1,52 (gerundet angezeigt 1,5)

g-mocken, 2023-12-30:
Ja, dass da hinter dem Fehler von 0,1kW ein Rundungsphänomen dahintersteckt, ist zu erwarten. Vielleicht besser alles in ganzen Watt anzeigen, dann ist auch der Fehler höchstens ein Watt.
Aber das eigentlich auffällige sind ja die OUT-IN = 340W, also ein deutlich größerer Fehler, was natürlich von "geratenen" 1840W = 230V * 8A * 1ph kommt, während der Gesamtverbrauch in Wahrheit derweil aber natürlich nur 1,5 oder 1,6 kW war, und das Haus dabei natürlich nicht 0W.
Warum die ZOE unter der Wallbox-Vorgabe bleibt, weiß ich nicht, aber diese ja ein analoges PWM-Signal mit (bei dieser Wallbox) nur ganzen 1A-Schritten ist, kann es natürlich sein, dass sie grundsätzlich um 1A daneben (hier: darunter) liegt. Das wären dann 1610W, was aber noch nicht ganz als Erklärung reicht, da mindestens 200W noch ins Haus gingen.

premultiply, 2023-12-30:
W vs. VA

VolkerK62, 2023-12-31:
>Warum die ZOE unter der Wallbox-Vorgabe bleibt, weiß ich nicht, 

Meine WB hat einen Zähler (und ein Display). Mein Auto lädt (egal ob mit oder ohne evcc) nie mit der Stromvorgabe.
Bei 16A Vorgabe zeigt die WB zwischen 14,8A und 15,5A an. Am Anfang dachte ich, die sei kaputt. Da kam der WB-Hersteller mit seinem Tesla vorbei und hat mir gezeigt, dass auch 16A möglich sind.
Insofern würde ich sagen: Völlig normal.

g-mocken, 2023-12-31:
Scheinleistung versus Wirkleistung, stimmt, das ist so ein fehlender Faktor, gerade bei Renault Zoe. 

Leider ist heute keinerlei Sonne, um es nochmal zu reproduzieren.