evcc-io/evcc #11356: [nightly] runtime Error tasmota

mdkeil, 2023-12-31:
### Describe the bug

aktuelle nightly hat wohl ein Problem mit tasmota meters.

### Steps to reproduce

yaml mit konfigurierten tasmota-meter

### Configuration details

```yaml
network:
  schema: http
  host: evcc
  port: 7070
interval: 10s

influx:
  url: http://192.168.1.2:8086
  database: evcc
  user: *****
  password: *****

mqtt:
  broker: 192.168.1.2:1883
  topic: evcc
  user: *****
  password: *****

log: debug
levels:
  #site: trace
  #salia: trace
  lp-2: info

vehicles:
  - name: my_peugeot
    type: template
    template: flobz
    title: e-208
    url: http://192.168.1.2:5050
    vin: *****
    capacity: 46
    #mode: pv
    #targetSoC: 80
  - name: my_kia
    type: custom
    title: Niro-EV
    soc:
      source: mqtt
      topic: nodered/evnotify/all
      jq: .soc_display
    status:
      source: mqtt
      topic: nodered/evnotify/status
    range:
      source: mqtt
      topic: nodered/evnotify/all
      jq: .range
    #akey: c53705
    #token: *****
    capacity: 64.8
  - name: my_dummy
    type: template
    template: offline
    title: evDummy

chargers:
  - name: my_charger
    type: template
[... 147 lines cut ...]
github.com/evcc-io/evcc/meter/tasmota.(*Connection).Voltages(0xc0015a0a20?)
	github.com/evcc-io/evcc/meter/tasmota/connection.go:230 +0x1a
github.com/evcc-io/evcc/charger.(*Tasmota).Voltages(0xc001855400?)
	github.com/evcc-io/evcc/charger/tasmota.go:92 +0x16
github.com/evcc-io/evcc/core.(*Loadpoint).updateChargeVoltages(0xc000db8000)
	github.com/evcc-io/evcc/core/loadpoint.go:1319 +0x7d
github.com/evcc-io/evcc/core.(*Loadpoint).Update(0xc000db8000, 0x0?, 0x1, 0x0?, 0x0?, 0x0, 0xc0019851b0, 0x0)
	github.com/evcc-io/evcc/core/loadpoint.go:1516 +0x66
github.com/evcc-io/evcc/core.(*Site).update(0xc000ae6000, {0x32708b0?, 0xc000db8000})
	github.com/evcc-io/evcc/core/site.go:794 +0x5ca
github.com/evcc-io/evcc/core.(*Site).Run(0xc000ae6000, 0xc000957aa0, 0x2540be400)
	github.com/evcc-io/evcc/core/site.go:904 +0x2e5
github.com/evcc-io/evcc/cmd.runRoot.func4()
	github.com/evcc-io/evcc/cmd/root.go:281 +0x25
created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
	github.com/evcc-io/evcc/cmd/root.go:280 +0x19b3
```

### What type of operating system are you running?

Docker container

### Version

0.123.2 (cf597421)

VolkerK62, 2023-12-31:
Da gab es eine Änderung  https://github.com/evcc-io/evcc/pull/11347

andig, 2023-12-31:
Danke für den Hinweis. Fix kommt sofort.

andig, 2023-12-31:
@thierolm hier hat mich die Logik überrascht. Es werden die Phasenwerte verwendet obwohl die Konfiguration 1p ist. Das dürfte eigentlich nicht der Fall sein. Ich würde das deshalb lieber mit einem Decorator ändern. Hättest Du Einwände?

thierolm, 2023-12-31:
Nein keine Einwände, aber vergiss Silvester nicht 😏

ulrichSchreiner, 2024-01-01:
hi,

ich habe den gleichen fehler, aber erst mit 0.123.3, die 0.123.2 ging noch. meine konfig ist zur obigen quasi vergleichbar:

~~~yaml
chargers:
- name: tasmota02
  type: template
  template: tasmota
  host: tasmota02
  channel: 1
- name: tasmota01
  type: template
  template: tasmota
  host: tasmota01
  channel: 1
  icon: bike
~~~

und es passiert exakt der gleiche panic an der gleichen zeile. wenn ich auf die 0.123.2 downgrade gehts wieder.

mdkeil, 2024-01-01:
Kann ich bestätigen.. #11366 hat leider nicht zum Erfolg geführt, Fehler bleibt der Gleiche.. habe es leider nicht früher geschafft, dass in der letzen nightly zu testen.

andig, 2024-01-01:
Jau. Ich habs nur für den Zähler gelöst, aber nicht für den charger. Seufz.

closed 2024-01-01 by andig

mdkeil, 2024-01-02:
Nightly bootet wieder ohne Probleme. Danke.