evcc-io/evcc #11360: Move restart button form help wanted to menu top level or in system submenue

mmierke, 2023-12-31:
### Move restart button form help wanted to menu top level

Until now I always restarted my docker containe after a change to the configuration. I have to do that very often, because I just started with evcc and I change a lot nowerdays.

However, yesterday I opened the **Need help?** menu entry and finally found a way to restart evcc from the GUI.

From a UX experience this is not self explaining. It would be much better to elevate that choice to the to level of the menu or under a system submenue.

andig, 2023-12-31:
Wir wollten das egtl. bewusst nicht so prominent machen da diese Restarts oft ein Aberglaube sind. Mein persönliches evcc läuft einfach über Wochen durch.

/cc @naltatis

naltatis, 2023-12-31:
> I have to do that very often, because I just started with evcc and I change a lot nowerdays.

Yes, what @andig says :D

If you feel the need to restart regularly, we should discuss the reasons and maybe fix them.

I understand that, especially in the beginning, it might be necessary to get the initial config right. That's something we are actively working on with config UI.

closed 2023-12-31 by naltatis