evcc-io/evcc #11371: Ladeverlauf hält sich nicht an Ladeplanung bei mehreren Unterbrechungen

cvg-de, 2023-12-31:
### Describe the bug

Der tatsächliche Ladeverlauf erfolgt nicht nach der Ladeplanung. Dies ist aufgetreten, als die Planung (korrekterweise) die Gesamtladung über Nacht in drei Zeitscheiben aufgeteilt hat:
Eine Stunde von 21-22 Uhr
Eine Stunde von 01-02 Uhr
Rest ab 03 Uhr
Die tatsächliche Ladung ist zunächst eine Stunde ab 21 Uhr erfolgt, und dann der gesamte Rest ab 01 Uhr.
Screenshots von Ladeplanung und tatsächlichem Verbrauch laut Tibber-App anbei.
[image]
[image]
[image]
[image]

### Steps to reproduce

1. Warten bis das nächste Mal der Strompreis in der Nacht so schwankend ist, und das Auto geladen werden muss
2. Planung eingeben und aktivieren
3. Ergebnis am nächsten Tag anschauen
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: <xyz>

interval: 10s # control cycle interval

meters:
- type: template
  template: lg-ess-home-8-10
  usage: grid
  host: LG-PV-Anlage
  password: ganzgeheim
  name: grid1
- type: template
  template: lg-ess-home-8-10
  usage: pv
  host: LG-PV-Anlage
  password: ganzgeheim
  name: pv2
- type: template
  template: lg-ess-home-8-10
  usage: battery
  host: LG-PV-Anlage
  password: ganzgeheim
  name: battery3

chargers:
- type: template
  template: hardybarth-ecb1
  host: ecb1-wallbox
  name: wallbox5
- type: hardybarth-ecb1
  uri: http://ecb1-wallbox.fritz.box
  chargecontrol: 2
  meter: 2
  name: wallbox6

vehicles:
- type: template
  template: audi
  title: Q4 etron
  user: xyz
  password: ganzgeheim
  vin: WAUZZZ1234567890
  capacity: 77
  name: ev4

loadpoints:
- title: Garage Spiralkabel
  charger: wallbox5
  mode: minpv
  phases: 3
  guardduration: 3m
  mincurrent: 6
  maxcurrent: 16
- title: Garage Steckdose
  charger: wallbox6
  mode: minpv
  phases: 3
  mincurrent: 6
  maxcurrent: 16

site:
  title: vongrone-evcc
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
  residualpower: 100

tariffs:
  currency: EUR
  grid:
    type: tibber
    token: "<behalteichfuerrmich>"
#    type: fixed
#    price: 0.3915

  feedin:
    type: fixed
    price: 0.0747
```

### Log details

```text
Dec 29 18:27:07 smarthome evcc[17731]: [lp-1  ] INFO 2023/12/29 18:27:07 car connected
Dec 29 18:27:17 smarthome evcc[17731]: [lp-1  ] INFO 2023/12/29 18:27:17 vehicle updated: unknown -> Q4 etron
Dec 29 18:27:26 smarthome evcc[17731]: [lp-1  ] INFO 2023/12/29 18:27:26 start charging ->
Dec 29 18:30:47 smarthome evcc[17731]: [lp-1  ] INFO 2023/12/29 18:30:47 stop charging <-
Dec 29 21:00:27 smarthome evcc[17731]: [lp-1  ] INFO 2023/12/29 21:00:27 start charging ->
Dec 29 22:00:27 smarthome evcc[17731]: [lp-1  ] INFO 2023/12/29 22:00:26 stop charging <-
Dec 30 01:00:27 smarthome evcc[17731]: [lp-1  ] INFO 2023/12/30 01:00:27 start charging ->
Dec 30 03:23:27 smarthome evcc[17731]: [lp-1  ] INFO 2023/12/30 03:23:27 stop charging <-
```

### What type of operating system are you running?

Linux

### Version

0.123.2

andig, 2023-12-31:
Es braucht bitte ein vollständiges Debuglog.

closed 2023-12-31 by andig

cvg-de, 2023-12-31:
Ok, ich werde beim nächsten Auftreten so einer Konstellation den Log-Level auf Debug stellen und dann hier ein Update senden.