evcc-io/evcc #11372: Weihnachts Update / Batterie Steuerung

puzich, 2023-12-31:
### Describe the bug

Im Weihnachts Update wird beschrieben, dass es ein Feature gibt, dass die Hausbatterie nicht mehr entlädt, wenn man das Auto auf "Schnell" laden möchte. Beim einschalten der Funktion (nach dem Klick auf den Batteriestandt) erhalte ich einen Panic von EVCC.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
#log: info
#levels:
#  bmw: debug
#  cache: error
##  modbus: trace
##log.easee: debug

# unique installation id
plant: *****
sponsortoken: *****
#interval: 30s # control cycle interval
#interval: 10s # control cycle interval
interval: 15s # control cycle interval

site:
  title: Home
  meters:
    grid: my_grid
    pv: my_pv
    battery: my_battery
  ##prioritySoc: 50
#  residualPower: 350
  residualPower: 100
  ##bufferSoc: 80
  ##batteryDischargeControl: true

meters:
  - name: my_grid
    type: template
    template: sofarsolar-g3
    usage: grid
    modbus: rs485serial
    id: 1
    device: /dev/ttyUSB0
    baudrate: 9600 # Pruefe Geraeteeinstellungen, typische Werte sind 9600, 19200, 38400, 57600, 115200
    comset: "8N1" # Kommunikationsparameter fuer den Adapter
    delay: 1s # Optional
  - name: my_pv
    type: template
    template: sofarsolar-g3
    usage: pv
    modbus: rs485serial
    id: 1
    device: /dev/ttyUSB0
    baudrate: 9600 # Pruefe Geraeteeinstellungen, typische Werte sind 9600, 19200, 38400, 57600, 115200
    comset: "8N1" # Kommunikationsparameter fuer den Adapter
    delay: 1s # Optional
  - name: my_battery
    type: template
[... 170 lines cut ...]
Dec 31 16:06:42 evcc evcc[14973]: #011net/http/server.go:3086 +0x4c0
Dec 31 16:06:43 evcc evcc[14973]: [site  ] DEBUG 2023/12/31 16:06:43 battery soc: 11%
Dec 31 16:06:43 evcc evcc[14973]: [site  ] DEBUG 2023/12/31 16:06:43 battery power: 0W
Dec 31 16:06:44 evcc evcc[14973]: [site  ] DEBUG 2023/12/31 16:06:44 grid power: 1210W
Dec 31 16:06:47 evcc evcc[14973]: [site  ] DEBUG 2023/12/31 16:06:47 grid powers: [320 350 340]W
Dec 31 16:06:48 evcc evcc[14973]: [site  ] DEBUG 2023/12/31 16:06:48 set battery discharge control: false
Dec 31 16:06:50 evcc evcc[14973]: [site  ] DEBUG 2023/12/31 16:06:50 grid currents: [1.46 1.49 1.42]A
Dec 31 16:06:51 evcc evcc[14973]: [site  ] DEBUG 2023/12/31 16:06:51 site power: 1310W
Dec 31 16:06:51 evcc evcc[14973]: [lp-2  ] ERROR 2023/12/31 16:06:51 vehicle odometer: asleep
Dec 31 16:06:51 evcc evcc[14973]: [lp-2  ] DEBUG 2023/12/31 16:06:51 charge currents: [0 0 0]A
Dec 31 16:06:51 evcc evcc[14973]: [lp-2  ] DEBUG 2023/12/31 16:06:51 charge total import: 2948.547kWh
Dec 31 16:06:51 evcc evcc[14973]: [lp-2  ] DEBUG 2023/12/31 16:06:51 charger status: A
Dec 31 16:06:51 evcc evcc[14973]: [lp-2  ] INFO 2023/12/31 16:06:51 car disconnected
Dec 31 16:06:51 evcc evcc[14973]: [lp-2  ] DEBUG 2023/12/31 16:06:51 set session soc limit: 0
Dec 31 16:06:51 evcc evcc[14973]: [lp-2  ] DEBUG 2023/12/31 16:06:51 set session energy limit: 0
Dec 31 16:06:51 evcc evcc[14973]: [lp-2  ] DEBUG 2023/12/31 16:06:51 set tesla_ecto1 plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
```

### What type of operating system are you running?

Linux

### Version

0.123.3

andig, 2023-12-31:
@puzich could you give remote access for info@evcc.io (exposed inverter port or SSH)? I'm unable to repro this locally.

puzich, 2023-12-31:
EMail mit Zugangsdaten ist raus.

closed 2023-12-31 by andig