evcc-io/evcc #11373: cannot create meter 'battery3'

leetillu, 2023-12-31:
### Describe the bug

After updating to 0.123.3 the Addon won't startup anymore.

### Steps to reproduce

1. Update
2. Restart
3. Fail

### Configuration details

```yaml
# open evcc at http://evcc.local:7070

network:

  schema: http

  host: evcc.local # .local suffix announces the hostname on MDNS

  port: 7070

log: debug

levels:

  cache: error

# unique installation id

plant: xxx

interval: 10s # control cycle interval

sponsortoken: xxx

# sponsors can set telemetry: true to enable anonymous data aggregation

# see https://github.com/evcc-io/evcc/discussions/4554

telemetry: false

meters:

- type: template

  template: solax 

  id: 1  

  host: 192.168.178.73  

  port: 502  

  usage: grid  

  modbus: tcpip  

  name: grid1

- type: template

  template: solax 

  id: 1  

  host: 192.168.178.73  

  port: 502  

  usage: pv  

  modbus: tcpip  

  name: pv2

- type: template

  template: solax 

  id: 1  

  host: 192.168.178.73  

  port: 502  

  usage: battery  

  modbus: tcpip  

  name: battery3

chargers:

- type: template

  template: go-e-v3 

  host: 192.168.178.70  

  name: wallbox5

- type: template

  template: go-e-v3 

  host: 192.168.178.68  

  name: wallbox6

vehicles:

- type: template

  template: kia 

  title: Kia Niro  

  user: xxx  

  password: xxx  

  vin: xxx 

  capacity: 64  

  phases: 3  

  icon: car  

  cache: 15m  

  language: de  

  mode: now  

  minCurrent: 6  

  maxCurrent: 16   

  name: ev4

- type: template

  template: kia 

  title: Kia ev6  

  user: xxx  

  password: xxx 

  vin: xxx

  capacity: 77  

  phases: 3  

  icon: car  

  cache: 15m  

  language: de  

  mode: pv  

  minCurrent: 6  

  maxCurrent: 16  

  name: ev5

loadpoints:

- title: Wallbox 2

  charger: wallbox5

  mode: now

  phases: 0

  mincurrent: 6

  maxcurrent: 16

- title: Wallbox 1

  charger: wallbox6

  mode: pv

  phases: 0

  mincurrent: 6

  maxcurrent: 16

site:

  title: Home

  meters:

    grid: grid1

    pv:

    - pv2

    battery:

    - battery3

  residualPower: 100

tariffs:

  currency: EUR # (default EUR)

  grid:

    # static grid price

    type: fixed

    price: 0.40 # [currency]/kWh

  feedin:

    # rate for feeding excess (pv) energy to the grid

    type: fixed

    price: 0.082 # [currency]/kWh

# mqtt message broker
mqtt:
  broker: localhost:1883
  topic: evcc
  user: mqtt
  password: xxx
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/12/31 17:02:33 evcc 0.123.3
[main  ] INFO 2023/12/31 17:02:33 using config file: /config/evcc.yaml
[main  ] INFO 2023/12/31 17:02:33 starting ui and api at :7070
[db    ] INFO 2023/12/31 17:02:34 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2023/12/31 17:02:34 connecting evcc-1072325781 at tcp://localhost:1883
[mqtt  ] DEBUG 2023/12/31 17:02:34 tcp://localhost:1883 connected
[main  ] FATAL 2023/12/31 17:02:34 cannot create meter 'battery3': cannot create meter 'template': cannot create meter 'custom': battery mode: 1 error(s) decoding:

* error decoding 'Timeout': time: invalid duration "<no value>"
[main  ] FATAL 2023/12/31 17:02:34 will attempt restart in: 5m0s
```

### What type of operating system are you running?

Linux

### Version

0.123.3

leetillu, 2023-12-31:
After restoring to previous version startup is without issues.

```
[lp-1  ] DEBUG 2023/12/31 17:11:41 charger status: A
[site  ] DEBUG 2023/12/31 17:11:50 ----
[lp-1  ] DEBUG 2023/12/31 17:11:50 charge power: 0W
[lp-2  ] DEBUG 2023/12/31 17:11:50 charge power: 0W
[site  ] DEBUG 2023/12/31 17:11:50 pv power: 0W
[site  ] DEBUG 2023/12/31 17:11:50 battery soc: 10%
[site  ] DEBUG 2023/12/31 17:11:50 battery power: 0W
[site  ] DEBUG 2023/12/31 17:11:51 grid power: 378W
[site  ] DEBUG 2023/12/31 17:11:51 site power: 478W
[lp-2  ] DEBUG 2023/12/31 17:11:51 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2023/12/31 17:11:51 charge total import: 645.072kWh
[lp-2  ] DEBUG 2023/12/31 17:11:51 charger status: A
[site  ] DEBUG 2023/12/31 17:12:00 ----
[lp-1  ] DEBUG 2023/12/31 17:12:00 charge power: 0W
[lp-2  ] DEBUG 2023/12/31 17:12:00 charge power: 0W
[site  ] DEBUG 2023/12/31 17:12:01 pv power: 0W
[site  ] DEBUG 2023/12/31 17:12:01 battery soc: 10%
[site  ] DEBUG 2023/12/31 17:12:01 battery power: 0W
[site  ] DEBUG 2023/12/31 17:12:01 grid power: 368W
[site  ] DEBUG 2023/12/31 17:12:01 site power: 468W
[lp-1  ] DEBUG 2023/12/31 17:12:01 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/12/31 17:12:01 charge total import: 1548.525kWh
[lp-1  ] DEBUG 2023/12/31 17:12:01 charger status: A
[site  ] DEBUG 2023/12/31 17:12:10 ----
[lp-1  ] DEBUG 2023/12/31 17:12:10 charge power: 0W
[lp-2  ] DEBUG 2023/12/31 17:12:10 charge power: 0W
[site  ] DEBUG 2023/12/31 17:12:10 pv power: 0W
[site  ] DEBUG 2023/12/31 17:12:10 battery soc: 10%
[site  ] DEBUG 2023/12/31 17:12:10 battery power: 0W
[site  ] DEBUG 2023/12/31 17:12:11 grid power: 373W
[site  ] DEBUG 2023/12/31 17:12:11 site power: 473W
[lp-2  ] DEBUG 2023/12/31 17:12:11 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2023/12/31 17:12:11 charge total import: 645.072kWh
[lp-2  ] DEBUG 2023/12/31 17:12:11 charger status: A
[site  ] DEBUG 2023/12/31 17:12:20 ----
```

andig, 2023-12-31:
Steht alles da:

> * error decoding 'Timeout': time: invalid duration "<no value>"