evcc-io/evcc #11377: Addon startet nach Update auf 0.123.3 nicht mehr

Bulli714, 2023-12-31:
### Describe the bug

Wie im Titel schon geschrieben habe ich das Update auf Version 0.123.3 durchgeführt und kann nun das Addon nicht mehr starten. 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
.
```

### Log details

```text
goroutine 131 [running]:
encoding/binary.bigEndian.Uint32(...)
	encoding/binary/binary.go:157
github.com/evcc-io/evcc/charger.(*Hesotec).getPhaseValues(0x1db9e40?, 0x56a0?, 0x1, 0x3ff0000000000000)
	github.com/evcc-io/evcc/charger/hesotec.go:202 +0xf8
github.com/evcc-io/evcc/charger.(*Hesotec).Voltages(0x4000891fc0?)
	github.com/evcc-io/evcc/charger/hesotec.go:190 +0x28
github.com/evcc-io/evcc/core.(*Loadpoint).updateChargeVoltages(0x40007c1c00)
	github.com/evcc-io/evcc/core/loadpoint.go:1319 +0x60
github.com/evcc-io/evcc/core.(*Loadpoint).Update(0x40007c1c00, 0x0?, 0x0, 0x0?, 0x0?, 0x0, 0x0, 0x0)
	github.com/evcc-io/evcc/core/loadpoint.go:1516 +0x4c
github.com/evcc-io/evcc/core.(*Site).update(0x4000bbc000, {0x2c68970?, 0x40007c1c00})
	github.com/evcc-io/evcc/core/site.go:794 +0x42c
github.com/evcc-io/evcc/core.(*Site).Run(0x4000bbc000, 0x40014524e0, 0x6fc23ac00)
	github.com/evcc-io/evcc/core/site.go:899 +0x1a4
github.com/evcc-io/evcc/cmd.runRoot.func4()
	github.com/evcc-io/evcc/cmd/root.go:281 +0x2c
created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
	github.com/evcc-io/evcc/cmd/root.go:280 +0x165c
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.123.3

naltatis, 2023-12-31:
Füg mal bitte deine Config an.

Bulli714, 2023-12-31:
Hier meine Config;

network:
  schema: http
  host: 192.168.0.4
  port: 7070

interval: 30s # control cycle interval

sponsortoken: xxx

log: debug
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: debug
  db: debug
  
modbusproxy:
  - port: 5021
    uri: 192.168.0.118:502

meters:
  - name: pv
    type: template
    template: sungrow-hybrid
    usage: pv
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.0.118 # Hostname
    port: 502 # Port 
  - name: grid
    type: template
    template: sungrow-hybrid
    usage: grid
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.0.118 # Hostname
    port: 502 # Port 
  - name: battery
    type: template
    template: sungrow-hybrid
    usage: battery
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.0.118 # Hostname  

chargers:
  - name: r10
    type: template
    template: hesotec      
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.0.10 # Hostname
    port: 502 # Port      

# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for documentation see https://docs.evcc.io/docs/devices/vehicles
vehicles:
  - type: template
    name: Enyaq
    template: enyaq
    title: Enyaq
    capacity: 77
    phases: 3
    cache: 15m
    timeout: 15s
    minCurrent: 6
    maxCurrent: 16
    vin: xxx
    user: xxx
    password: xxx
    mode: pv # enable PV-charging when vehicle is identified
    # minSoc: 20 # immediately charge to 0% regardless of mode unless "off" (disabled)
    # targetSoc: 80 # limit charge to 90%
    icon: car

site:
  title: Ross
  meters:
    grid: grid
    pv:
    - pv
    battery: battery
  residualPower: 100
    
# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
  - title: Einfahrt # display name for UI
    charger: r10 # charger
    mode: pv # set default charge mode, use "off" to disable by default if charger is publicly available
    vehicle: Enyaq # set default vehicle (disables vehicle detection)
    phases: 3 # electrical connection (normal charger: default 3 for 3 phase, 1p3p charger: 0 for "auto" or 1/3 for fixed phases)
    minCurrent: 6 # minimum charge current (default 6A)
    maxCurrent: 16 # maximum charge current (default 16A)

mdkeil, 2023-12-31:
Das gleiche Verhalten wie hier.. Fehlerbehebung sollte ähnlich sein.
#11356