evcc-io/evcc #11380: Solax template missing timeout

premultiply, 2023-12-31:
Es fehlt im solax template unter params `timeout`.. wird im Abschnitt battery_control benötigt..

_Originally posted by @mdkeil in https://github.com/evcc-io/evcc/discussions/11376#discussioncomment-7984772_

Stepsilein, 2024-01-01:
... uuuund... was zm Teufel muss der Nichtnerd nun wo in der evcc.yaml ändern, damit das Gsnz wieder läuft ???
Leute, wenn weiter so im Elfenbeintum auf Nerdniveau agiert wird, erreicht ihr nie Endanwender wirklich.
Es wäre ja sooo einfach mal zu erklären, was wo wie geändert werden muss... um ggf. die schlechtere Lösung des Batterie leersaugen Problems zu lösen, oder.
Es nervt total.
Kommt was, sonst gehe ich endgültig auf Backup und das war's.
Gibt da ganz andere Lösung betr. Batterie blocken.
Fakt ist:
Änderung MUSS mit Erläuterung der nötigen Änderung in der evcc.yaml kommen ... oder nur ihr kommt damit klar.
Kann doch nicht so schwer sein da wenigstens How to mit zu liefern.
Bin es leid aus kryptischen Fehlerlogs ständig händisch herum zu basteln hinterher.
Besser machen und nun nicht beschweren.
Wir deutlich muss man den noch werden :-(

andig, 2024-01-01:
Closed in https://github.com/evcc-io/evcc/commit/23284f79a9781643ff5c6270ba3355c99f32c266.

closed 2024-01-01 by andig