evcc-io/evcc #11381: Kostal Modbus

N-Bruder, 2023-12-31:
### Describe the bug

Für die aktive Batteriesteuerung mass der Kostal Wechselrichter auf Modbus Big-Endian gestellt werden. Allerdings muss man für die Kommunikation zwischen Wechselrichter und Smart energy Meter die Modbus auf Standard belassen. 
Es funktioniert sonst auch alles mit dem Standard Modbus  in EVCC nur die aktive Batteriesteuerung

### Steps to reproduce

.

### Configuration details

```yaml
.
```

### Log details

```text
[main  ] INFO 2023/12/31 22:21:41 evcc 0.123.3
[main  ] INFO 2023/12/31 22:21:41 using config file: /etc/evcc.yaml
[main  ] INFO 2023/12/31 22:21:41 starting ui and api at :7070
[db    ] INFO 2023/12/31 22:21:41 using sqlite database: /home/bruder/.evcc/evcc.db
[lp-1  ] WARN 2023/12/31 22:21:46 locking phase config to 1p for switchable charger
[lp-1  ] DEBUG 2023/12/31 22:21:46 charge total import: 5199.422kWh
[lp-2  ] DEBUG 2023/12/31 22:21:46 charge total import: 0.000kWh
[site  ] WARN 2023/12/31 22:21:46 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site  ] INFO 2023/12/31 22:21:46 site config:
[site  ] INFO 2023/12/31 22:21:46   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/12/31 22:21:46     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/12/31 22:21:46     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/12/31 22:21:46     pv 2:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/12/31 22:21:46     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/12/31 22:21:46   vehicles:
[site  ] INFO 2023/12/31 22:21:46     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[lp-1  ] INFO 2023/12/31 22:21:46 loadpoint 1:
[lp-1  ] INFO 2023/12/31 22:21:46   mode:        pv
[lp-1  ] INFO 2023/12/31 22:21:46   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/12/31 22:21:46   meters:      charge ✓
[lp-1  ] INFO 2023/12/31 22:21:46     charge:    power ✓ energy ✓ currents ✓
[lp-2  ] INFO 2023/12/31 22:21:46 loadpoint 2:
[lp-2  ] INFO 2023/12/31 22:21:46   mode:        pv
[lp-2  ] INFO 2023/12/31 22:21:46   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-2  ] INFO 2023/12/31 22:21:46   meters:      charge ✓
[lp-2  ] INFO 2023/12/31 22:21:46     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/12/31 22:21:46 phase timer inactive
[lp-1  ] DEBUG 2023/12/31 22:21:46 pv timer inactive
[lp-1  ] DEBUG 2023/12/31 22:21:46 guard timer inactive
[lp-1  ] INFO 2023/12/31 22:21:46 vehicle updated: unknown -> Zoe
[lp-2  ] DEBUG 2023/12/31 22:21:46 phase timer inactive
[lp-2  ] DEBUG 2023/12/31 22:21:46 pv timer inactive
[lp-2  ] DEBUG 2023/12/31 22:21:46 guard timer inactive
[lp-1  ] INFO 2023/12/31 22:21:46 vehicle updated: Zoe -> unknown
[lp-2  ] INFO 2023/12/31 22:21:46 vehicle updated: unknown -> Zoe
[main  ] FATAL 2023/12/31 22:21:46 could not open port- check that evcc is not already running (listen tcp :7070: bind: address already in use)
```

### What type of operating system are you running?

Linux

### Version

0.123.3

StevieC121176, 2024-01-01:
Ich würde hier eher den Fehler bei Kostal suchen. Ich habe BigEndian eingestellt (was bei sunspec meiner Meinung nach standard ist). Ja, in der Kostal Weboberfläche werden dann wirre Werte angezeigt, aber die nutze ich eh nicht. Weil
a: alle relevanten Werte in EVCC dargestellt werden und
b: ich 2 Kostal WR habe und dann stimmen die Werte bei den einzelnen WR sowieso nicht mehr (und seit Jahren wird das nicht behoben)

andig, 2024-01-01:
Laut https://docs.evcc.io/docs/devices/meters#plenticore-hybrid muss der WR *immer* und unabhängig von der Batteriesteuerung auf BigEndian stehen. Insofern keine Überraschung und hat nichts mit Batteriesteuerung zu tun?