evcc-io/evcc #11386: targetsoc verhindert Fahrzeugerkennung

Horgul123, 2024-01-01:
### Describe the bug

Wenn beim vehicle der targetsoc in der evcc.yaml steht startet EVCC ohne Fehler, das Fahrzeug wird aber nicht mehr erkannt (V0.123.3, aber schon seit 2 oder 3 Versionen vorher aufgefallen, nur kannte ich da die Lösung nicht)

### Steps to reproduce

in der evcc.yaml beim vehicle einen targetsoc definieren und dann ein Fahrzeug anschließen.
Beim Start von EVCC sollte da ein Fehler angezeigt werden.

### Configuration details

```yaml
(fehlerhafte Config nicht mehr vorhanden)
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

### Version

0.123.3

VolkerK62, 2024-01-01:
Ich würde vorschlagen, den Titel zu ändern. Es geht ja um einen fehlenden HInweis/Warnung.

andig, 2024-01-01:
Das ist ein guter Punkt. Bei Konfigurationsfehlern sollten wir stattdessen mit entsprechender Fehlermeldung abbrechen.

Horgul123, 2024-01-01:
@VolkerK62 : Danke für den Hinweis, hab den Titel ergänzt

@andig: Finde ich auch :)

andig, 2024-01-01:
Ich kann das nicht nachvollziehen. Diese Config ist ungültig:

```yaml
vehicles:
  - name: vehicle_1
    title: blauer e-Golf
    type: ...
    onidentify:
      targetsoc: 17
```

Gibt völlig korrekt einen Fehler und bricht ab:

```
[main  ] FATAL 2024/01/01 12:46:18 cannot create vehicle 'vehicle_1': cannot create vehicle 'vehicle_1': cannot create vehicle 'custom': 1 error(s) decoding:

* 'onIdentify' has invalid keys: targetsoc
```

Ich bräuchte also bitte die konkrete Config bei der evcc dennoch startet. Bis dahin erstmal zu.

closed 2024-01-01 by andig

VolkerK62, 2024-01-01:
Das ist einfach ;-)
Hängt evtl von template oder nativ ab??

```
vehicles:
- name: auto
  type: template
  template: hyundai
  title: (T)Raumschiff
  icon: car
  user: ****
  password: ****
  language: de
  capacity: 72.6
  phases: 3
  mode: pv
  maxcurrent: 20
  minsoc: 30
```

```
pi@raspberrypi:~ $ sudo journalctl -fau evcc
-- Journal begins at Thu 2023-12-28 03:21:37 CET. --
Jan 01 13:03:06 raspberrypi systemd[1]: Started evcc.
Jan 01 13:03:06 raspberrypi evcc[8298]: [main  ] INFO 2024/01/01 13:03:06 evcc 0.123.3 (fd937c64)
Jan 01 13:03:06 raspberrypi evcc[8298]: [main  ] INFO 2024/01/01 13:03:06 using config file: /etc/evcc.yaml
Jan 01 13:03:06 raspberrypi evcc[8298]: [main  ] INFO 2024/01/01 13:03:06 starting ui and api at :7070
Jan 01 13:03:07 raspberrypi evcc[8298]: [db    ] INFO 2024/01/01 13:03:07 using sqlite database: /var/lib/evcc/evcc.db
Jan 01 13:03:07 raspberrypi evcc[8298]: [mqtt  ] INFO 2024/01/01 13:03:07 connecting evcc-103329787 at tcp://localhost:1883
Jan 01 13:03:07 raspberrypi evcc[8298]: [mqtt  ] DEBUG 2024/01/01 13:03:07 tcp://localhost:1883 connected
Jan 01 13:03:07 raspberrypi evcc[8298]: [proxy-5200] DEBUG 2024/01/01 13:03:07 modbus proxy for 192.168.178.26:502 listening at :5200
Jan 01 13:03:07 raspberrypi evcc[8298]: [main  ] ERROR 2024/01/01 13:03:07 creating vehicle auto failed: cannot create vehicle 'template': invalid key: minsoc
Jan 01 13:03:07 raspberrypi evcc[8298]: [lp-1  ] WARN 2024/01/01 13:03:07 poll mode '{always 5h0m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
Jan 01 13:03:07 raspberrypi evcc[8298]: [lp-1  ] DEBUG 2024/01/01 13:03:07 charge total import: 1373.900kWh
Jan 01 13:03:07 raspberrypi evcc[8298]: [lp-2  ] DEBUG 2024/01/01 13:03:07 charge total import: 1.278kWh
Jan 01 13:03:07 raspberrypi evcc[8298]: [lp-3  ] DEBUG 2024/01/01 13:03:07 charge total import: 0.000kWh
Jan 01 13:03:07 raspberrypi evcc[8298]: [lp-4  ] DEBUG 2024/01/01 13:03:07 charge total import: 938.963kWh
Jan 01 13:03:07 raspberrypi evcc[8298]: [lp-5  ] DEBUG 2024/01/01 13:03:07 charge total import: 27.857kWh
Jan 01 13:03:07 raspberrypi evcc[8298]: [lp-6  ] DEBUG 2024/01/01 13:03:07 charge total import: 48.733kWh
```

Horgul123, 2024-01-01:
So startet EVCC bei mir (Daten teilweise anonymisiert ;) )

`vehicles:
- type: template
  template: enyaq
  title: Olis Eny
  user: oliver@XXXXX.de
  password: XXXXX
  vin: ABC
  targetsoc: 85
  capacity: 80
  name: ev3
`

andig, 2024-01-01:
Wunderbar, gelöst!

Horgul123, 2024-01-01:
Jetzt aus Neugier: Wieso wurde bei Dir ei Fehler geworfen und bei uns nicht?
Wegen dem "onidentify:" ?

andig, 2024-01-01:
Templates vs „nackte“ Config haben sich unterschiedlich verhalten.