evcc-io/evcc #11388: Smart cost limit (in context: smart charging) can no longer be set in web GUI since latest version v0.123.3

steinthomas, 2024-01-01:
### Describe the bug

Since the latest version v0.123.3, the smart cost limit (ct/kWh) can no longer be set via the web GUI for smart charging in the context of dynamic electricity tariffs (e.g. Tibber). It has already been removed in the YAML Config because it will be configured via the web GUI.

Previously, the Smart Cost Limit could be set via the current electricity price by clicking on it. This is also displayed in the new version, but is no longer a link (to the configuration page for Smart charging). Everything else works fine.

### Steps to reproduce

1. Open Web-GUI
2. Try to configure the smart cost limit by clicking on the current electricity price

### Configuration details

```yaml
smartCostLimit not included/configured in yaml file
```

### Log details

```text
service cannot be stopped, because its running in container...

/app # evcc --log debug
[main  ] INFO 2024/01/01 11:23:52 evcc 0.123.3
[main  ] INFO 2024/01/01 11:23:52 using config file: /etc/evcc.yaml
[main  ] INFO 2024/01/01 11:23:52 starting ui and api at :7070
[mqtt  ] INFO 2024/01/01 11:23:52 connecting evcc at tcp://192.168.1.12:1883
[mqtt  ] DEBUG 2024/01/01 11:23:52 tcp://192.168.1.12:1883 connected
[lp-2  ] DEBUG 2024/01/01 11:23:53 charge total import: 26.689kWh
[lp-3  ] DEBUG 2024/01/01 11:23:53 charge total import: 7.774kWh
[site  ] WARN 2024/01/01 11:23:53 vehicle 'Vogue Carry 3' does not support automatic detection
[site  ] WARN 2024/01/01 11:23:53 vehicle 'Ora Funky Cat' does not support automatic detection
[site  ] INFO 2024/01/01 11:23:53 site config:
[site  ] INFO 2024/01/01 11:23:53   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2024/01/01 11:23:53     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/01/01 11:23:53     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/01/01 11:23:53   vehicles:
[site  ] INFO 2024/01/01 11:23:53     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2024/01/01 11:23:53     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/01/01 11:23:53 loadpoint 1:
[lp-1  ] INFO 2024/01/01 11:23:53   mode:        pv
[lp-1  ] INFO 2024/01/01 11:23:53   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✓
[lp-1  ] INFO 2024/01/01 11:23:53   meters:      charge ✓
[lp-1  ] INFO 2024/01/01 11:23:53     charge:    power ✓ energy ✗ currents ✓
[lp-2  ] INFO 2024/01/01 11:23:53 loadpoint 2:
[lp-2  ] INFO 2024/01/01 11:23:53   mode:        pv
[lp-2  ] INFO 2024/01/01 11:23:53   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✓
[lp-2  ] INFO 2024/01/01 11:23:53   meters:      charge ✓
[lp-2  ] INFO 2024/01/01 11:23:53     charge:    power ✓ energy ✓ currents ✓
[lp-3  ] INFO 2024/01/01 11:23:53 loadpoint 3:
[lp-3  ] INFO 2024/01/01 11:23:53   mode:        pv
[lp-3  ] INFO 2024/01/01 11:23:53   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-3  ] INFO 2024/01/01 11:23:53   meters:      charge ✓
[lp-3  ] INFO 2024/01/01 11:23:53     charge:    power ✓ energy ✓ currents ✗
[lp-1  ] DEBUG 2024/01/01 11:23:53 phase timer inactive
[lp-1  ] DEBUG 2024/01/01 11:23:53 pv timer inactive
[lp-1  ] DEBUG 2024/01/01 11:23:53 guard timer inactive
[lp-1  ] INFO 2024/01/01 11:23:53 vehicle updated: unknown -> Ora Funky Cat
[lp-2  ] DEBUG 2024/01/01 11:23:53 phase timer inactive
[lp-2  ] DEBUG 2024/01/01 11:23:53 pv timer inactive
[lp-2  ] DEBUG 2024/01/01 11:23:53 guard timer inactive
[lp-1  ] INFO 2024/01/01 11:23:53 vehicle updated: Ora Funky Cat -> unknown
[lp-2  ] INFO 2024/01/01 11:23:53 vehicle updated: unknown -> Ora Funky Cat
[lp-3  ] DEBUG 2024/01/01 11:23:53 phase timer inactive
[lp-3  ] DEBUG 2024/01/01 11:23:53 pv timer inactive
[lp-3  ] DEBUG 2024/01/01 11:23:53 guard timer inactive
[lp-3  ] INFO 2024/01/01 11:23:53 vehicle updated: unknown -> Vogue Carry 3
[site  ] WARN 2024/01/01 11:23:53 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/01/01 11:23:53 ----
[main  ] FATAL 2024/01/01 11:23:53 could not open port- check that evcc is not already running (listen tcp :7070: bind: address already in use)
```

### What type of operating system are you running?

Docker container

### Version

0.123.3

VolkerK62, 2024-01-01:
das ist #11330 
Ein Neustart behebt das Problem.

steinthomas, 2024-01-01:
Ja, nach einem Neustart v. evcc (bzw. des Docker Containers) ist der Smart Charging Link wieder anklickbar. Danke für den Hinweis. Bug or a Feature!? :-)

VolkerK62, 2024-01-01:
Bug. Siehe 11330

closed 2024-01-01 by andig