evcc-io/evcc #11413: Planner: Ladeplanung noch mit maxcurrent ohne min/max Logik?

docolli, 2024-01-02:
### Describe the bug

Kann es sein, dass die Ladeplanung noch mit ´maxcurrent´ vom vehicle arbeitet und auch hier die Logik nach https://github.com/evcc-io/evcc/issues/11280#issuecomment-1871230515 eingebaut werden muss?
[image]

@naltatis 

### Steps to reproduce

1. Definiere ´maxcurrent´ mit unterschiedlichen Werten (bei mir vehicle 22A, loadpoint 20A)
2. Gehe in die Ladeplanung

### Configuration details

```yaml
vehicles:
- type: template
  template: carwings 
  title: Nissan Leaf  
  user: *** 
  password: ***  
  capacity: 40  
  phases: 1  
  icon: car  
  cache: 15m  
  #mode: pv  
  #minSoc: 30  
  #targetSoc: 80  
  minCurrent: 6  
  maxCurrent: 22  
  name: ev1

loadpoints:
- title: Garage Carmen
  charger: wallbox1
  vehicle: ev1
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 20
#  resetOnDisconnect: false
  soc:
    poll:
      mode: always
      interval: 720m
```

### Log details

```text
kommt, wenn notwendig
```

### What type of operating system are you running?

Linux

### Version

0.123.3

andig, 2024-01-02:
Stimmt der Plan nicht oder die Anzeige im UI? Was hättest Du erwartet?

Der Plan wird so erzeugt:

	goal, _ := lp.GetPlanGoal()
	maxPower := lp.EffectiveMaxPower()
	requiredDuration := lp.GetPlanRequiredDuration(goal, maxPower)

Dabei

```go
func (lp *Loadpoint) EffectiveMaxPower() float64 {
	return Voltage * lp.effectiveMaxCurrent() * float64(lp.maxActivePhases())
}

func (lp *Loadpoint) effectiveMaxCurrent() float64 {
	maxCurrent := lp.GetMaxCurrent()

	if v := lp.GetVehicle(); v != nil {
		if res, ok := v.OnIdentified().GetMaxCurrent(); ok {
			maxCurrent = min(maxCurrent, res)
		}
	}

	if c, ok := lp.charger.(api.CurrentLimiter); ok {
		if _, res, err := c.GetMinMaxCurrent(); err == nil {
			maxCurrent = min(maxCurrent, res)
			lp.publish(keys.EffectiveMaxCurrent, maxCurrent)
		}
	}

	return maxCurrent
}
```

Sieht für mich richtig aus.

closed 2024-01-02 by andig

docolli, 2024-01-02:
Ich hätte erwartet, dass dort @ 4,7kW (=20A) steht. 
Aber ich muss mich **entschuldigen**, bin selber schuld (Layer 8 Problem)... Ich hatte übersehen, dass ich im GUI beim Loadpoint als max. Ladestrom 22A eingestellt habe. Der hat natürlich Vorrang vor dem `maxcurrent` des loadpoint in der config.yaml.

Braucht es den Wert `maxcurrent` im loadpoint in der config.yaml noch? Wenn nein, kommentiere ich den jetzt aus. Evtl. kann man wie bei den anderen alten config Werten, wenn er noch gesetzt ist, evcc nicht starten und eine Meldungs im Log ausgeben.