evcc-io/evcc #11416: Error: charger disable: invalid status: 400 upon disconnect car from Charger (Easee)

Frostexxx, 2024-01-02:
### Describe the bug

After disconnecting the car from the charger (Easee Home) the error "charger disable: invalid status: 400" is shown repeadetly. Also the loading session is still shown as active in evcc although the car has been unplugged. 

### Steps to reproduce

1. Disconnect car from charger (Easee Home)
2.
3.
...

### Configuration details

```yaml
open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: *****************************************************

interval: 10s # control cycle interval

sponsortoken: ********************************************************************************************************************************

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: shelly-3em
  usage: grid
  host: 192.168.178.67
  name: grid1

chargers:
- type: template
  template: easee
  user: ***************@*********
  password: ***************
  charger: EH3LJKDG
  name: wallbox3
- type: template
  template: easee
  user: ***************@*********
  password: ***************
  charger: EHKY3HLC
  name: wallbox4

vehicles:
- type: template
  template: hyundai
  title: Ioniq5
  user: ***************@*********
  password: ***************
  capacity: 77
  phases: 3
  icon: car
  cache: 15m
  language: de
  mode: pv
  minCurrent: 6
  maxCurrent: 16
[... 250 lines cut ...]
[lp-1  ] DEBUG 2024/01/02 12:40:13 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/01/02 12:40:13 charge total import: 2748.383kWh
[lp-1  ] DEBUG 2024/01/02 12:40:13 charger status: A
[lp-1  ] DEBUG 2024/01/02 12:40:13 set ev2 plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
[site  ] DEBUG 2024/01/02 12:40:20 ----
[lp-1  ] DEBUG 2024/01/02 12:40:20 charge power: 0W
[lp-2  ] DEBUG 2024/01/02 12:40:20 charge power: 0W
[site  ] DEBUG 2024/01/02 12:40:20 grid power: 281W
[site  ] DEBUG 2024/01/02 12:40:20 grid powers: [-77 205 153]W
[site  ] DEBUG 2024/01/02 12:40:20 grid currents: [-1.15 1.24 0.77]A
[site  ] DEBUG 2024/01/02 12:40:20 pv power: 0W
[site  ] DEBUG 2024/01/02 12:40:20 site power: 281W
[lp-2  ] DEBUG 2024/01/02 12:40:20 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/01/02 12:40:20 charge total import: 0.000kWh
[lp-2  ] DEBUG 2024/01/02 12:40:20 charger status: A
[site  ] DEBUG 2024/01/02 12:40:30 ----
```

### What type of operating system are you running?

Linux

### Version

0.123.3

andig, 2024-01-02:
Logfile passt nicht zum Fehler

closed 2024-01-02 by andig

GrimmiMeloni, 2024-01-03:
Fehler 400 passt zumindest nicht zu #11431. Könnte also auch was anderes sein. Da müßte jetzt in der Tat mal ein Log mit dem Fehler her.