evcc-io/evcc #11417: cannot create charger type 'easee': i/o timeout

jekruege, 2024-01-02:
### Describe the bug

Seit der Version 0.123.3 startet evcc bei mir nicht mehr. Die EASEE Wallbox scheint das Problem zu sein.

### Steps to reproduce

install package
run evcc

### Configuration details

```yaml
Kein Schimmer wie man die exportiert bekommt. Kann im Nano Editor nicht kopieren.
```

### Log details

```text
[main  ] INFO 2024/01/02 14:20:43 evcc 0.123.4
[main  ] INFO 2024/01/02 14:20:43 using config file: /etc/evcc.yaml
[main  ] INFO 2024/01/02 14:20:43 starting ui and api at :7070
[db    ] INFO 2024/01/02 14:20:44 using sqlite database: /home/admin/.evcc/evcc.db
[main  ] FATAL 2024/01/02 14:20:55 cannot create charger 'wallbox1': cannot create charger type 'template': cannot create charger type 'easee': i/o timeout
[main  ] FATAL 2024/01/02 14:20:55 will attempt restart in: 5m0s
[server] FATAL 2024/01/02 14:20:55 unlinkat /tmp/evcc: operation not permitted
[main  ] FATAL 2024/01/02 14:20:55 could not open port- check that evcc is not already running (listen tcp :7070: bind: address already in use)
```

### What type of operating system are you running?

Linux

### Version

0.123.4

0x3d13f, 2024-01-02:
Zunächst mal scheint evcc mehrfach zu laufen:
`could not open port- check that evcc is not already running (listen tcp :7070: bind: address already in use)`
Das würde ich erst mal abstellen.

andig, 2024-01-02:
Logfiles die das Problem nicht zeigen sind leider nicht verwertbar :(

andig, 2024-01-02:
> cannot create charger type 'easee': i/o timeout

Easee Server antworten Dir nicht. Da kann Dir nur der Easee Support helfen.

closed 2024-01-02 by andig

jekruege, 2024-01-02:
In der Version 0.123.2 antwortet aber der EASEE Server bzw. es funktioniert alles (ich bin wieder auf der Version).  

Ich hatte auch schon früher mal Kontakt mit EASEE. Wenn es zu viele Anfragen seitens EVCC gibt,  kann man eine halbe Stunde die Wallbox vom Strom nehmen, dann wird die API Queue geleert. Das funktioniert auch.

Na ich scheine ja der einzige zu sein. Dann liegt es wohl doch irgendwie bei mir.

jekruege, 2024-01-02:
> Zunächst mal scheint evcc mehrfach zu laufen (oder etwas anderes blockiert den Port 7070): `could not open port- check that evcc is not already running (listen tcp :7070: bind: address already in use)` Das würde ich erst mal abstellen.

Ja, weil es immer eingeplant ist und ich zum Test evcc an der Kommandozeile aufgerufen habe.

andig, 2024-01-02:
> In der Version 0.123.2 antwortet aber der EASEE Server bzw. es funktioniert alles (ich bin wieder auf der Version).

These: Easee hat dich wieder entblockiert. Ob Du überhaupt geblockt wurdest weiss aber nur Easee. Im Zweifel würde ein trace log für gute und schlechte Version zum Vergleich helfen (`easee: trace`).

jekruege, 2024-01-02:
Jetzt hatte ich es mit dem Trace versucht und auf einmal geht es auch mit der neuen Version. Also vermutlich doch EASEE Server ... ist nur eigenartig, dass es nur in der neuen Version auftrat.

EASEE Trace war auch unauffällig. Danke! :)

[easee ] TRACE 2024/01/02 17:28:34 POST https://streams.easee.com/hubs/chargers/negotiate
[easee ] TRACE 2024/01/02 17:28:35 {"negotiateVersion":0,"connectionId":"_mCJwoXcuCdPchzCIremgg","availableTransports":[{"transport":"WebSockets","transferFormats":["Text","Binary"]},{"transport":"ServerSentEvents","transferFormats":["Text"]},{"transport":"LongPolling","transferFormats":["Text","Binary"]}]}
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:14:41 +0000 UTC) SMART_CHARGING true
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:07:48 +0000 UTC) CABLE_LOCKED true
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:25:33 +0000 UTC) CHARGER_OP_MODE 6
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:11:03 +0000 UTC) TOTAL_POWER 0
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:07:34 +0000 UTC) SESSION_ENERGY 14.659675598144531
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 15:00:00 +0000 UTC) ENERGY_PER_HOUR 0.0880158320069313
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:07:48 +0000 UTC) WI_FI_RSSI 0
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:14:42 +0000 UTC) CELL_RSSI -104
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:19:53 +0000 UTC) OUTPUT_PHASE 10
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:14:53 +0000 UTC) DYNAMIC_CIRCUIT_CURRENT_P1 23
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:14:53 +0000 UTC) DYNAMIC_CIRCUIT_CURRENT_P2 0
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:14:53 +0000 UTC) DYNAMIC_CIRCUIT_CURRENT_P3 0
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:23:07 +0000 UTC) TEMP_MAX 16
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:08:09 +0000 UTC) SOFTWARE_RELEASE 318
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:08:09 +0000 UTC) CHARGER_RAT 1
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:08:10 +0000 UTC) LOCK_CABLE_PERMANENTLY true
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:26:00 +0000 UTC) IN_CURRENT_T2 0.00800000037997961
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:25:59 +0000 UTC) IN_CURRENT_T3 0.00800000037997961
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:25:59 +0000 UTC) IN_CURRENT_T4 0.009999999776482582
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:25:59 +0000 UTC) IN_CURRENT_T5 0.006000000052154064
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:22:49 +0000 UTC) OUTPUT_CURRENT 16
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:23:07 +0000 UTC) IN_VOLT_T2_T3 234.32899475097656
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:23:07 +0000 UTC) IN_VOLT_T2_T4 229.21499633789062
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:23:07 +0000 UTC) IN_VOLT_T2_T5 226.54800415039062
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:14:41 +0000 UTC) LEDMODE 21
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:07:48 +0000 UTC) CABLE_RATING 32
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:22:48 +0000 UTC) DYNAMIC_CHARGER_CURRENT 16
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:22:50 +0000 UTC) CIRCUIT_TOTAL_ALLOCATED_PHASE_CONDUCTOR_CURRENT_L1 16
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:14:53 +0000 UTC) CIRCUIT_TOTAL_ALLOCATED_PHASE_CONDUCTOR_CURRENT_L2 0
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:14:53 +0000 UTC) CIRCUIT_TOTAL_ALLOCATED_PHASE_CONDUCTOR_CURRENT_L3 0
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:25:59 +0000 UTC) CIRCUIT_TOTAL_PHASE_CONDUCTOR_CURRENT_L1 0.00800000037997961
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:25:59 +0000 UTC) CIRCUIT_TOTAL_PHASE_CONDUCTOR_CURRENT_L2 0.009999999776482582
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:25:59 +0000 UTC) CIRCUIT_TOTAL_PHASE_CONDUCTOR_CURRENT_L3 0.007000000216066837
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:19:53 +0000 UTC) REASON_FOR_NO_CURRENT 79
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:07:48 +0000 UTC) WI_FI_APENABLED false
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 15:38:23 +0000 UTC) LIFETIME_ENERGY 4145.5535569444455
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:08:10 +0000 UTC) MAX_CURRENT_OFFLINE_FALLBACK_P1 32
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:08:10 +0000 UTC) MAX_CURRENT_OFFLINE_FALLBACK_P2 32
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:08:10 +0000 UTC) MAX_CURRENT_OFFLINE_FALLBACK_P3 32
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:07:48 +0000 UTC) ERROR_CODE 0
[... cut ...]
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:07:34 +0000 UTC) LOCAL_PRE_AUTHORIZE_ENABLED false
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:07:34 +0000 UTC) LOCAL_AUTHORIZE_OFFLINE_ENABLED false
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:07:34 +0000 UTC) ALLOW_OFFLINE_TX_FOR_UNKNOWN_ID true
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:08:10 +0000 UTC) MAX_CHARGER_CURRENT 32
[easee ] TRACE 2024/01/02 17:28:35 ProductUpdate EHLDCW8Z: (2024-01-02 16:07:33 +0000 UTC) LED_STRIP_BRIGHTNESS 75