evcc-io/evcc #11418: Trial token is invalid

pellenbeck, 2024-01-02:
### Describe the bug

It looks like the trial token from https://docs.evcc.io/docs/sponsorship is invalid since today. Will the token be renewed?
Or is it just an example code?

### Steps to reproduce

1.
2.
3.
...
See https://docs.evcc.io/docs/sponsorship

### Configuration details

```yaml
sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJleHAiOjE3MDQyMDQwMDAsImlhdCI6MTcwMjIxNjgwMCwiaXNzIjoiZXZjYy5pbyIsInN1YiI6IlRyaWFsIFNwb25zb3JzaGlwIPCfkJkiLCJzcGUiOnRydWUsInNyYyI6Im1hIn0.t00DfXA3bcdIBBDj_SRQRHcI0eTJWL0KgS1sJ40pYW4
```

### Log details

```text
sponsortoken
rpc error: code = Unknown desc = token is expired by 45m36s
```

### What type of operating system are you running?

Linux

andig, 2024-01-02:
Danke für den Hinweis- aktualisieren wir bei Gelegenheit!

pellenbeck, 2024-01-02:
@andig das heißt um gerade eine Wallbox zu testen müsste ich einen Sponsortoken beantragen?

naltatis, 2024-01-02:
Sponsortoken ist in der Doku aktualisiert.

closed 2024-01-02 by naltatis

andig, 2024-01-02:
> @andig das heißt um gerade eine Wallbox zu testen müsste ich einen Sponsortoken beantragen?

Da könnt ihr auch immer eine Mail an info@evcc.io schreiben.