evcc-io/evcc #11419: resetondisconnect is not working

dm82m, 2024-01-02:
### Describe the bug

See: https://github.com/evcc-io/evcc/discussions/10978#discussioncomment-7992246

If I disconnect my car from the Easee Wallbox, mode is not reset to 'off' as expected or if it was before 0.123.x.

### Steps to reproduce

1. Connect the car, set the correct car in Evcc and set mode from off to PV.
2. Disconnect the car from Easee wallbox.
3. Car keeps beeing set and mode keeps also beeing set in evcc.
4. According to breaking changes of 0.123 it should work as before (resetondisconnect) as it set mode to off and removed resetondisconnect from config file.

### Configuration details

```yaml
loadpoints:
  - title: Dirk # display name for UI
    charger: chargerDirk # charger
    mode: off # set default charge mode, use "off" to disable by default if charger is publicly available
    phases: 0 # electrical connection (normal charger: default 3 for 3 phase, 1p3p charger: 0 for "auto" or 1/3 for fixed phases)
    minCurrent: 6 # minimum charge current (default 6A)
    maxCurrent: 16 # maximum charge current (default 16A)
    # remaining settings are experts-only and best left at default values
    priority: 0 # relative priority for concurrent charging in PV mode with multiple loadpoints (higher values have higher priority)
    soc:
      # polling defines usage of the vehicle APIs
      # Modifying the default settings it NOT recommended. It MAY deplete your vehicle's battery
      # or lead to vehicle manufacturer banning you from API use. USE AT YOUR OWN RISK.
      poll:
        # poll mode defines under which condition the vehicle API is called:
        #   charging: update vehicle ONLY when charging (this is the recommended default)
        #   connected: update vehicle when connected (not only charging), interval defines how often
        #   always: always update vehicle regardless of connection state, interval defines how often (only supported for single vehicle)
        mode: charging
        # poll interval defines how often the vehicle API may be polled if NOT charging
        interval: 60m
      estimate: true # set false to disable interpolating between api updates (not recommended)
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: 0 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 3m # threshold must be exceeded for this long
      threshold: 0 # maximum import power (W)
    guardDuration: 5m # switch charger contactor not more often than this (default 5m)
```

### Log details

```text
[site  ] DEBUG 2024/01/02 15:56:14 ----
[lp-1  ] DEBUG 2024/01/02 15:56:14 charge power: 0W
[lp-2  ] DEBUG 2024/01/02 15:56:14 charge power: 0W
[site  ] DEBUG 2024/01/02 15:56:14 pv power: 57W
[site  ] DEBUG 2024/01/02 15:56:14 battery soc: 25%
[site  ] DEBUG 2024/01/02 15:56:14 battery power: 1W
[site  ] DEBUG 2024/01/02 15:56:14 grid power: 441W
[site  ] DEBUG 2024/01/02 15:56:14 grid currents: [0.851 2.56 0.626]A
[site  ] DEBUG 2024/01/02 15:56:14 site power: 542W
[lp-1  ] DEBUG 2024/01/02 15:56:14 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/01/02 15:56:14 charge total import: 1049.808kWh
[lp-1  ] DEBUG 2024/01/02 15:56:14 charger status: B
[lp-1  ] DEBUG 2024/01/02 15:56:14 set dima-car plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
[lp-1  ] DEBUG 2024/01/02 15:56:14 pv charge current: 0A = 0A + -2.35A (542W @ 1p)
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.123.4

andig, 2024-01-02:
Please read the release notes. Use `mode` instead.

andig, 2024-01-02:
Please provide logfile showing the issue. Will reopen when log available.

closed 2024-01-02 by andig

dm82m, 2024-01-02:
@andig hier das log: ich habe im log zeitraum das ladekabel ab- und wieder angesteckt. und zwar komplett. von der easee weg und vom fahrzeug weg. zustand in evcc bleibt wie beschrieben: fahrzeug bleibt eingestellt und mode bleibt eingestellt. im log erkenne ich persönlich nichts. sag mir bescheid wenn ich mehr logging einstellen soll und wenn ja auch was genau ihr braucht:

```
Jan 02 17:09:14 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:09:14 pv power: 0W
Jan 02 17:09:14 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:09:14 battery soc: 25%
Jan 02 17:09:14 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:09:14 battery power: 0W
Jan 02 17:09:14 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:09:14 grid power: 4921W
Jan 02 17:09:14 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:09:14 grid currents: [1.01 20 0.647]A
Jan 02 17:09:14 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:09:14 site power: 5021W
Jan 02 17:09:14 homeassistant addon_49686a9f_evcc[450]: [lp-1  ] DEBUG 2024/01/02 18:09:14 charge currents: [0.007 0.009 0.009]A
Jan 02 17:09:14 homeassistant addon_49686a9f_evcc[450]: [lp-1  ] DEBUG 2024/01/02 18:09:14 charge total import: 1049.808kWh
Jan 02 17:09:14 homeassistant addon_49686a9f_evcc[450]: [lp-1  ] DEBUG 2024/01/02 18:09:14 charger status: B
Jan 02 17:09:14 homeassistant addon_49686a9f_evcc[450]: [lp-1  ] DEBUG 2024/01/02 18:09:14 set dima-car plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
Jan 02 17:09:14 homeassistant addon_49686a9f_evcc[450]: [lp-1  ] DEBUG 2024/01/02 18:09:14 pv charge current: 0A = 0A + -21.8A (5021W @ 1p)
Jan 02 17:09:44 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:09:44 ----
Jan 02 17:09:44 homeassistant addon_49686a9f_evcc[450]: [lp-1  ] DEBUG 2024/01/02 18:09:44 charge power: 0W
Jan 02 17:09:44 homeassistant addon_49686a9f_evcc[450]: [lp-2  ] DEBUG 2024/01/02 18:09:44 charge power: 0W
Jan 02 17:09:44 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:09:44 pv power: 0W
Jan 02 17:09:44 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:09:44 battery soc: 25%
Jan 02 17:09:44 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:09:44 battery power: 0W
Jan 02 17:09:44 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:09:44 grid power: 6154W
Jan 02 17:09:44 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:09:44 grid currents: [1.03 25.2 0.655]A
Jan 02 17:09:44 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:09:44 site power: 6254W
Jan 02 17:09:44 homeassistant addon_49686a9f_evcc[450]: [lp-2  ] DEBUG 2024/01/02 18:09:44 charge currents: [0 0 0]A
Jan 02 17:09:44 homeassistant addon_49686a9f_evcc[450]: [lp-2  ] DEBUG 2024/01/02 18:09:44 charge total import: 0.000kWh
Jan 02 17:09:44 homeassistant addon_49686a9f_evcc[450]: [lp-2  ] DEBUG 2024/01/02 18:09:44 charger status: A
Jan 02 17:10:14 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:10:14 ----
Jan 02 17:10:14 homeassistant addon_49686a9f_evcc[450]: [lp-1  ] DEBUG 2024/01/02 18:10:14 charge power: 0W
Jan 02 17:10:14 homeassistant addon_49686a9f_evcc[450]: [lp-2  ] DEBUG 2024/01/02 18:10:14 charge power: 0W
Jan 02 17:10:14 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:10:14 pv power: 0W
Jan 02 17:10:14 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:10:14 battery soc: 25%
Jan 02 17:10:14 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:10:14 battery power: 0W
Jan 02 17:10:14 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:10:14 grid power: 5561W
Jan 02 17:10:14 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:10:14 grid currents: [1.05 22.7 0.656]A
Jan 02 17:10:14 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:10:14 site power: 5661W
Jan 02 17:10:14 homeassistant addon_49686a9f_evcc[450]: [lp-1  ] DEBUG 2024/01/02 18:10:14 charge currents: [0.007 0.009 0.009]A
Jan 02 17:10:14 homeassistant addon_49686a9f_evcc[450]: [lp-1  ] DEBUG 2024/01/02 18:10:14 charge total import: 1049.808kWh
Jan 02 17:10:14 homeassistant addon_49686a9f_evcc[450]: [lp-1  ] DEBUG 2024/01/02 18:10:14 charger status: B
Jan 02 17:10:14 homeassistant addon_49686a9f_evcc[450]: [lp-1  ] DEBUG 2024/01/02 18:10:14 set dima-car plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
Jan 02 17:10:14 homeassistant addon_49686a9f_evcc[450]: [lp-1  ] DEBUG 2024/01/02 18:10:14 pv charge current: 0A = 0A + -24.6A (5661W @ 1p)
Jan 02 17:10:44 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:10:44 ----
Jan 02 17:10:44 homeassistant addon_49686a9f_evcc[450]: [lp-1  ] DEBUG 2024/01/02 18:10:44 charge power: 0W
Jan 02 17:10:44 homeassistant addon_49686a9f_evcc[450]: [lp-2  ] DEBUG 2024/01/02 18:10:44 charge power: 0W
Jan 02 17:10:44 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:10:44 pv power: 0W
Jan 02 17:10:44 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:10:44 battery soc: 25%
Jan 02 17:10:44 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:10:44 battery power: 0W
Jan 02 17:10:44 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:10:44 grid power: 3544W
[... cut ...]
Jan 02 17:12:44 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:12:44 grid currents: [1.22 14.8 0.66]A
Jan 02 17:12:44 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:12:44 site power: 3784W
Jan 02 17:12:44 homeassistant addon_49686a9f_evcc[450]: [lp-2  ] DEBUG 2024/01/02 18:12:44 charge currents: [0 0 0]A
Jan 02 17:12:44 homeassistant addon_49686a9f_evcc[450]: [lp-2  ] DEBUG 2024/01/02 18:12:44 charge total import: 0.000kWh
Jan 02 17:12:44 homeassistant addon_49686a9f_evcc[450]: [lp-2  ] DEBUG 2024/01/02 18:12:44 charger status: A

```

dm82m, 2024-01-02:
jetzt noch mal nach frischen neustart. man sieht im log, dass ich das fahrzeug auswähle und dadurch via "onIdentify" der mode auf pv gesetzt wird. evcc bekommt vom abstecken des fahrzeugs an der easee wallbox aber nichts mit. fahrzeug bleibt gesetzt. mode bleibt auf pv gesetzt und geht nicht auf off zurück.

```
Jan 02 17:41:07 homeassistant addon_49686a9f_evcc[450]: Using config file: /config/evcc.yaml
Jan 02 17:41:07 homeassistant addon_49686a9f_evcc[450]: starting evcc: 'EVCC_DATABASE_DSN=/config/evcc.db evcc --config /config/evcc.yaml'
Jan 02 17:41:07 homeassistant addon_49686a9f_evcc[450]: [main  ] INFO 2024/01/02 18:41:07 evcc 0.123.4
Jan 02 17:41:07 homeassistant addon_49686a9f_evcc[450]: [main  ] INFO 2024/01/02 18:41:07 using config file: /config/evcc.yaml
Jan 02 17:41:07 homeassistant addon_49686a9f_evcc[450]: [main  ] INFO 2024/01/02 18:41:07 starting ui and api at :7070
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [lp-1  ] DEBUG 2024/01/02 18:41:12 charge total import: 1049.808kWh
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [lp-2  ] DEBUG 2024/01/02 18:41:12 charge total import: 0.000kWh
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [site  ] INFO 2024/01/02 18:41:12 site config:
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [site  ] INFO 2024/01/02 18:41:12   meters:      grid ✓ pv ✓ battery ✓
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [site  ] INFO 2024/01/02 18:41:12     grid:      power ✓ energy ✓ currents ✓
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [site  ] INFO 2024/01/02 18:41:12     pv 1:      power ✓ energy ✓ currents ✗
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [site  ] INFO 2024/01/02 18:41:12     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [site  ] INFO 2024/01/02 18:41:12   vehicles:
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [site  ] INFO 2024/01/02 18:41:12     vehicle 1: range ✓ finish ✗ status ✗ climate ✗ wakeup ✗
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [lp-1  ] INFO 2024/01/02 18:41:12 loadpoint 1:
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [lp-1  ] INFO 2024/01/02 18:41:12   mode:        pv
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [lp-1  ] INFO 2024/01/02 18:41:12   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [lp-1  ] INFO 2024/01/02 18:41:12   meters:      charge ✓
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [lp-1  ] INFO 2024/01/02 18:41:12     charge:    power ✓ energy ✓ currents ✓
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [lp-2  ] INFO 2024/01/02 18:41:12 loadpoint 2:
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [lp-2  ] INFO 2024/01/02 18:41:12   mode:        off
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [lp-2  ] INFO 2024/01/02 18:41:12   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [lp-2  ] INFO 2024/01/02 18:41:12   meters:      charge ✓
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [lp-2  ] INFO 2024/01/02 18:41:12     charge:    power ✓ energy ✓ currents ✓
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:41:12 set buffer soc: 95
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:41:12 set buffer start soc: 0
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:41:12 set priority soc: 70
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [lp-1  ] DEBUG 2024/01/02 18:41:12 phase timer inactive
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [lp-1  ] DEBUG 2024/01/02 18:41:12 pv timer inactive
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [lp-1  ] DEBUG 2024/01/02 18:41:12 guard timer inactive
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [lp-2  ] DEBUG 2024/01/02 18:41:12 phase timer inactive
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [lp-2  ] DEBUG 2024/01/02 18:41:12 pv timer inactive
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [lp-2  ] DEBUG 2024/01/02 18:41:12 guard timer inactive
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:41:12 ----
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [lp-1  ] DEBUG 2024/01/02 18:41:12 charge power: 0W
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [lp-2  ] DEBUG 2024/01/02 18:41:12 charge power: 0W
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:41:12 pv power: 0W
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:41:12 battery soc: 25%
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:41:12 battery power: 0W
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:41:12 grid power: 439W
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:41:12 grid currents: [1.14 2.29 0.653]A
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:41:12 site power: 539W
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [lp-1  ] DEBUG 2024/01/02 18:41:12 charge currents: [0 0 0]A
Jan 02 17:41:12 homeassistant addon_49686a9f_evcc[450]: [lp-1  ] DEBUG 2024/01/02 18:41:12 charge total import: 1049.808kWh
[... cut ...]
Jan 02 17:45:42 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:45:42 grid currents: [1.13 2.42 0.655]A
Jan 02 17:45:42 homeassistant addon_49686a9f_evcc[450]: [site  ] DEBUG 2024/01/02 18:45:42 site power: 584W
Jan 02 17:45:42 homeassistant addon_49686a9f_evcc[450]: [lp-2  ] DEBUG 2024/01/02 18:45:42 charge currents: [0 0 0]A
Jan 02 17:45:42 homeassistant addon_49686a9f_evcc[450]: [lp-2  ] DEBUG 2024/01/02 18:45:42 charge total import: 0.000kWh
Jan 02 17:45:42 homeassistant addon_49686a9f_evcc[450]: [lp-2  ] DEBUG 2024/01/02 18:45:42 charger status: A
```

dm82m, 2024-01-02:
also nachdem das fahrzeug um 18:43:48 gesetzt wurde hab ich kurz gewartet und dann das fahrzeug wiederabgesteckt -> da kommt im log gar nichts mehr dazu

dm82m, 2024-01-02:
auch in der evcc gui bleibt der status "Verbunden.". evcc bekommt die status änderung auf der easee nicht mehr mit. ich vermute ehrlich gesagt auch, dass das die gleiche ursache für https://github.com/evcc-io/evcc/issues/11420 ist. da brauchen wir vmtl. wirklich den @GrimmiMeloni mit seinem wissen.

GrimmiMeloni, 2024-01-02:
Im Log sieht man leider erstmal nichts. Setz bitte mal easee auf trace und wiederhole den Test.

demichve, 2024-01-02:
Ich habe das gleiche Problem und bin auf 0.123.2. Trotz abziehen des Ladekabels bleibt der Status in evcc auf „verbunden“.
Ich hatte das Problem auf die Serverprobleme bei Easee geschoben, aber lässt sich reproduzieren. Ein Neustart von EVCC hilft.

GrimmiMeloni, 2024-01-02:
Ich würde das gerne mal versuchen hier lokal zu reproduzieren.
Wie habt Ihr die Autos konfiguriert? Noch über die evcc.yaml oder bereits über die UI?

dm82m, 2024-01-02:
evcc.yaml ; meine config siehst du komplett hier: https://github.com/evcc-io/evcc/issues/11420

GrimmiMeloni, 2024-01-02:
caused by #11431