evcc-io/evcc #11423: kuriose Batterieansicht in 0.123.4

VolkerK62, 2024-01-02:
### Describe the bug

Habe es auf 2 verschiedenen Browsern und am Handy ausprobiert. Und auch nochmal einen Restart gemacht.
Bild ist immer gleich.

[image]

### Steps to reproduce

.

### Configuration details

```yaml
.
```

### Log details

```text
.
```

### What type of operating system are you running?

Linux

### Version

0.123.4

VolkerK62, 2024-01-02:
Wenn man die Prozente anklickt und verändert, ist die Anzeige wieder wie gewohnt, aber nur auf dem Gerät, auf dem man diese Änderungen vorgenommen hat.

docolli, 2024-01-02:
Kein Problem hier bei mir unter Firefox 121.0:
[image]

Schon mal versucht alles neu zu laden indem man mit STRG+F5 aktualisiert?

VolkerK62, 2024-01-02:
Nachdem ich jetzt auf allen Geräten einmal den Soc im UI geändert habe, ist überall die Anzeige ok. Auch nach Neustart des Gerätes und löschen vom Cache ... usw.
Keine Ahnung, was das war.
Ich mach erstmal wieder zu.

naltatis, 2024-01-02:
@VolkerK62 In welchem Browser / OS war das? Beim nächsten mal bitte einmal einen Auszug aus `/api/state` beifügen. Vielleicht ist das was Aufschlussreiches drin.

VolkerK62, 2024-01-02:
@docolli  funktional war alles ok. Im MQTT Explorer wurden die richtige Werte angezeigt. War ein reines Anzeigeproblem.

@naltatis  Aufgefallen ist es direkt nach dem Update auf 0.123.4. auf dem Laptop (Win10) im Chrome. Dann hab ich auf dem Android Handy (auch Chrome) gecheckt. Gleiche kuriose Anzeige. Im MQTT Explorer wurden die erwarteten Werte angezeigt. Dann hab ich auf dem Laptop im Chrome mal die überlappenden Prozentwerte angeklickt und einen anderen Wert eingestellt und schon war die Anzeige wieder ok. Auf dem Handy neu aufgerufen. Immer noch kuriose Anzeige. Prozente angeklickt, alles wieder ok. Dann auf dem Laptop Firefox geöffnet. Gleiche kuriose Anzeige. Dort ebenfalls korrigiert. evcc neu gestartet, Laptop neu gestartet. Anzeige weiterhin ok. Dann nochmal im Edge geschaut, dort war die Anzeige aber ok, obwohl ich da vorher nichts geändert habe. An /api/state habe ich in dem Moment gar nicht gedacht.

Hofyyy, 2024-01-02:
Gleicher Fehler bei mir. Chrome auf dem Mac. und safari als desktop link auf IOs. als wenn die defauls nicht passen. 

[image]

docolli, 2024-01-02:
Was enthält die /api/state?

Hofyyy, 2024-01-02:
Ich kann das leider gerade nur so posten. sry

{"result":{"auth":{"vehicles":{}},"battery":[{"power":4,"soc":57,"capacity":5,"controllable":true}],"batteryCapacity":5,"batteryConfigured":true,"batteryEnergy":0,"batteryMode":"unknown","batteryPower":4,"batterySoc":57,"currency":"EUR","greenShareHome":0.013559322033898305,"greenShareLoadpoints":0,"gridConfigured":true,"gridPower":291,"homePower":295,"loadpoints":[{"chargeCurrent":0,"chargeCurrents":[0,0,0],"chargeDuration":300000000000,"chargePower":0,"chargeRemainingDuration":0,"chargeRemainingEnergy":16084.444444444443,"chargeTotalImport":6088.351,"chargedEnergy":269.99999999952706,"chargerFeatureHeating":false,"chargerFeatureIntegratedDevice":false,"chargerIcon":null,"charging":false,"connected":true,"connectedDuration":0,"disableThreshold":0,"effectiveLimitSoc":100,"effectiveMaxCurrent":16,"effectiveMinCurrent":6,"effectivePlanSoc":100,"effectivePlanTime":"2024-01-03T06:00:00Z","effectivePriority":0,"enableThreshold":0,"enabled":false,"guardAction":"inactive","guardRemaining":0,"limitEnergy":0,"limitSoc":0,"maxCurrent":16,"minCurrent":6,"mode":"pv","phaseAction":"inactive","phaseRemaining":0,"phasesActive":1,"phasesConfigured":null,"phasesEnabled":3,"planEnergy":0,"planProjectedStart":"2024-01-03T01:32:25+01:00","planTime":"0001-01-01T00:00:00Z","priority":0,"pvAction":"inactive","pvRemaining":0,"sessionCo2PerKWh":null,"sessionEnergy":269.99999999952706,"sessionPrice":0.06933599999987855,"sessionPricePerKWh":0.25680000000000003,"sessionSolarPercentage":0,"title":"WALLBE","vehicleCapacity":18.8,"vehicleClimaterActive":null,"vehicleDetectionActive":false,"vehicleFeatureOffline":false,"vehicleIcon":"car","vehicleName":"Bounty","vehicleOdometer":80975,"vehiclePresent":true,"vehicleRange":18,"vehicleSoc":23,"vehicleTargetSoc":0,"vehicleTitle":"BMWi3"},{"chargeCurrent":0,"chargeCurrents":[0,0.2,0.4],"chargeDuration":0,"chargePower":0,"chargeVoltages":[2.7,0,0],"chargedEnergy":36075.898,"chargerFeatureHeating":false,"chargerFeatureIntegratedDevice":false,"chargerIcon":null,"charging":false,"connected":false,"connectedDuration":9223372036854775807,"disableThreshold":0,"effectiveLimitSoc":100,"effectiveMaxCurrent":16,"effectiveMinCurrent":6,"effectivePlanSoc":0,"effectivePlanTime":"0001-01-01T00:00:00Z","effectivePriority":0,"enableThreshold":0,"enabled":false,"guardAction":"inactive","guardRemaining":0,"limitEnergy":0,"limitSoc":0,"maxCurrent":16,"minCurrent":6,"mode":"now","phaseAction":"inactive","phaseRemaining":0,"phasesActive":3,"phasesConfigured":null,"phasesEnabled":3,"planEnergy":0,"planProjectedStart":"0001-01-01T00:00:00Z","planTime":"0001-01-01T00:00:00Z","priority":0,"pvAction":"inactive","pvRemaining":0,"sessionCo2PerKWh":null,"sessionEnergy":36075.898,"sessionPrice":10.032707233800002,"sessionPricePerKWh":0.2781,"sessionSolarPercentage":0,"title":"TWC3","vehicleCapacity":0,"vehicleClimaterActive":null,"vehicleDetectionActive":false,"vehicleFeatureOffline":false,"vehicleIcon":"","vehicleName":"","vehicleOdometer":0,"vehiclePresent":false,"vehicleRange":0,"vehicleSoc":0,"vehicleTargetSoc":0,"vehicleTitle":""},{"chargeCurrent":16,"chargeDuration":0,"chargePower":0,"chargeTotalImport":71.61043799999999,"chargedEnergy":0,"chargerFeatureHeating":false,"chargerFeatureIntegratedDevice":true,"chargerIcon":"heater","charging":false,"connected":true,"connectedDuration":0,"disableThreshold":0,"effectiveLimitSoc":100,"effectiveMaxCurrent":16,"effectiveMinCurrent":6,"effectivePlanSoc":0,"effectivePlanTime":"0001-01-01T00:00:00Z","effectivePriority":0,"enableThreshold":0,"enabled":true,"guardAction":"inactive","guardRemaining":0,"limitEnergy":0,"limitSoc":0,"maxCurrent":16,"minCurrent":6,"mode":"now","phaseAction":"inactive","phaseRemaining":0,"phasesActive":3,"phasesConfigured":null,"phasesEnabled":3,"planEnergy":0,"planProjectedStart":"0001-01-01T00:00:00Z","planTime":"0001-01-01T00:00:00Z","priority":0,"pvAction":"inactive","pvRemaining":0,"sessionCo2PerKWh":null,"sessionEnergy":0,"sessionPrice":null,"sessionPricePerKWh":null,"sessionSolarPercentage":0,"title":"Waermepumpe","vehicleCapacity":0,"vehicleDetectionActive":false,"vehicleIcon":"","vehicleName":"","vehicleOdometer":0,"vehiclePresent":false,"vehicleTitle":""}],"pv":[{"power":0}],"pvConfigured":true,"pvEnergy":0,"pvPower":0,"residualPower":-50,"siteTitle":"Am Eschenacker 3","smartCostActive":false,"smartCostType":"priceforecast","sponsor":"Hofyyy","sponsorTokenExpires":0,"statistics":{"30d":{"avgCo2":0,"avgPrice":0.1788957897606828,"chargedKWh":49.50067106510246,"solarPercentage":1.314335255849076},"365d":{"avgCo2":0,"avgPrice":0.1788957897606828,"chargedKWh":49.50067106510246,"solarPercentage":1.314335255849076},"total":{"avgCo2":0,"avgPrice":0.1788957897606828,"chargedKWh":49.50067106510246,"solarPercentage":1.314335255849076}},"tariffFeedIn":0,"tariffGrid":0.2447,"tariffPriceHome":0.2413820338983051,"tariffPriceLoadpoints":0.2447,"vehicles":{"Bounty":{"title":"BMWi3","plans":[{"soc":100,"time":"2024-01-03T06:00:00Z"}]},"Bunny":{"title":"TeslaY"},"SmartEQ":{"title":"SmartEQ"}},"version":"0.123.4"}}

docolli, 2024-01-02:
Wir suchen buffersoc und prioritysoc.
[image]
Die finde ich bei Dir nicht, sind also nicht gesetzt!

Seit 0.123.0 kommen diese Werte NICHT MEHR aus der config.yaml! Sind also nach einem Upgrade erstmal undefiniert (das könnte man im Code ändern) und MÜSSEN einmal manuell im GUI gesetzt werden.

VolkerK62, 2024-01-03:
Jetzt wo du es sagst (NaN), sehe ich es auch. Ich habe reingeklickt, einen Wert eingestellt, dann hat es gut ausgesehen. Der neu eingestellte Wert war `prioritysoc`. Buffersoc war wie erwartet auf 100%
Meine aktuell eingestellten Werte (auch vor dem update) sind: buffer=100%, priority=0%.
Ich habe diese Werte seit 0.123.0 schon verändert. Ob im UI, kann ich nicht sicher sagen, aber auf jeden Fall per API.
Seit 0.123.0 habe ich fast täglich auf das jeweilige Nightly aktualisiert. 
Als ich die komische Ansicht gesehen habe, dachte ich erst, dass das durch das Rausnehmen des Steuerungshinweises passiert ist.

VolkerK62, 2024-01-03:
Update auf 0.123.5.  und danach ist der Fehler wieder da.
`prioritysoc` und `bufferstartsoc` fehlen in `/api/state`

```
{
    "result": {
        "auth": {
            "vehicles": {}
        },
        "aux": [
            {
                "power": 0
            }
        ],
        "auxPower": 0,
        "battery": [
            {
                "power": -45.23312100742248,
                "soc": 40,
                "capacity": 7.5,
                "controllable": false
            }
        ],
        "batteryCapacity": 7.5,
        "batteryConfigured": true,
        "batteryEnergy": 0,
        "batteryMode": "normal",
        "batteryPower": -45.23312100742248,
        "batterySoc": 40,
        "bufferSoc": 100,
        "currency": "EUR",
        "greenShareHome": 0,
        "greenShareLoadpoints": 0,
        "gridConfigured": true,
        "gridCurrents": [
            0.85,
            0.69,
            0.54
        ],
        "gridEnergy": 3932.7282000000005,
        "gridPower": 251.3555657087958,
        "gridPowers": [
            151.78,
            45.89,
            51.16
        ],
        "homePower": 206.12244470137333,
        "loadpoints": [
            {
                "chargeCurrent": 0,
                "chargeCurrents": [
                    0,
                    0,
                    0
                ],
                "chargeDuration": 0,
                "chargePower": 0,
                "chargeTotalImport": 1373.9,
                "chargeVoltages": [
                    228.7,
                    0,
                    0
                ],
                "chargedEnergy": 0,
                "chargerFeatureHeating": false,
                "chargerFeatureIntegratedDevice": false,
                "chargerIcon": null,
                "charging": false,
                "connected": false,
                "connectedDuration": 9223372036854776000,
[... 390 lines cut ...]
            },
            "WasserSG": {
                "title": "WasserSG",
                "limitSoc": 60
            },
            "WasserWP": {
                "title": "WasserWP"
            },
            "auto": {
                "title": "(T)Raumschiff",
                "limitSoc": 100
            },
            "bike": {
                "title": "E-Bikes",
                "minSoc": 66,
                "limitSoc": 100
            },
            "offline": {
                "title": "ohne Bluelink"
            }
        },
        "version": "0.123.5 (7819662a)"
    }
}
```

VolkerK62, 2024-01-03:
Im Log nach dem Start after update finde ich diese (richtigen) Einträge:
```
Jan 03 05:51:06 raspberrypi evcc[28288]: [site  ] DEBUG 2024/01/03 05:51:06 set buffer soc: 100
Jan 03 05:51:06 raspberrypi evcc[28288]: [site  ] DEBUG 2024/01/03 05:51:06 set buffer start soc: 0
Jan 03 05:51:06 raspberrypi evcc[28288]: [site  ] DEBUG 2024/01/03 05:51:06 set smart cost limit: 0
Jan 03 05:51:06 raspberrypi evcc[28288]: [site  ] DEBUG 2024/01/03 05:51:06 set priority soc: 0
```

Velandriel, 2024-01-03:
Die Ansicht sieht bei mir auch so aus:
[image]

naltatis, 2024-01-03:
Danke für die Werte. Schau ich mir an.

PR #11441 (2024-01-03): Restore: publishing initial site values
fixes #11438
fixes #11428
fixes #11423

Revert from https://github.com/evcc-io/evcc/commit/f14b83cd8ba63d772f71cca6d1adaf195ac68248#diff-7aa5baae0bd3243f85bfc6db64612f7450eb0170fa310ca2c93dc3968fc02e02
closed 2024-01-03 by andig