evcc-io/evcc #11425: Planner: Angezeigte Restzeit im GUI bezieht sich auf effectiveLimitSoc und nicht auf effectivePlanSoc

docolli, 2024-01-02:
### Describe the bug

Wenn ein Plan (Zielladen) aktiv ist, bezieht sich die Restzeitanzeige im GUI auf das gewählte limitSoc und nicht auf das im Plan gewählte Soc. Zumindest wenn das im Plan gewählte Soc größer ist, als das limitSoc (damit habe ich die Beobachtung gemacht). Das ist verwirrend, im meinem gezeigten Fall, kann man denken, dass die Ladung in 30 min. fertig ist. 

[image]
Start des Plans, Restzeit ~35min.

[image]
Auto hat 79%, Restzeit sei nur noch 8min. Das ist definitiv nicht richtig.

[image]
Auto hat 80%, Restzeit nur noch 1s ! Obwohl Plan auf 90% Ladung steht.

[image]
Danach wird nur noch die Ladedauer angezeigt, Restzeit (bis 90%) gibt es nicht mehr in der Auswahl.

[image]

[image]
Es ist 17:36h, die Ladung ist bereits fertig (was ich gut finde, lieber etwas Zeitpuffer drin). Wäre aber schön, wenn man das reale Ladeende über die reale Restzeit sehen könnte.

Wenn ein Plan läuft, sollte meiner Meinung nach der Ladelimitbalken nicht mehr im GUI erscheinen, da er in diesem Zustand keine Relevanz hat. Es gilt jetzt das gewählte Limit im Plan.
Auch die Anzeige "Ladelimit 80%" mit geschätzer Reichweite am rechten Ende des Ladebalkens ist, wenn ein Plan ausgeführt wird, nicht mehr relevant. Hier sollte man sich dann auch auf die Planwerte beziehen.

Wir hatten das schon mal für minSoc hier https://github.com/evcc-io/evcc/issues/10628
Schon klar, dass dies zur Zeit nicht so leicht umsetzbar ist, aber nehmt die Info als evtl. zu beachtenden Aspekt, wenn ihr die wiederholenden Pläne implementiert.

### Steps to reproduce

1. Ladelimit auf 80% setzen
2. Ladeplanung mit Ladeziel 90% erstellen
3. Auto mit Soc < 80% (z.B. 70%) anstecken
4. Restzeit im GUI beachten
...

### Configuration details

```yaml
Kommt, wenn nötig
```

### Log details

```text
Kommt, wenn nötig
```

### What type of operating system are you running?

Linux

### Version

0.123.4

naltatis, 2024-01-03:
Das ist ja eher ein generelles Thema, was "Restzeit" inhaltlich bedeutet. Momentan ist das ja auch abhängig vom Fahrzeug. Heißt, wenn das Fahrzeug eine eigene Zeitschätzung hat (bspw. Tesla), wird hier der gleiche Wert wie in der Tesla App angezeigt. Die weiß natürlich nichts von den Limits und Plänen in evcc.

Hier müssten wir, vor allem in Hinblick auf die unterschiedlichen Limits, mal überlegen, was dieser Wert aus Nutzersicht aussagen soll. Ich skizziere mal das von mir erwartete Verhalten:

1. Limit aktiv: Zeit bis zum Erreichen des Ladelimits (bei aktueller Ladeleistung)
2. Fahrzeuglimit (API) < Limit: Zeit bis zum Erreichen des Fahrzeuglimits (bei aktueller Ladeleistung)
3. Plan aktiv: Zeit bis zum Ende des letzten Plan-Zeitfensters

Dafür müssten wir ein wenig umbauen und vmtl. auch die Restzeitangaben von der Fahrzeug-API (bspw. Tesla) rauswerfen. Diese ist ja nur in Szenario 2 korrekt.

docolli, 2024-01-03:
Dieses skizzierte Verhalten würde ich als Nutzer auch erwarten. Das mit der eigenen Zeitschätzung in den Apps der Fzg-Hersteller kannte ich noch nicht, aber ja, das sollte einigermaßen synchron sein.

Dann würde aber die Restzeit (u.a. Anzeige im GUI, API, MQTT) konsequenterweise weg vom loadpoint runter zum vehicle wandern müssen. Diese ist eine Eigenschaft des angeschlossenen Fahrzeugs und hängt (außer von der Ladleistung des loadpoints) von mehreren Fahrzeug-Eigenschaften ab (wie Kapazität des Akkkus, Maximale Ladeleistung). Somit wäre es wieder kompatibel mit der Fzg-API in Szenario 2.

docolli, 2024-01-03:
Beim Rumspielen mit dem Ladelimitslider kam mit gerade eine Idee.  Wenn ich den Slider auf den gerade im Plan eingestellte Limit verschiebe, dann stimmen die Werte ja.
[image]

Könnte man nicht einfach den Slider (und das Textfeld "Ladelimit" das effectiveLimitSoc anzeigen lassen und die Restzeit wird auch immer mit dem effectiveLImitSoc berechnet? Ich denke mit den wiederkehrenden Plänen kann das vereinheitlicht werden, das jetzige Ladelimit ist ja schon so eine Art "Basis-Plan" mit Soc Limit, nur ohne Zeitangabe, sondern gilt immer.

naltatis, 2024-01-03:
@docolli ich hab das Gefühl, dass hier jetzt einige Themen gemischt werden. Wir haben im letzten Release bewusst das Ladelimit vom Plan-Ziel getrennt, um mehr Flexibilität zu ermöglichen: temporäres Übersteuern, unterschiedliche Ziele pro Plan.

Ich bin mir noch nicht ganz sicher, welches Problem du hier lösen möchtest. Gehts nur um die Restzeitermittlung?

docolli, 2024-01-03:
Ja, sorry, stecke halt nicht so tief drin, wie Ihr... 😉 Und die Gedanken fliesen...

Primär geht es mir hier schon um die Restzeit. Und dabei ist mir heute morgen aufgefallen, dass wenn der Slider für das Ladelimit im GUI immer auf dem aktuell gültigen Limit (Default oder von einem Plan) steht, dass dann bereits die Berechung der Restzeit passend ist.

Es können ja für jeden Zeitpunkt des Tages nur die Default Limit Werte oder die Werte von 1 Plan gültig sein. Auch wenn es mehrere (konkurrierende) Pläne gibt, kann nur 1 davon durchgeführt werden. Aber das wird jetzt OT.

>Wir haben im letzten Release bewusst das Ladelimit vom Plan-Ziel getrennt, um mehr Flexibilität zu ermöglichen: temporäres Übersteuern, unterschiedliche Ziele pro Plan.

Da habt ihr sicherlich einen Thread, wo ihr das diskutiert und wir mit diesem Thema weiter machen können. Ich verstehe jetzt den Anwendungsfall "temporäres Übersteuern" bzw. "unterschiedliche Ziele pro Plan" nicht. Wo ist den denn das Thema "zukünftige Ladeplanung" zu finden?

naltatis, 2024-01-03:
Die passende Diskussion gibts hier: https://github.com/evcc-io/evcc/issues/5492

andig, 2024-01-03:
> Dafür müssten wir ein wenig umbauen und vmtl. auch die Restzeitangaben von der Fahrzeug-API (bspw. Tesla) rauswerfen. Diese ist ja nur in Szenario 2 korrekt.

Das würde mir sehr gelegen kommen :)

premultiply, 2024-01-04:
Du willst also nur noch intern estimieren?

andig, 2024-01-13:
OT: wollten wir das nicht schon länger?

andig, 2024-01-13:
@naltatis gibts hier noch was zu tun?

naltatis, 2024-01-13:
Einziges Todo das ich sehe wäre der Umbau der Restzeitberechnung. Aus UI Sicht wäre das transparent. Gerne als separates Issue.

andig, 2024-02-03:
Kein weiteres Todo, closing.

closed 2024-02-03 by andig