evcc-io/evcc #11428: Einstellung Batteriesteuerung (Ein/Aus) wird nicht gespeichert

StevieC121176, 2024-01-02:
### Describe the bug

Die Batteriesteuerung (Ein/Aus) wird nicht gespeichert. Aktualisiere ich beispielsweise das Browserfenster oder gehe ich in das Ladelog und wieder zurück zu den Batterieeinstellungen, ist die Batteriesteuerung immer wieder auf "Aus" gestellt.
Ich habe es mit dem Safariebrowser unter iOS und MacOS sowie dem Chromebrowser probiert

### Steps to reproduce

...

### Configuration details

```yaml
meters:
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.118  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv1
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.118  
  port: 1502  
  usage: battery  
  modbus: tcpip
  capacity: 12.8 
  minsoc: 10  
  maxsoc: 100  
  watchdog: 60s 
  name: battery2
- type: template
  template: kostal-ksem 
  id: 71  
  host: 192.168.178.116  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid3
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.117  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv4
```

</details>
Version: `0.123.3 (9751f2b2)`
```

### Log details

```text
.
```

### What type of operating system are you running?

Linux

### Version

0.123.3 (9751f2b2)

andig, 2024-01-02:
@naltatis kann ich bestätigen.

[image]

andig, 2024-01-02:
Warum verstehe ich nicht- `batteryDischargeControl` wird Site -> Energyflow -> BatterySettingsModal durchgereicht.

PR #11441 (2024-01-03): Restore: publishing initial site values
fixes #11438
fixes #11428
fixes #11423

Revert from https://github.com/evcc-io/evcc/commit/f14b83cd8ba63d772f71cca6d1adaf195ac68248#diff-7aa5baae0bd3243f85bfc6db64612f7450eb0170fa310ca2c93dc3968fc02e02
closed 2024-01-03 by andig