evcc-io/evcc #11431: Easee: panic in ProductUpdate processing

GrimmiMeloni, 2024-01-02:
Seit Version 0.123.2 wird beim Laden weder das Auto (Tesla) erkannt noch der Ladepunkt angezeigt (Ladeleistung permanent auf 0.
Das Auto wird zwar geladen, die Anzeigen in der TeslaApp als auch auf der EaseeApp sind OK.
Ein vorhergehender Check der Installation über:
sudo systemctl status evcc
sudo evcc -l debug meter
sudo evcc -l debug charger
sudo evcc -l debug vehicle
brachte keine Fehler, alles wurde korrekt erkannt.
Nach dem Downgrade von 0.123.4 auf 0.123.1 (im log ab 17:09 zusehen) war wieder alles wie sonst.
die Config und das Log hab ich mal im Windowsformat angehängt.
[config.txt](https://github.com/evcc-io/evcc/files/13812030/config.txt)
[NoLoadpoint.txt](https://github.com/evcc-io/evcc/files/13812031/NoLoadpoint.txt)
Ich habe keine Idee mehr was ich noch ausprobieren könnte

_Originally posted by @JHammerschmid in https://github.com/evcc-io/evcc/discussions/11427_

GrimmiMeloni, 2024-01-02:
Confirmed locally. Testing a fix locally right now....

andig, 2024-01-02:
> Ist es korrekt, daß once innerhalb der Methode definiert ist? Muß die Referenz nicht in einen globalen Scope?

Woah, ja!

andig, 2024-01-02:
Dafür darf der zu: 7819662ad1dff4bdaadb4be23dd6f6fb67ee7408

closed 2024-01-02 by andig