evcc-io/evcc #11454: BatteryDischargeControl mit Sofar HYD 15 via Sofar ModBus TCP LAN Stick API-Fehler

sahomm, 2024-01-03:
### Describe the bug

Bei dem Setzen des Parameter BatteryDischargeConect wird ein API-Fehler in der GUI ausgegeben. Gemäß Log wurde dieser jedoch gesetzt.

Genutzt wird ein Sofar HYD 15 KTL 3PH, angebunden ist dieser via Modus TCP mittels des Sofar Ethernetdongle.

[image]

### Steps to reproduce

1. Parameter Batteriesteuerung setzen
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: xxx

interval: 10s # control cycle interval

meters:
  - name: my_grid
    type: template
    template: sofarsolar-g3
    usage: grid

    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.181.135 # Hostname
    port: 8899 # Port
#    delay: 10s # optional 

  - name: my_pv
    type: template
    template: sofarsolar-g3
    usage: pv
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.181.135 # Hostname
    port: 8899 # Port
#    delay: 10s # optional 

  - name: my_battery
    type: template
    template: sofarsolar-g3
    usage: battery

    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.181.135 # Hostname
    port: 8899 # Port
#    delay: 10s # optional 

chargers:
[... 172 lines cut ...]
[site  ] DEBUG 2024/01/03 18:41:53 grid powers: [210 200 200]W
[site  ] DEBUG 2024/01/03 18:41:54 grid currents: [0.88 0.84 0.85]A
[site  ] DEBUG 2024/01/03 18:41:54 site power: 610W
[lp-1  ] DEBUG 2024/01/03 18:41:54 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/01/03 18:41:54 charger status: A
[site  ] DEBUG 2024/01/03 18:42:01 ----
[lp-1  ] DEBUG 2024/01/03 18:42:01 charge power: 0W
[site  ] DEBUG 2024/01/03 18:42:01 pv power: 0W
[site  ] DEBUG 2024/01/03 18:42:02 battery soc: 60%
[site  ] DEBUG 2024/01/03 18:42:02 battery power: 0W
[site  ] DEBUG 2024/01/03 18:42:02 grid power: 610W
[site  ] DEBUG 2024/01/03 18:42:03 grid powers: [210 200 200]W
[site  ] DEBUG 2024/01/03 18:42:04 grid currents: [0.88 0.84 0.85]A
[site  ] DEBUG 2024/01/03 18:42:04 site power: 610W
[lp-1  ] DEBUG 2024/01/03 18:42:04 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/01/03 18:42:04 charger status: A
```

### What type of operating system are you running?

Linux

### Version

0.123.5

andig, 2024-01-03:
Probiers mal mit curl. Was steht in der Antwort?

sahomm, 2024-01-04:
nach:  curl -X POST -v http://localhost:7070/api/batterydischargecontrol/true
*   Trying ::1:7070...
* connect to ::1 port 7070 failed: Verbindungsaufbau abgelehnt
*   Trying 127.0.0.1:7070...
* connect to 127.0.0.1 port 7070 failed: Verbindungsaufbau abgelehnt
* Failed to connect to localhost port 7070: Verbindungsaufbau abgelehnt
* Closing connection 0
curl: (7) Failed to connect to localhost port 7070: Verbindungsaufbau abgelehnt

andig, 2024-01-04:
Das ergibt ja keinen Sinn. Dein Screenshot zeigt, dass der Request geht.

sahomm, 2024-01-04:
ich war mir mir nicht sicher ob der Request an den WR zur Steuerung gehen soll, sry.

> Dein Screenshot zeigt, dass der Request geht

Stimmt, dies sagt ja auch das Log, hier ein Auszug:
```
[lp-1  ] DEBUG 2024/01/04 10:05:19 charge power: 0W
[site  ] DEBUG 2024/01/04 10:05:20 pv power: 300W
[site  ] DEBUG 2024/01/04 10:05:21 battery soc: 54%
[site  ] DEBUG 2024/01/04 10:05:21 battery power: 0W
[site  ] DEBUG 2024/01/04 10:05:21 grid power: 480W
[site  ] DEBUG 2024/01/04 10:05:22 grid powers: [160 160 160]W
[site  ] DEBUG 2024/01/04 10:05:22 grid currents: [0.66 0.62 0.71]A
[site  ] DEBUG 2024/01/04 10:05:23 site power: 480W
[lp-1  ] DEBUG 2024/01/04 10:05:23 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/01/04 10:05:23 charger status: A
**[site  ] DEBUG 2024/01/04 10:05:24 set battery discharge control: false**
[site  ] DEBUG 2024/01/04 10:05:29 ----
[lp-1  ] DEBUG 2024/01/04 10:05:29 charge power: 0W
[site  ] DEBUG 2024/01/04 10:05:30 pv power: 300W
[site  ] DEBUG 2024/01/04 10:05:31 battery soc: 54%
[site  ] DEBUG 2024/01/04 10:05:31 battery power: 0W
[site  ] DEBUG 2024/01/04 10:05:31 grid power: 480W
[site  ] DEBUG 2024/01/04 10:05:32 grid powers: [160 160 160]W
[site  ] DEBUG 2024/01/04 10:05:33 grid currents: [0.66 0.66 0.66]A
[site  ] DEBUG 2024/01/04 10:05:33 site power: 480W
[lp-1  ] DEBUG 2024/01/04 10:05:33 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/01/04 10:05:33 charger status: A
[site  ] DEBUG 2024/01/04 10:05:39 ----
[lp-1  ] DEBUG 2024/01/04 10:05:39 charge power: 0W
[site  ] DEBUG 2024/01/04 10:05:40 pv power: 300W
[site  ] DEBUG 2024/01/04 10:05:41 battery soc: 54%
[site  ] DEBUG 2024/01/04 10:05:41 battery power: 0W
[site  ] DEBUG 2024/01/04 10:05:41 grid power: 490W
[site  ] DEBUG 2024/01/04 10:05:42 grid powers: [170 160 160]W
**[site  ] DEBUG 2024/01/04 10:05:42 set battery discharge control: true**
[site  ] DEBUG 2024/01/04 10:05:42 grid currents: [0.7 0.66 0.71]A
[site  ] DEBUG 2024/01/04 10:05:43 site power: 490W
[lp-1  ] DEBUG 2024/01/04 10:05:43 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/01/04 10:05:43 charger status: A
[site  ] DEBUG 2024/01/04 10:05:49 ----
[lp-1  ] DEBUG 2024/01/04 10:05:49 charge power: 0W
[site  ] DEBUG 2024/01/04 10:05:57 pv power: 300W
[site  ] DEBUG 2024/01/04 10:05:57 battery soc: 54%
[site  ] DEBUG 2024/01/04 10:05:57 battery power: 0W
[site  ] DEBUG 2024/01/04 10:05:58 grid power: 470W
[site  ] DEBUG 2024/01/04 10:05:59 grid powers: [150 160 160]W
[site  ] DEBUG 2024/01/04 10:06:00 grid currents: [0.62 0.66 0.66]A
[site  ] DEBUG 2024/01/04 10:06:00 site power: 470W
[lp-1  ] DEBUG 2024/01/04 10:06:00 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/01/04 10:06:00 charger status: A
[site  ] DEBUG 2024/01/04 10:06:00 ----
```

Nur warum kommt die Meldung gem. Screenshot?

andig, 2024-01-04:
Mach doch bitte mal den Request wie angefragt.

sahomm, 2024-01-04:
ich dachte diesen Request meinst du:

> curl -X POST -v http://localhost:7070/api/batterydischargecontrol/true
> Trying ::1:7070...
> connect to ::1 port 7070 failed: Verbindungsaufbau abgelehnt
> Trying 127.0.0.1:7070...
> connect to 127.0.0.1 [...]

oder welchen curl Request meinst du?

andig, 2024-01-04:
Genau den, aber bitte auf evcc. Anscheinend läuft das nicht auf localhost, sonst gäbe es eine Antwort.

sahomm, 2024-01-04:
merkwürdig, ich hatte diesen gestern auf der Konsole des EVCC gemacht.
Aber du liegest genau richtig, nach heutigem Versuch kommt folgende Ausgabe:
```
curl -X POST -v http://localhost:7070/api/batterydischargecont
rol/true
*   Trying ::1:7070...
* Connected to localhost (::1) port 7070 (#0)
> POST /api/batterydischargecontrol/true HTTP/1.1
> Host: localhost:7070
> User-Agent: curl/7.74.0
> Accept: */*
> 
* Mark bundle as not supporting multiuse
< HTTP/1.1 400 Bad Request
< Content-Type: application/json; charset=UTF-8
< Vary: Accept-Encoding
< Date: Thu, 04 Jan 2024 10:07:44 GMT
< Content-Length: 77
< 
{"error":"dod: modbus: response data size '17' does not match expected '4'"}
* Connection #0 to host localhost left intact
```

merkwürdigerweise funktioniert FALSE:
```
curl -X POST -v http://localhost:7070/api/batterydischargecont
rol/false
*   Trying ::1:7070...
* Connected to localhost (::1) port 7070 (#0)
> POST /api/batterydischargecontrol/false HTTP/1.1
> Host: localhost:7070
> User-Agent: curl/7.74.0
> Accept: */*
> 
* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< Content-Type: application/json; charset=UTF-8
< Vary: Accept-Encoding
< Date: Thu, 04 Jan 2024 10:10:58 GMT
< Content-Length: 17
< 
{"result":false}
* Connection #0 to host localhost left intact
```

andig, 2024-01-04:
Die Umschaltung funktioniert nicht. Bitte mal auf der Kommandozeile:

    evcc meter --log error,modbus:trace --battery-mode hold

durchführen.

@naltatis das ist wieder ein Fall, wo wir aus dem UI einen Request machen der einen Fehler liefert, den Fehler aber nirgendwo anzeigen. Könnten wir das error Result- wenn es eines gibt- zumindest mit in die Logmeldung aufnehmen, auch wenns ein technischer Fehler ist? Wir sind sonst blind...

sahomm, 2024-01-04:
Ich habe -c zum Verweis auf die evcc.yaml ergänzt. Hier der Result:

```
evcc -c /etc/evcc.yaml meter --log error,modbus:trace --batter
y-mode hold
[main  ] INFO 2024/01/04 12:42:42 evcc 0.123.5 (7819662a)
[main  ] INFO 2024/01/04 12:42:42 using config file: /etc/evcc.yaml
[modbus] TRACE 2024/01/04 12:42:42 modbus: send 00 01 00 00 00 06 01 03 06 08 00 01
[modbus] TRACE 2024/01/04 12:42:43 modbus: recv 00 01 00 00 00 05 01 03 02 00 35
[modbus] TRACE 2024/01/04 12:42:43 modbus: send 00 02 00 00 00 09 01 10 10 4d 00 01 02 12 5c
[modbus] TRACE 2024/01/04 12:42:43 modbus: recv 00 02 00 00 00 13 01 10 10 4d 00 01 95 1e 00 00 00 00 00 00 00 00 00 00 00
[main  ] FATAL 2024/01/04 12:42:43 set battery mode: dod: modbus: response data size '17' does not match expected '4'
```

andig, 2024-01-04:
It seems writing `02 12`(530) to DOD (`10 4d`) does not work. Seems we should remove the `scale: 100` from the template.

@PeterPablo original request was https://github.com/evcc-io/evcc/issues/10379#issuecomment-1805276942. Does this analysis make sense?

@sahomm could you give remote access to WR (port forwarting) to info@evcc.io so we can test this?

andig, 2024-01-04:
Leider klappt auch das nicht:

```
evcc meter --log error,modbus:trace -c battery.yaml -b normal -w 30s

MinSoc: 20
[modbus] TRACE 2024/01/04 14:28:22 modbus: send 00 01 00 00 00 09 01 10 10 4d 00 01 02 00 50
[modbus] TRACE 2024/01/04 14:28:22 modbus: recv 00 01 00 00 00 13 01 10 10 4d 00 01 95 1e 00 00 00 00 00 00 00 00 00 00 00
[main  ] FATAL 2024/01/04 14:28:22 set battery mode: dod: modbus: response data size '17' does not match expected '4'
```

Minsoc 20% entspricht DOD 80%, also 0x50. Leider wird auch das vom WR abgelehnt. Jetzt bin ich ratlos.

sahomm, 2024-01-04:
In folgendem Thread aus dem PV-Forum wurde davon gesprochen, dass beim Schreiben auf den Sofar die Daten in einem bestimmten Frame gemeinsam geschrieben werden müssen.

https://www.photovoltaikforum.com/thread/217362-sofar-hyd-serie-kommunikation-%C3%BCber-modbus-tcp-erfahrungen/?pageNo=4￼
SOFAR HYD Serie - Kommunikation über Modbus TCP - Erfahrungen - Photovoltaikforum
photovoltaikforum.com

Kann es evtl damit zusammenhängen?

andig, 2024-01-04:
Wen dem so wäre- mit den Parametern kann die komplette Batteriekonfiguration geändert werden- wäre mit das zu heiss und eignet sich ohnehin nicht für ein Template. Dann müsste die Umsetzung leider wieder raus bis es einen besseren Ansatz gibt.

sahomm, 2024-01-04:
das verstehe ich. Leider bin ich hier wenig Hilfe. Solle jedoch bedarf bestehen weiter zum Test auf den WR zugreifen zu wollen, bitte melden.

VIELN DANK FÜR DEINE MÜHE!!!!

Beste Grüße

Sascha

closed 2024-01-04 by andig

PeterPablo, 2024-01-05:
Ich hatte damals nur die Register anhand der Dokumentation herausgesucht. Bezüglich Umsetzung mit MODBUS und dessen Eigenheiten habe ich bisher leider keine Erfahrungen. Der SofarSolar-Support hatte mir ja nach einigen Tagen die Dokumentation zur Verfügung gestellt. Ich würde dort mal nachfragen ob sie einen Tipp haben.

sahomm, 2024-01-05:
@PeterPablo vielen Dank für deine Mühe! Solltest du ein Sofar HYD 15 zum testen benötigen, laß es mich wissen, ich gebe dir gerne meinen frei.

Wie oben geschrieben, hat es im PV-Forum scheinbar bereits Erfolge gegeben. Ggf. kannst du/ihr von den Erfahrungen partizipieren oder laß(t) es mich wissen wie ich dich ggf. unterstützen kann.

andig, 2024-01-05:
Wir haben das jetzt wieder ausgebaut da leider nur der ganze Batterieblock geschrieben werden kann.