evcc-io/evcc #11455: Battery Discharge Slider SolaX

Stepsilein, 2024-01-03:
### Describe the bug

Nachdem gestern der Slider zumindest "optisch" in Funktion blieb, hat die Änderung in Evcc dann in Folge die SolaX Modbus Integration nachhaltig zerschossen.
Folge: Keine SolaX Entitäten bis auf 2 verfügbar.
Nach viek Gefrickel, de jnd neu Installation, jnzähligen HA und System Neustarts, läufg nun Alles wieder, aber der Schalter betr. Akku blocken springt immer wieder nach raus auf Dasboard und zurück zum Schalter, auf Aus.

Da ich diese Funktion zu der ich betr. deren Zugriff auf welche ??? Entität vln SolaX leider keinerlei Doku/Info finde, gerne wissen möchte um einzuschätzen, ob sinnvoll/gut, hier dazu meine Frage:
Welcher SolaX Wert wird genutzt?
Ich denke leider nicht der "Battery Discharge max. Current" Wert (Ampere), oder?
Bis dazu Klarheit betr. Wert per Info kommt, kann ich nur warnen und rühre den Schalter nicht mehr an.
Da ist mir mein eigener dazu erstellter Schalter betr. Battery Discharge dann sicherer und funktioniert auch ohne Probleme mit der SolaX Modbus Integration.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
Neuste 5er Evcc
HA neustes System
evcc.yaml an selber Stelle wie immer/früher
```

### Log details

```text
-
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

Latrst from today

VolkerK62, 2024-01-03:
>Welcher SolaX Wert wird genutzt?

guckst du hier: https://github.com/evcc-io/evcc/commit/840f33ade57da6a04374c02968a86776351c387c

Stepsilein, 2024-01-03:
Danke für den Link.
Soweit ich das laienhaft verstehe, wird mit dem "Use Mode", also auf Backup ? umgestellt oder Ähnliches.
Sehe da leider nicht klar durch, bin da zu unwisends betr. Programmierung usw.
Frage erweitert:
1.
Es ist also nicht der Battery Discharge max Current Wert der genutzt wird?
2.
Warum nicht?
Der Mide in SolaX blockt doch je nach Einstellung kaum bis Nichts, je min. SLC Einstellung 🤔.

Nicht nur ich wäre sehr dankbar für eine Klartext Info dazu.
Man kommt doch an die vielen SolaX Entitäten nur via dem SolaX Modbus Addon.
Läuft das nicht, hat Evcc keine SolaX Werte und zeigt oben nur: "keine Daten".
Ergo dürften von Evcc genutzte Werte auch die SolaX Bezeichnungen haben, oder?

Wenn aber "nur" der Backup Mode als Batterie Blocker in Evcc genutzt wird, würde es mir teilweise ein Wenig erklären, warum der Schalter dann wiederum nur km Evcc Schnell Modus greift, statt generell.
Fakt ist, ich möchte wie Viele, nie und unter keinen Umständen, das Auto aus dem Hausakku laden.
Dazu fiel mir bisher nur ein, den SolaX Battery Discharge Wert auf 0 A zu setzen.
Das funktioniert 100%.
Leider jeweils ein Schreibvorgang auf den SolaX Eprom und ein 2. beim wieder auf 25A hoch setzten nach dem Laden.

Gibt es dazu Bedenken, oder warum geht man nicht diesen recht einfachen, aber 100% blockenden Weg?

Liebe Grüße

andig, 2024-01-03:
Details fehlen

closed 2024-01-03 by andig