evcc-io/evcc #11458: Cool wäre wenn ihr die dynamischen Strompreise von Smart Energy implementieren könntet! https://www.smartenergy.at/

AlexGabl, 2024-01-03:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

andig, 2024-01-04:
Einfacher gehts nicht- gerne PR.

andig, 2024-01-04:
@AlexGabl sieht aus als würde das API einfach die SPOT-Preise weiter geben? Falls ja kannst Du das jetzt schon mit EntsoE machen. Oder gibts hier einen Mehrwert?

AlexGabl, 2024-01-05:
Also anstatt:
 grid:
    type: awattar

 grid:
    type: EntsoE

??? Danke!

andig, 2024-01-06:
Mit entsprechender Konfiguration, genau. Siehe https://github.com/evcc-io/evcc/blob/master/evcc.dist.yaml

closed 2024-01-06 by andig