evcc-io/evcc #11466: Battery charge graphic broken

TomS2, 2024-01-04:
### Describe the bug

When showing the current battery charge lavel the graphic showing the level is broken:
[image]

### Steps to reproduce

1. Show energy statistics information in main screen
2. Click battery percentage
3. See broken graphical representation of battery SoC

### Configuration details

```yaml
# open evcc at http://evcc.local:7070  
network:  
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS  
  port: 7070  
  
log: debug  
levels:  
  cache: error  
  
# unique installation id (64 characters from sponsortoken)
plant: *******************************************************************
  
interval: 10s # control cycle interval  

sponsortoken: ************************************************************************************************************************************************************************
meters:  
# use enphase envoy's grid values
- name: envoy_grid
  type: template
  template: enphase
  usage: grid
  host: 192.168.10.20 # IP-Adresse oder Hostname
  token: ****************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************
  
# use enphase envoy's pv values
- name: envoy_pv
  type: template
  template: enphase
  usage: pv
  host: 192.168.10.20 # IP-Adresse oder Hostname
  token: ****************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************
  
# use enphase envoy's battery values
- name: envoy_bat
  type: template
  template: enphase
  usage: battery
  capacity: 10.5 # Akkukapazitaet in kWh (Optional)
  host: 192.168.10.20 # IP-Adresse oder Hostname
  token: ****************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************
  
chargers:
- name: Master
  type: template
  template: easee
  user: ***************
  password: *******************
  charger: EH962043 
      
- name: Slave
  type: template
  template: easee
  user: ***************
  password: *******************
  charger: EH603852 
      
vehicles:  
- type: template  
  template: hyundai   
  title: IONIQ5    
  identifiers: 
  - ***************
  user: *****************  
  password: **********  
  vin: *****************  
  capacity: 72.4    
  language: de    
  name: ioniq5
  
- type: template
  template: peugeot
  title: Peugeot-208e
  identifiers:
  - **************
  user: ****************************
  password: ***********
  capacity: 50  
  name: 208e
  
loadpoints:
- title: Privat
  charger: Master
#  vehicle: ioniq5
  mode: pv
  mincurrent: 10
  
- title: Gast
  charger: Slave
#  vehicle: 208e
  mode: pv
  mincurrent: 10
  
site:  
  title: Mein Zuhause  
  meters:  
    grid: envoy_grid
    pv: envoy_pv
    battery: envoy_bat
```

### Log details

```text
Does not work in docker container ;-)
```

### What type of operating system are you running?

Docker container

### Version

evcc version 0.123.5

StevieC121176, 2024-01-04:
sh #11423

TomS2, 2024-01-04:
Thanks for directing me ...

closed 2024-01-04 by TomS2