evcc-io/evcc #11475: Fenecon  PV Konfiguration

kaimaik, 2024-01-04:
### Describe the bug

Hallo zusammen,
ich hab ein Problem mit dem Fenecon  und der Konfiguration.

Es sieht so aus als ob die Komunikation nicht über den Wechselrichter sondern über die Batterie läuft

Kann da jemnad helfen ?

### Steps to reproduce

evcc configure
> Advanced mode (Ask more details, requires technical know-how)
> Configure a single device (has to be added manually to a configuration file!)
> PV inverter (or corresponding meter)
> FENECON
Please provide the following settings:

 IP address or hostname [? for help] 192.168.178.31
Password [? for help] ****

### Configuration details

```yaml
In which mode should this process guide you through the configuration? Standard mode (As simple and quick as possible)

- Setup meters (house installation)

Choose one of the following integrated PV systems, or 'My device is not in this list' if you have none of them

 Choose an integrated PV system: FENECON

Please provide the following settings:

 IP address or hostname 192.168.178.31

Testing the grid configuration of FENECON ...

Testing the pv configuration of FENECON ...
   Error: strconv.ParseFloat: parsing "<nil>": invalid syntax

 Testing of the pv configuration of FENECON failed. Do you want to add pv anyway? (y/N) 
 Testing of the pv configuration of FENECON failed. Do you want to add pv anyway? No

Testing the battery configuration of FENECON ...

FENECON was successfully added.

 Do you want to add a PV inverter (or corresponding meter)? No

 Do you want to add another battery inverter (or corresponding meter)? Yes

- Configuration battery inverter (or corresponding meter)

 Choose a battery inverter (or corresponding meter): FENECON

Please provide the following settings:

 IP address or hostname 192.168.178.31

Testing the battery configuration of FENECON ...

FENECON FENECON was successfully added.

 Do you want to add another battery inverter (or corresponding meter)? (y/N)
```

### Log details

```text
- Configuration PV inverter (or corresponding meter)

 Choose a PV inverter (or corresponding meter): FENECON

Please provide the following settings:

 IP address or hostname 192.168.178.31
 Password [? for help] ****

Testing the pv configuration of FENECON ...
   Error: strconv.ParseFloat: parsing "<nil>": invalid syntax
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.123.5

andig, 2024-01-04:
@kaimaik bitte Ergebnis von

    evcc meter --log trace

ergänzen.

kaimaik, 2024-01-04:
kaimaik@raspberrypi:~ $ evcc meter --log trace
[main  ] INFO 2024/01/04 12:21:12 evcc 0.123.5
[main  ] INFO 2024/01/04 12:21:12 using config file: /home/kaimaik/evcc.yaml
[db    ] INFO 2024/01/04 12:21:13 using sqlite database: /home/kaimaik/.evcc/evcc.db
[db    ] TRACE 2024/01/04 12:21:13 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/01/04 12:21:13 CREATE TABLE `settings` (`key` text,`value` text,PRIMARY KEY (`key`)) 0 <nil>
[db    ] TRACE 2024/01/04 12:21:13 SELECT * FROM `settings` 0 <nil>
[db    ] TRACE 2024/01/04 12:21:13 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/01/04 12:21:13 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/01/04 12:21:13 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/01/04 12:21:13 CREATE TABLE `configs` (`id` integer PRIMARY KEY AUTOINCREMENT,`class` integer,`type` text) 0 <nil>
[db    ] TRACE 2024/01/04 12:21:13 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/01/04 12:21:13 CREATE TABLE `config_details` (`config_id` integer,`key` text,`value` text,CONSTRAINT `fk_configs_details` FOREIGN KEY (`config_id`) REFERENCES `configs`(`id`) ON DELETE CASCADE ON UPDATE CASCADE) 0 <nil>
[db    ] TRACE 2024/01/04 12:21:13 CREATE INDEX `idx_unique` ON `config_details`(`config_id`,`key`) 0 <nil>
[db    ] TRACE 2024/01/04 12:21:13 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_devices_details\" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT fk_devices_details    %") -1 <nil>
[db    ] TRACE 2024/01/04 12:21:13 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%\"device_id\" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "% device_id    %") -1 <nil>
[db    ] TRACE 2024/01/04 12:21:13 SELECT * FROM `configs` WHERE `configs`.`class` = 2 ORDER BY id 0 <nil>
grid1
-----
[http  ] TRACE 2024/01/04 12:21:13 GET http://192.168.178.31/rest/channel/_sum/GridActivePower
[http  ] TRACE 2024/01/04 12:21:13 {"address":"_sum/GridActivePower","type":"INTEGER","accessMode":"RO","text":"Grid exchange power. Negative values for sell-to-grid; positive for buy-from-grid","unit":"W","value":746}
Power: 746W

pv2
---
[http  ] TRACE 2024/01/04 12:21:13 GET http://192.168.178.31/rest/channel/_sum/ProductionActivePower
[http  ] TRACE 2024/01/04 12:21:13 {"address":"_sum/ProductionActivePower","type":"INTEGER","accessMode":"RO","text":"Total production; always positive","unit":"W","value":null}
Power: strconv.ParseFloat: parsing "<nil>": invalid syntax

battery3
--------
[http  ] TRACE 2024/01/04 12:21:13 GET http://192.168.178.31/rest/channel/_sum/EssDischargePower
[http  ] TRACE 2024/01/04 12:21:13 {"address":"_sum/EssDischargePower","type":"INTEGER","accessMode":"RO","text":"Actual AC-side battery discharge power of Energy Storage System. Negative values for charge; positive for discharge","unit":"W","value":-19}
[http  ] TRACE 2024/01/04 12:21:13 GET http://192.168.178.31/rest/channel/_sum/EssSoc
[http  ] TRACE 2024/01/04 12:21:13 {"address":"_sum/EssSoc","type":"INTEGER","accessMode":"RO","text":"Range 0..100","unit":"%","value":0}
Power: -19W
Soc:   0%

battery4
--------
[http  ] TRACE 2024/01/04 12:21:13 GET http://192.168.178.31/rest/channel/_sum/EssDischargePower
[http  ] TRACE 2024/01/04 12:21:13 {"address":"_sum/EssDischargePower","type":"INTEGER","accessMode":"RO","text":"Actual AC-side battery discharge power of Energy Storage System. Negative values for charge; positive for discharge","unit":"W","value":-19}
[http  ] TRACE 2024/01/04 12:21:13 GET http://192.168.178.31/rest/channel/_sum/EssSoc
[http  ] TRACE 2024/01/04 12:21:13 {"address":"_sum/EssSoc","type":"INTEGER","accessMode":"RO","text":"Range 0..100","unit":"%","value":0}
Power:    -19W
Soc:      0%
Capacity: 9.0kWh

charge6
-------
[modbus] TRACE 2024/01/04 12:21:13 modbus: send 01 04 00 34 00 02 30 05
[modbus] TRACE 2024/01/04 12:21:18 modbus: recv
[modbus] TRACE 2024/01/04 12:21:18 modbus: send 01 04 00 48 00 02 f1 dd
[modbus] TRACE 2024/01/04 12:21:23 modbus: recv
[modbus] TRACE 2024/01/04 12:21:23 modbus: send 01 04 00 06 00 02 91 ca
[modbus] TRACE 2024/01/04 12:21:29 modbus: recv
[modbus] TRACE 2024/01/04 12:21:29 modbus: send 01 04 00 00 00 02 71 cb
[modbus] TRACE 2024/01/04 12:21:34 modbus: recv
Power:          read failed: serial: timeout
Energy:         read failed: serial: timeout
Current L1..L3: read failed: serial: timeout
Voltage L1..L3: read failed: serial: timeout ###

kaimaik, 2024-01-04:
kaimaik@raspberrypi:~ $ evcc meter --log trace
[main  ] INFO 2024/01/04 12:21:12 evcc 0.123.5
[main  ] INFO 2024/01/04 12:21:12 using config file: /home/kaimaik/evcc.yaml
[db    ] INFO 2024/01/04 12:21:13 using sqlite database: /home/kaimaik/.evcc/evcc.db
[db    ] TRACE 2024/01/04 12:21:13 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/01/04 12:21:13 CREATE TABLE `settings` (`key` text,`value` text,PRIMARY KEY (`key`)) 0 <nil>
[db    ] TRACE 2024/01/04 12:21:13 SELECT * FROM `settings` 0 <nil>
[db    ] TRACE 2024/01/04 12:21:13 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/01/04 12:21:13 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/01/04 12:21:13 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/01/04 12:21:13 CREATE TABLE `configs` (`id` integer PRIMARY KEY AUTOINCREMENT,`class` integer,`type` text) 0 <nil>
[db    ] TRACE 2024/01/04 12:21:13 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/01/04 12:21:13 CREATE TABLE `config_details` (`config_id` integer,`key` text,`value` text,CONSTRAINT `fk_configs_details` FOREIGN KEY (`config_id`) REFERENCES `configs`(`id`) ON DELETE CASCADE ON UPDATE CASCADE) 0 <nil>
[db    ] TRACE 2024/01/04 12:21:13 CREATE INDEX `idx_unique` ON `config_details`(`config_id`,`key`) 0 <nil>
[db    ] TRACE 2024/01/04 12:21:13 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT \"fk_devices_details\" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT fk_devices_details    %") -1 <nil>
[db    ] TRACE 2024/01/04 12:21:13 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%\"device_id\" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "% device_id    %") -1 <nil>
[db    ] TRACE 2024/01/04 12:21:13 SELECT * FROM `configs` WHERE `configs`.`class` = 2 ORDER BY id 0 <nil>
grid1
-----
[http  ] TRACE 2024/01/04 12:21:13 GET http://192.168.178.31/rest/channel/_sum/GridActivePower
[http  ] TRACE 2024/01/04 12:21:13 {"address":"_sum/GridActivePower","type":"INTEGER","accessMode":"RO","text":"Grid exchange power. Negative values for sell-to-grid; positive for buy-from-grid","unit":"W","value":746}
Power: 746W

pv2
---
[http  ] TRACE 2024/01/04 12:21:13 GET http://192.168.178.31/rest/channel/_sum/ProductionActivePower
[http  ] TRACE 2024/01/04 12:21:13 {"address":"_sum/ProductionActivePower","type":"INTEGER","accessMode":"RO","text":"Total production; always positive","unit":"W","value":null}
Power: strconv.ParseFloat: parsing "<nil>": invalid syntax

battery3
--------
[http  ] TRACE 2024/01/04 12:21:13 GET http://192.168.178.31/rest/channel/_sum/EssDischargePower
[http  ] TRACE 2024/01/04 12:21:13 {"address":"_sum/EssDischargePower","type":"INTEGER","accessMode":"RO","text":"Actual AC-side battery discharge power of Energy Storage System. Negative values for charge; positive for discharge","unit":"W","value":-19}
[http  ] TRACE 2024/01/04 12:21:13 GET http://192.168.178.31/rest/channel/_sum/EssSoc
[http  ] TRACE 2024/01/04 12:21:13 {"address":"_sum/EssSoc","type":"INTEGER","accessMode":"RO","text":"Range 0..100","unit":"%","value":0}
Power: -19W
Soc:   0%

battery4
--------
[http  ] TRACE 2024/01/04 12:21:13 GET http://192.168.178.31/rest/channel/_sum/EssDischargePower
[http  ] TRACE 2024/01/04 12:21:13 {"address":"_sum/EssDischargePower","type":"INTEGER","accessMode":"RO","text":"Actual AC-side battery discharge power of Energy Storage System. Negative values for charge; positive for discharge","unit":"W","value":-19}
[http  ] TRACE 2024/01/04 12:21:13 GET http://192.168.178.31/rest/channel/_sum/EssSoc
[http  ] TRACE 2024/01/04 12:21:13 {"address":"_sum/EssSoc","type":"INTEGER","accessMode":"RO","text":"Range 0..100","unit":"%","value":0}
Power:    -19W
Soc:      0%
Capacity: 9.0kWh

charge6
-------
[modbus] TRACE 2024/01/04 12:21:13 modbus: send 01 04 00 34 00 02 30 05
[modbus] TRACE 2024/01/04 12:21:18 modbus: recv
[modbus] TRACE 2024/01/04 12:21:18 modbus: send 01 04 00 48 00 02 f1 dd
[modbus] TRACE 2024/01/04 12:21:23 modbus: recv
[modbus] TRACE 2024/01/04 12:21:23 modbus: send 01 04 00 06 00 02 91 ca
[modbus] TRACE 2024/01/04 12:21:29 modbus: recv
[modbus] TRACE 2024/01/04 12:21:29 modbus: send 01 04 00 00 00 02 71 cb
[modbus] TRACE 2024/01/04 12:21:34 modbus: recv
Power:          read failed: serial: timeout
Energy:         read failed: serial: timeout
Current L1..L3: read failed: serial: timeout
Voltage L1..L3: read failed: serial: timeout

andig, 2024-01-04:
> {"address":"_sum/ProductionActivePower","type":"INTEGER","accessMode":"RO","text":"Total production; always positive","unit":"W","value":null}

@premultiply können wir die Werte mittels JQ noch gegen `null` absichern?

premultiply, 2024-01-04:
Ja, mit sowas wie z. B. `(.value // 0)`

closed 2024-01-04 by andig

kaimaik, 2024-01-14:
Konfiguration läuft jetzt :-)

Vielen Dank euch

PR #11479 (2024-01-04): OpenEMS: null as 0
Fixes #11475