evcc-io/evcc #11477: 403: Out of call volume quota

box99h, 2024-01-04:
### Describe the bug

After approximately 30min charging time I'm receiving subsequent errors:

- Wallbox: vehicle range: 403: Out of call volume quota. Quota will be replenished in xx:xx:xx. (unexpected status: 403 (Forbidden)) 
- Wallbox: vehicle soc: 403: Out of call volume quota. Quota will be replenished in 00:39:41. (unexpected status: 403 (Forbidden)) (ignored by estimator)

evcc is running as docker in version 0.123.5.
Car is a BMW iX1### 

### Steps to reproduce

1. Charge for approx. 30min

### Configuration details

```yaml
# schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval

# log settings
log: trace
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error
  ocpp: trace
# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
  - name: my_grid
    type: template
    template: kostal-ksem
    usage: grid

    # Modbus TCP
    modbus: tcpip
    id: 71
    host: 10.10.2.133 # Hostname
    port: 502 # Port 

  - name: my_pv
    type: template
    template: kostal-plenticore
    usage: pv
    
    # Modbus TCP
    modbus: tcpip
    id: 71
    host: 10.10.2.218 # Hostname
    port: 1502 # Port 

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for documentation see https://docs.evcc.io/docs/devices/chargers
chargers:
   - type: template
     template: ocpp
     name: evbox
     stationid: EVB-Pxxxxxx
     connector: 1
     idtag: ‚xxxxxxxxxxxxxxxxxx‘
     connecttimeout: 5m
     timeout: 2m
     metervalues: Current.Import.L1,Current.Import.L2,Current.Import.L3,Energy.Active.Import.Register,Power.Active.Import,Voltage.L1,Voltage.L2,Voltage.L3
     getconfiguration: false

loadpoints:
- title: Wallbox
  charger: evbox
  mode: off
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  soc:
    poll:
      mode: charging
      interval: 15m
    estimate: true

# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for documentation see https://docs.evcc.io/docs/devices/vehicles
vehicles:
  - name: iX1
    type: template
    template: bmw
    user: xxxxxxxxxxx
    password: xxxxxxxxxx
    capacity: 65
    identifiers:
    - xxxxxxxxxxxxx
```

### Log details

```text
-
```

### What type of operating system are you running?

Docker container

VolkerK62, 2024-01-04:
same as #11424   
but `cache` can´t bei the reason here.

andig, 2024-01-04:
Missing logfile.

closed 2024-01-04 by andig

andig, 2024-01-08:
Ich sehe das bei meinem BMW jetzt auch- scheint also als hätte BMW die Quota deutlich reduziert :/