evcc-io/evcc #11478: Vehicle always shown as offline

davidgiga1993, 2024-01-04:
### Describe the bug

After updating to 0.123.2 (I was 4 versions behind before) the vehicle is always shown as `(Offline)` . Additionally the max current limit doesn't seem to work anymore/only sometimes.

There was no additional change, just update evcc.

I currently don't have access to the evcc host, will provide further information soon.

### Steps to reproduce

1. Plugin pmcc to vehicle 
2. Wait for vehicle to start charge (using minCharge profile)
3.Max charge limit not applied, vehicle still shown as offline even though evcc shows SoC
...

### Configuration details

```yaml
Will be provided
```

### Log details

```text
Will be provided
```

### What type of operating system are you running?

Linux

### Version

0.123.2

andig, 2024-01-04:
I guess you have minsoc and targetsoc as part of the vehicle config? Then logfile should show an error and you‘ll need to remove those.

davidgiga1993, 2024-01-04:
> I guess you have minsoc and targetsoc as part of the vehicle config? Then logfile should show an error and you‘ll need to remove those.

Yes I already migrated the yml file, evcc starts, just the described behavior appears

davidgiga1993, 2024-01-04:
Also I just received a `PorscheMobile: charger logic error: current mismatch (got -1A, expected 32A)`

davidgiga1993, 2024-01-04:
Just went back to the old yml and `0.121.5` and everything works fine. Thus this is a bug and should be handled as such..

davidgiga1993, 2024-01-04:
Did a couple of restarts of evcc and at some point if the vehicle is shown without the `(offline)` tag the max current gets set correctly with the version mentioned in the task. I'll take a look at the code when exactly this happens

andig, 2024-01-04:
Then lats get a trace log of the charger. -1A cant be correct.