evcc-io/evcc #11486: Easee: reset of SignalR connection on start of charge and car disconnect

GrimmiMeloni, 2024-01-04:
### Describe the bug

I have recently started to notice the occurrence of a new(?) SignalR method that is called by the Easee API. As a side effect the missing handler on evcc side leads to a disconnect/reconnect of the SignalR connection, which is not great.

In fact we might be missing out on some events due to this, as I have seen this happen reproducible during start of charge while a lot of ProductUpdates flow in.

### Steps to reproduce

1. Activate Trace Log for easee
2. connect vehicle
3. start a charge

### Configuration details

```yaml
nothing specific here. Just a working easee template.
```

### Log details

```text
[easee ] TRACE 2023/12/27 18:02:14 level=info connection=ccqKAuBBx7nrdtJmDynKlw event=getMethod error=missing method name=SubscribeToMyProduct reaction=send completion with error
[easee ] TRACE 2023/12/27 18:02:15 level=info connect=Connection closed with an error.
[easee ] TRACE 2023/12/27 18:02:15 POST https://streams.easee.com/hubs/chargers/negotiate
```

### What type of operating system are you running?

Docker container

### Version

0.123.5

andig, 2024-01-04:
What does the Easee doc/support have to say about this?

GrimmiMeloni, 2024-01-04:
There's no documentation on this to be found. I have asked our Easee contact if they can shed some light on this.

FWIW, I at least understand the flow we are observing. If our SignalR lib fails to find a suitable method for a received invocation, it replies with a Completion message incl. an Error. Based on my reading of the [spec](https://github.com/dotnet/aspnetcore/blob/main/src/SignalR/docs/specs/HubProtocol.md#errors), this should **not** lead to a disconnect - but that is also somewhat up for the hub (controlled by Easee) to decide how to handle. At least our client library I believe does not actively close the connection.

Either way, the PR I linked to earlier gets rid of the disconnect which is already an improvement. But we can of course let this soak for a moment and wait if Easee comes back with some answer.

PR #11487 (2024-01-05): Easee: add shims for unknown SignalR methods
Fix #11486 

I am somewhat curious to observe what Easee may (or may not) be sending our way, so I have added a Trace level log. I did the same for ChargerUpdate (which was already present as a NoOp). It does not hurt to have these, and stabilizing the connection is always a good thing.
closed 2024-01-05 by andig