evcc-io/evcc #11492: Easee: smart charging: timeout

g-rudi, 2024-01-05:
### Describe the bug

Hi 
Ich bekomme immer diese Meldung "WARN 2024/01/05 09:04:16 smart charging: timeout" im Log File und via UI.

Habe das mit dem normalen Build 0.123.5 und mit dem nightly build (9ae77a57). 

Gruss 
g-rudi

### Steps to reproduce

1. Start EVCC
2. Vehicle is not connected

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: <ID>

interval: 30s # control cycle interval

sponsortoken: <TOKEN>

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: solaredge-inverter 
  id: 1  
  host: 192.168.1.70  
  port: 1502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: solaredge-inverter 
  id: 1  
  host: 192.168.1.70  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv2

chargers:
- type: template
  template: easee 
  user: user 
  password: pw
  charger: EHM7CNNJ  
  timeout: 10s  
  name: wallbox4

vehicles:
- type: template
  template: bmw 
  title: iX1  
  icon: car  
  user: user
  password: pw
  vin: VIN
  capacity: 63  
  phases: 3  
  cache: 5m  
  mode: minpv  
  minCurrent: 6  
  maxCurrent: 16  
  priority: 1  
  region: EU  
  name: ev3

loadpoints:
- title: Garage
  charger: wallbox4
  mode: minpv
  phases: 0
  mincurrent: 6
  maxcurrent: 16

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pv:
```

### Log details

```text
[main  ] INFO 2024/01/05 09:03:51 evcc 0.123.5 (9ae77a57)
[main  ] INFO 2024/01/05 09:03:51 using config file: /evcccfg/evcc.yaml
[main  ] INFO 2024/01/05 09:03:51 starting ui and api at :7070
[db    ] INFO 2024/01/05 09:03:51 using sqlite database: /root/.evcc/evcc.db
[lp-1  ] DEBUG 2024/01/05 09:03:55 charge total import: 2257.932kWh
[site  ] INFO 2024/01/05 09:03:55 site config:
[site  ] INFO 2024/01/05 09:03:55   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2024/01/05 09:03:55     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/01/05 09:03:55     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/01/05 09:03:55   vehicles:
[site  ] INFO 2024/01/05 09:03:55     vehicle 1: range ✓ finish ✗ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2024/01/05 09:03:55 loadpoint 1:
[lp-1  ] INFO 2024/01/05 09:03:55   mode:        minpv
[lp-1  ] INFO 2024/01/05 09:03:55   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2024/01/05 09:03:55   meters:      charge ✓
[lp-1  ] INFO 2024/01/05 09:03:55     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2024/01/05 09:03:55 phase timer inactive
[lp-1  ] DEBUG 2024/01/05 09:03:55 pv timer inactive
[lp-1  ] DEBUG 2024/01/05 09:03:55 guard timer inactive
[site  ] DEBUG 2024/01/05 09:03:55 ----
[easee ] WARN 2024/01/05 09:04:05 smart charging: timeout
[lp-1  ] DEBUG 2024/01/05 09:04:05 charge power: 0W
[site  ] DEBUG 2024/01/05 09:04:06 pv power: 119W
[site  ] DEBUG 2024/01/05 09:04:06 grid power: 374W
[site  ] DEBUG 2024/01/05 09:04:06 site power: 374W
[lp-1  ] DEBUG 2024/01/05 09:04:06 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/01/05 09:04:06 charge total import: 2257.932kWh
[easee ] WARN 2024/01/05 09:04:16 smart charging: timeout
[lp-1  ] DEBUG 2024/01/05 09:04:16 charger status: A
[lp-1  ] INFO 2024/01/05 09:04:16 car disconnected
[lp-1  ] DEBUG 2024/01/05 09:04:16 set session soc limit: 0
[lp-1  ] DEBUG 2024/01/05 09:04:16 set session energy limit: 0
[site  ] DEBUG 2024/01/05 09:04:25 ----
[easee ] WARN 2024/01/05 09:04:36 smart charging: timeout
[lp-1  ] DEBUG 2024/01/05 09:04:36 charge power: 0W
[site  ] DEBUG 2024/01/05 09:04:36 pv power: 114W
[site  ] DEBUG 2024/01/05 09:04:36 grid power: 381W
[site  ] DEBUG 2024/01/05 09:04:36 site power: 381W
[lp-1  ] DEBUG 2024/01/05 09:04:36 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/01/05 09:04:36 charge total import: 2257.932kWh
[easee ] WARN 2024/01/05 09:04:46 smart charging: timeout
[lp-1  ] DEBUG 2024/01/05 09:04:46 charger status: A
[site  ] DEBUG 2024/01/05 09:04:55 ----
[easee ] WARN 2024/01/05 09:05:05 smart charging: timeout
[lp-1  ] DEBUG 2024/01/05 09:05:05 charge power: 0W
[site  ] DEBUG 2024/01/05 09:05:06 pv power: 99W
[site  ] DEBUG 2024/01/05 09:05:06 grid power: 382W
[site  ] DEBUG 2024/01/05 09:05:06 site power: 382W
[lp-1  ] DEBUG 2024/01/05 09:05:06 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/01/05 09:05:06 charge total import: 2257.932kWh
[easee ] WARN 2024/01/05 09:05:16 smart charging: timeout
[lp-1  ] DEBUG 2024/01/05 09:05:16 charger status: A
[site  ] DEBUG 2024/01/05 09:05:25 ----
[easee ] WARN 2024/01/05 09:05:35 smart charging: timeout
[lp-1  ] DEBUG 2024/01/05 09:05:35 charge power: 0W
[site  ] DEBUG 2024/01/05 09:05:36 pv power: 113W
[site  ] DEBUG 2024/01/05 09:05:36 grid power: 360W
[site  ] DEBUG 2024/01/05 09:05:36 site power: 360W
[lp-1  ] DEBUG 2024/01/05 09:05:36 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/01/05 09:05:36 charge total import: 2257.932kWh
[lp-1  ] DEBUG 2024/01/05 09:05:42 set charge mode: pv
[easee ] WARN 2024/01/05 09:05:46 smart charging: timeout
[lp-1  ] DEBUG 2024/01/05 09:05:46 charger status: A
[site  ] DEBUG 2024/01/05 09:05:46 ----
[easee ] WARN 2024/01/05 09:05:56 smart charging: timeout
[lp-1  ] DEBUG 2024/01/05 09:05:56 charge power: 0W
[site  ] DEBUG 2024/01/05 09:05:56 pv power: 101W
[site  ] DEBUG 2024/01/05 09:05:56 grid power: 345W
[site  ] DEBUG 2024/01/05 09:05:56 site power: 345W
[lp-1  ] DEBUG 2024/01/05 09:05:56 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/01/05 09:05:56 charge total import: 2257.932kWh
[easee ] WARN 2024/01/05 09:06:06 smart charging: timeout
[lp-1  ] DEBUG 2024/01/05 09:06:06 charger status: A
[site  ] DEBUG 2024/01/05 09:06:06 ----
[easee ] WARN 2024/01/05 09:06:17 smart charging: timeout
[lp-1  ] DEBUG 2024/01/05 09:06:17 charge power: 0W
[site  ] DEBUG 2024/01/05 09:06:17 pv power: 100W
[site  ] DEBUG 2024/01/05 09:06:17 grid power: 379W
[site  ] DEBUG 2024/01/05 09:06:17 site power: 379W
[lp-1  ] DEBUG 2024/01/05 09:06:17 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/01/05 09:06:17 charge total import: 2257.932kWh
[easee ] WARN 2024/01/05 09:06:27 smart charging: timeout
[lp-1  ] DEBUG 2024/01/05 09:06:27 charger status: A
[site  ] DEBUG 2024/01/05 09:06:27 ----
[easee ] WARN 2024/01/05 09:06:37 smart charging: timeout
[lp-1  ] DEBUG 2024/01/05 09:06:37 charge power: 0W
[site  ] DEBUG 2024/01/05 09:06:37 pv power: 104W
[site  ] DEBUG 2024/01/05 09:06:37 grid power: 347W
[site  ] DEBUG 2024/01/05 09:06:37 site power: 347W
[lp-1  ] DEBUG 2024/01/05 09:06:37 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/01/05 09:06:37 charge total import: 2257.932kWh
[easee ] WARN 2024/01/05 09:06:47 smart charging: timeout
[lp-1  ] DEBUG 2024/01/05 09:06:47 charger status: A
[site  ] DEBUG 2024/01/05 09:06:55 ----
[easee ] WARN 2024/01/05 09:07:06 smart charging: timeout
[lp-1  ] DEBUG 2024/01/05 09:07:06 charge power: 0W
[site  ] DEBUG 2024/01/05 09:07:06 pv power: 122W
[site  ] DEBUG 2024/01/05 09:07:06 grid power: 345W
[site  ] DEBUG 2024/01/05 09:07:06 site power: 345W
[lp-1  ] DEBUG 2024/01/05 09:07:06 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/01/05 09:07:06 charge total import: 2257.932kWh
[easee ] WARN 2024/01/05 09:07:16 smart charging: timeout
[lp-1  ] DEBUG 2024/01/05 09:07:16 charger status: A
[site  ] DEBUG 2024/01/05 09:07:25 ----
[easee ] WARN 2024/01/05 09:07:36 smart charging: timeout
[lp-1  ] DEBUG 2024/01/05 09:07:36 charge power: 0W
[site  ] DEBUG 2024/01/05 09:07:36 pv power: 128W
[site  ] DEBUG 2024/01/05 09:07:36 grid power: 399W
[site  ] DEBUG 2024/01/05 09:07:36 site power: 399W
[lp-1  ] DEBUG 2024/01/05 09:07:36 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/01/05 09:07:36 charge total import: 2257.932kWh
[easee ] WARN 2024/01/05 09:07:46 smart charging: timeout
[lp-1  ] DEBUG 2024/01/05 09:07:46 charger status: A
[site  ] DEBUG 2024/01/05 09:07:55 ----
[easee ] WARN 2024/01/05 09:08:05 smart charging: timeout
[lp-1  ] DEBUG 2024/01/05 09:08:05 charge power: 0W
[site  ] DEBUG 2024/01/05 09:08:06 pv power: 125W
[site  ] DEBUG 2024/01/05 09:08:06 grid power: 423
```

### What type of operating system are you running?

Docker container

### Version

0.123.5 (9ae77a57) Nightly build

andig, 2024-01-05:
Bitte trace log für Easee.

g-rudi, 2024-01-05:
Attached:

[trace.log](https://github.com/evcc-io/evcc/files/13840015/trace.log)

andig, 2024-01-05:
/cc @naltatis

GrimmiMeloni, 2024-01-05:
Ich habe das auch schon ein paar Mal gesehen. Die gute Nachricht ist, es hat keinen Einfluss auf die Funktionalität. Selbst die LED Steuerung an sich funktioniert eigentlich immer. Es ist meiner Ansicht nach nur die Easee API die hier keine (korrekten) Antworten liefert. Ich beobachte das bei mir nochmal gezielter und schaue ob wir hier noch etwas verbessern können, oder ob die Antworten wirklich so random sind, wie sie mir erscheinen.

g-rudi, 2024-01-05:
Besten Dank @GrimmiMeloni 

Das grössere Problem ist jedoch, dass die Steuerung der Ladeleistung nicht funktioniert. Egal was ich im UI eingebe (minPV, PV,aus, etc), hat keine Auswirkungen. Der BMW iX1 lädt immer mit der Stromstärke, welche ich im Auto definiert habe.

Nun weiss ich nicht, wo der Fehler liegt. Ev ist sogar das Auto das Problem und nicht Easee bzw,. evcc

GrimmiMeloni, 2024-01-05:
> Das grössere Problem ist jedoch, dass die Steuerung der Ladeleistung nicht funktioniert.

Dieser Fehler geht leider aus dem Log nicht hervor. Bitte erstelle nochmal ein Trace Log, wenn die Regelung nicht greift. Dann kann ich ggf. mehr sagen.

andig, 2024-01-05:
> Ich habe das auch schon ein paar Mal gesehen.

Was hier auffällt ist, dass das Logfile überhaupt keine `CommandResponse` enthält. Fast so, als würde Easee nie auf smartcharging reagieren? Ob man das dem Easee Support als Frage stellen kann?

GrimmiMeloni, 2024-01-06:
Das ist wieder so ein Fall wo alles zusammen kommt. Der Charger sagt `SMART_CHARGING false`, und wir senden per API eine Sekunde später ein `true` doch die API gibt uns nur die leere (NoOp) Antwort, obwohl true nicht dem State auf dem Charger entspricht. Das kennen wir ja schon vom Dynamic Charger Current.  
Zusätzlich ist das insbesondere beim Abstecken auch eine Timing Sache. Da resettet der Charger eigenständig den Wert - das ist quasi der kleine Bruder von "Charger out of sync".

Ich werde das ganze zumindest mal so anpassen, das wir Smart Charging nicht mehr auf dem Charger setzen wenn kein Auto angesteckt ist. In Status A kann man die LED eh nicht steuern. Vielleicht reicht das schon um das ganze zumindest zu beruhigen.

andig, 2024-01-06:
Alternativ: wir schmeissen das SmartCharging als weitere Fehlerquelle raus. Für die Funktion ist es unerheblich.

/cc @naltatis

GrimmiMeloni, 2024-01-06:
Ja, erstmal probieren ob besagter Change nicht schon hilft. Ich mag den blauen LED Strip. 😅

g-rudi, 2024-01-06:
> Ja, erstmal probieren ob besagter Change nicht schon hilft. Ich mag den blauen LED Strip. 😅

Werde morgen den nightly build installieren und mal testen.

g-rudi, 2024-01-07:
Getestet in 0.123.7. Kann geschlossen werden. Issue wurde gefixt.

closed 2024-01-07 by g-rudi