evcc-io/evcc #11494: new battery mode broken

DanielTuerk, 2024-01-05:
### Describe the bug

The UI isn't displayed correctly (in Safari and Chrome on OSX).
[image]

By switching on the new battery mode, I got the error:
"Request failed with status code 400: API request failed http://wallbox:7070/api/batterydischargecontrol/true"

Looks like a broken evcc config due to the update, non compatible change in a hotfix version?

[site  ] WARN 2024/01/05 09:49:11 battery configured but residualPower is missing or <= 0

by adding "residualPower: 100" to the yaml I got:
"Request failed with status code 400: API request failed http://wallbox:7070/api/batterydischargecontrol/true"
No log entry for that warning on the UI.

So at this point I reverted the version.

### Steps to reproduce

1. click on the percentage of the battery
2. switch the toggle for the new battery mode (https://github.com/evcc-io/evcc/pull/11378)

### Configuration details

```yaml
meters:
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.178.132
  port: 502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.178.132
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.178.132
  port: 502
  usage: battery
  modbus: tcpip
  name: battery3

chargers:
- type: template
  template: heidelberg
  id: 2
  device: /dev/ttyUSB0
  baudrate: 19200
  comset: 8E1
  modbus: rs485serial
  name: wallbox6

loadpoints:
- title: Garage
  charger: wallbox6
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
  residualPower: 100
```

### Log details

```text
[main  ] INFO 2024/01/05 09:49:07 evcc 0.123.5
[main  ] INFO 2024/01/05 09:49:07 using config file: /etc/evcc.yaml
[main  ] INFO 2024/01/05 09:49:07 starting ui and api at :7070
[db    ] INFO 2024/01/05 09:49:08 using sqlite database: /home/pv/.evcc/evcc.db
[lp-1  ] DEBUG 2024/01/05 09:49:11 charge total import: 6350.235kWh
[site  ] WARN 2024/01/05 09:49:11 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site  ] INFO 2024/01/05 09:49:11 site config:
[site  ] INFO 2024/01/05 09:49:11   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/01/05 09:49:11     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/01/05 09:49:11     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/01/05 09:49:11     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/01/05 09:49:11   vehicles:
[site  ] INFO 2024/01/05 09:49:11     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
[site  ] INFO 2024/01/05 09:49:11     vehicle 2: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
[site  ] INFO 2024/01/05 09:49:11     vehicle 3: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/01/05 09:49:11 loadpoint 1:
[lp-1  ] INFO 2024/01/05 09:49:11   mode:        pv
[lp-1  ] INFO 2024/01/05 09:49:11   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✓
[lp-1  ] INFO 2024/01/05 09:49:11   meters:      charge ✓
[lp-1  ] INFO 2024/01/05 09:49:11     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2024/01/05 09:49:11 phase timer inactive
[lp-1  ] DEBUG 2024/01/05 09:49:11 pv timer inactive
[lp-1  ] DEBUG 2024/01/05 09:49:11 guard timer inactive
[site  ] WARN 2024/01/05 09:49:11 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/01/05 09:49:11 ----
[server] FATAL 2024/01/05 09:49:11 unlinkat /tmp/evcc: operation not permitted
```

### What type of operating system are you running?

Linux

### Version

0.123.5

andig, 2024-01-05:
Duplikat

andig, 2024-01-05:
Duplikat zu https://github.com/evcc-io/evcc/issues/11423

closed 2024-01-05 by andig

DanielTuerk, 2024-01-05:
Dennoch war die evcc.yaml nicht kompatibel mit dem fehlenden Eintrag "residualPower", passt doch nicht für ein Hotfix Release. Oder verstehe ich hier was falsch?

Erst danach kommt der 400er Response in der UI.

andig, 2024-01-05:
> Oder verstehe ich hier was falsch?

siehe release notes!