evcc-io/evcc #11500: v0.123.x seem to set min soc for system battery to 20% on restart EVCC or start EV "fast" charging

meckigabel, 2024-01-05:
### Describe the bug

Since one of the new releases after x-mas 2023, EVCC seems to set min soc of my Victron ESS system for the battery to 20% when starting EVCC or when setting EV charging to "fast". If the battery SOC when making the EV setting is below 20% this causes the system to charge the battery from the grid until it reaches 20% soc (the car charges "fast" in the background as expected).
I am not sure if this is a bug or if this is just me setting the settings wrong... However, this behaviour did not seem to exist on the o.122.x versions of EVCC.

### Steps to reproduce

1. System: Victron ESS (em24 meter, 3xMultiPlus II 48/5000/70-50, MPPT SmartSolar 450/200, 3x2xPylontech 5000 battery)
2. EVCC Version v0.123.5
3. yaml does not contain any resetondisconnect or prioritysoc or buffersoc or the like anymore

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxxxx

interval: 10s # control cycle interval

sponsortoken: xxxxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: victron-energy 
  usage: grid  
  host: xxxxx  
  port: 502  
  name: grid1
- type: template
  template: victron-energy 
  usage: pv  
  host: xxxx  
  port: 502  
  name: pv2
- type: template
  template: victron-energy 
  usage: battery  
  host: xxxx
  port: 502  
  name: battery3

chargers:
- type: template
  template: abl 
  id: 3  
  device: /dev/ttyUSB0  
  baudrate: 38400  
  comset: 8E1  
  modbus: rs485serial  
  name: wallbox4
- type: template
  template: abl 
  id: 4  
  device: /dev/ttyUSB0  
  baudrate: 38400  
  comset: 8E1  
  modbus: rs485serial  
  name: wallbox5

loadpoints:
- title: eMH3_left
  charger: wallbox4
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  #resetOnDisconnect: true
- title: eMH3_right
  charger: wallbox5
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  #resetOnDisconnect: true

site:
  title: xxxxxx
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
#  prioritySoc: 50 # Hausbatterie bekommt bis zum Soc 50% Priorität beim laden
#  bufferStartSoc: 80 # hat die Hausbatterie Soc 80% erreicht startet der ladevorgang
#  bufferSoc: 60 # Hausbatterie wird oberhalb Soc 60% als Puffer genutzt

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.32

  feedin:
    type: fixed
    price: 0.08

# mqtt message broker
mqtt:
  broker: xxxx
  topic: evcc # root topic for publishing, set empty to disable
  user: xxxx
  password: xxxxx

vehicles:
- type: custom
  name: eqb
  title: xxxxx
  capacity: 66 # battery capacity kWh
  phases: 3 # how many phases it uses
  icon: car
  soc: # battery soc (%)
   source: http
   uri: http://xxxxx/v1/state/mercedesme.0.xxxxx.state.soc.displayValue/plain?extraPlain=true
   method: GET # default HTTP method
   timeout: 5s
  range: # optional electric range (km)
   source: http
   uri: http://xxx/v1/state/mercedesme.0.xxxx.state.overallRange.displayValue/plain?extraPlain=true
   method: GET # default HTTP method
   timeout: 5s 
   
  #onIdentify: # set defaults when vehicle is identified
   #mode: pv
   #mincurrent: 6
   #targetSoc: 100
```

### Log details

```text
there is no error message coming up
```

### What type of operating system are you running?

Linux

### Version

0.123.5

VolkerK62, 2024-01-05:
have a look here   https://github.com/evcc-io/evcc/issues/11252

mdkeil, 2024-01-05:
you have to set `minsoc:`  under
```
- type: template
  template: victron-energy 
  usage: battery  
  host: xxxx
  port: 502  
  name: battery3
  minsoc: 10 for example 
```
If you didn't set minsoc under yaml, the default value ist 20. If you change victron minsoc after evcc start, evcc will not noticed.

mdkeil, 2024-01-05:
@premultiply 
Meiner Meinung nach wäre es sinnvoller als Ladesperre den maxdischargecurrent auf 0 zu setzen.. anschließend entweder auf -1 (kein Limit) oder eleganter weise auf den letzen Wert (der in einer variable zwischengespeichert wird)

andig, 2024-01-05:
@meckigabel you have apparently activated battery discharge control, so this is expected.