evcc-io/evcc #11501: Fenecon Batterie Steuerung

JBufe, 2024-01-05:
Bezgl. der Fenecon Batterie Ansteuerung ein Hinweis. Mittels einer kostenpflichtigen Fenecon APP lässt sich die Batterie problemlos über Modvus TCIp ansteuern. Besteht die Möglichkeit diese Bastelanleitung in eure Software zu integrieren? Bei Bedarf kann ich gerne die Modvus Tabelle mit den Variablen zur Verfügung stellen. 

Dies würde mir den Umweg über ioBroker ersparen, um das umladen aus der Hausbatterie in die Autobatterie zu verhindern.

andig, 2024-01-05:
Doku würde helfen das zu bewerten.

andig, 2024-01-08:
@JBufe lass uns hier gerne weiter machen wenn es Erkenntnisse dazu gibt.

JBufe, 2024-01-13:
Hallo Andi,

 

anbei die Modbus Tabelle inkl. Schriebzugriff für das Fenecon EMS Home System

 

Mfg

 

Jörg Bufe

 

Closed #11501 <https://github.com/evcc-io/evcc/issues/11501>  as completed.

JBufe, 2024-01-13:
Hallo Andi,

Sorry ich war in Ägypten und konnte deswegen nicht antworten. Ich habe den FEMS Home und nach meinem Wissenstands ist gilt das Modbus Protokoll für die gesamte Baureihe. Ob diese identisch mit OpenEMS kann ich aufgrund fehlender Daten nicht beantworten.  Ich habe dir die Modbus Tabelle per E-Mail zukommen lassen, bzw. beigefügt.
[Modbus-TCP-fems65881.xlsx](https://github.com/evcc-io/evcc/files/13928314/Modbus-TCP-fems65881.xlsx)

andig, 2024-01-14:
@premultiply wollen wir von HTTP auf Modbus umsteigen? Lohnt sich das?

andig, 2024-01-14:
/cc @sfeilmeier

premultiply, 2024-01-14:
> wollen wir von HTTP auf Modbus umsteigen? Lohnt sich das?

Es spricht jedenfalls nicht dagegen.

andig, 2024-01-15:
Ich würde gerne einen Kommentar von Fenecon abwarten bevor wir das tun- insbesondere um keinen Breaking Change einzuführen.

premultiply, 2024-01-17:
> Diese App ist nicht im Standard-Lieferumfang des FEMS enthalten. Sie kann jedoch nachträglich jederzeit nachgerüstet werden. Kontaktieren Sie uns hierfür!

JBufe, 2024-01-18:
In eigener Sache einen Hinweis an H. Feilmeier. Das Fenecon Home System ist echt gut und ich habe den Kauf bisher nicht bereut. Auch das APP Modell ist durchdacht und sinnvoll. Nur bezgl.der APP Preise solltet Fenecon noch mal in sich gehen. Pauschal knapp 400 € pro APP ist echt happig. Beim Modbus Schreibzugriff für die handvoll Variablen schon extrem mutig. Das Tibber Modul wäre noch interessant aber bei 400 € sehe ich keinen Return of Invest  bezogen auf die Lebensdauer der Anlage. In meinem Fall realisiere ich dann lieber eine Lösung mittels ioBroker. Damit erreichte ich zwar nicht die Effizienz wie ihr,  aber die erreichten 80% rechnen sich wenigsten. Nimmt man dann noch Wärmepumpen Apps dazu wird das ganze schnell unverschämt teuer. Ich denke mittels akteaktiver Preise könnte nach mehr Nutzer dafür begeistern.

JBufe, 2024-01-18:
P S. den TCP/IP Modbus Schreibzugriff have ich mir gekauft. Funktioniert fehlerfrei und erlaubt via ioBroker den Eigenbau diverser Funktionen. Unter anderem die Batterie beim laden des Autos zu sperren. Sven das ist alles von hinten durch die Brust und Auge geschossen,  weil solche Features von den Herstellern der PV Anlage nicht frei verfügbar sind, bzw. nur zu gönnen Preisen. Somit wird die sinnvolle Kopplung der System nur erschwert bzw. durch Workarounds ersetzt.

Beispiel EVCC. Ein tolles Open Source Produkt mit Tibber Integration. Dasx wird auch Fenecon nicht mehr besser hinbekommen. Aber durch die Politik von Fenecon verdoppeln sich die Entladezyklen beim Laden eines Autos sehr leicht. Daraus resultiert in etwa eine Halbierung der Akku Lebensdauer plus Energieverschwendung von knapp 10% durch die Ladegeräte beim Zwischenladen der Hausbatterie. Das ist weder nachhaltig noch ökologisch sinnvoll.

andig, 2024-01-19:
Zusammenfassend: Batteriesteuerung ist universell einsetzbar, benötigt aber kostenpflichtigen Schreibzugriff. Dieser ist über Modbus oder HTTP verfügbar. Implementierung ist also möglich.

Vor dem Hintergrund der vmtl. geringen Anzahl an Fenecon Usern mit Schreibzugriff würden wir zum jetzigen Zeitpunkt von einer Implementierung in evcc absehen. Für Hintergrunddiskussion sind wir unter info@evcc.io erreichbar.

closed 2024-01-19 by andig

andig, 2024-11-11:
Afaik this requires a license.

andig, 2024-11-11:
Na dann sind wir an der gleichen Stelle wie bei anderen Geräten auch: es braucht Doku und Implementierung.

muxodo, 2025-04-29:
> > Na dann sind wir an der gleichen Stelle wie bei anderen Geräten auch: es braucht Doku und Implementierung.
> 
> Ich hätte ein Template lokal angepasst und als PR anzubieten, falls gewünscht. Es realisiert nur die Batteriesperre, nutzt die read/write REST-API und befindet sich im Test. Doku, was seitens openEMS nötig ist, fehlt noch. Weicht in der Realisierung jedoch nicht vom aktuellen OpenEMS-Template ab.

Gibt es eine Möglichkeit, dass ich das bei mir lokal einbaue? Schreibzugriff FEMS vorhanden.

andig, 2025-04-29:
PR welcome