evcc-io/evcc #11504: Battery control: lock battery when smart charging is ongoing

mclane, 2024-01-05:
Actually, battery discharging can be locked if charge mode is "Fast" or in case of planned charging. However, battery is discharged if "smart charging (e.g. CO2 limited charging) is on,

**Describe the solution you'd like**
Battery discharge should also be locked in case of "smart charging" (specifically if controlled by CO2-limit)

**Describe alternatives you've considered**
If this is not desired or possible; I would lock the battery via an external logic (e.g. node-red).

andig, 2024-01-05:
Good catch!

VolkerK62, 2024-01-05:
Would it make sense, also include the minSoc loading. It is similar to "fast".

andig, 2024-01-05:
Mhhm. It is and it's not. Dunno...

mdkeil, 2024-01-05:
I think all charging-situation with full power should hold battery.

premultiply, 2024-01-06:
I think not. 🤷🏻‍♂️
It may import most expensive energy from grid at any time just because you connected the car.

mdkeil, 2024-01-06:
Sorry, but I don't understand your reason.

mdkeil, 2024-01-06:
Ok, in my external control I further check chargePower (>4500)

closed 2024-01-06 by andig