evcc-io/evcc #11508: Goodwe Hybrid: Use AC Power and not DC

maatinh, 2024-01-05:
### Describe the bug

evcc is using Modbus Register 35191 to read current PV energy. This is the DC side of the inverter. Instead register 35138 "Total power of inverter" should be used which reflects the AC output of the inverter.
There is a difference in both values especially in the morning there is ~50W on DC and 0W on AC. The same in the evening. During the day the diff can be huge in some cases:

[image]

[image]

### Steps to reproduce

none

### Configuration details

```yaml
use goodwe-hybrid template
```

### Log details

```text
none required
```

### What type of operating system are you running?

Docker container

### Version

0.123.3

andig, 2024-01-05:
> evcc is using Modbus Register 35191 to read current PV energy. This is the DC side of the inverter. Instead register 35138 "Total power of inverter" should be used which reflects the AC output of the inverter.

This seems wrong: Power is a different physical measurement from Energy.

@premultiply we should remove pv energy from all templates- it may be wrong and it's not even used by evcc.

maatinh, 2024-01-05:
I'm referring to this line: https://github.com/evcc-io/evcc/blob/0705fca00aa19e009c31747f38ccf04e53ed6c27/templates/definition/meter/goodwe-hybrid.yaml#L67C14-L67C14

This should use register 35138.

andig, 2024-01-05:
That is energy. 35138 acoording to what you wrote is power. Energy is NOT Power. Closing.

closed 2024-01-05 by andig

premultiply, 2024-01-05:
As this is a hybrid inverter AC output metering is useless. It will return just a mixture of solar, battery and other self consumption. The only way to monitor a hybrid inverter is to get the dc metering on solar and battery paths.

maatinh, 2024-01-05:
EVCC displays generation in the UI now where there is actually none effectively. What evcc is reporting now is totally consumed internally within the inverter. DC side metering is useless unless there is some power on AC side or into batteries.

I'm referring to what the UI display as "Erzeugung" in german UI. This represents wrong value for goodwe hybrid. Wherever the value comes from.