evcc-io/evcc #11514: Nach Update Fehlermeldungen

AlexGabl, 2024-01-05:
### Describe the bug

Nach Update von EVCC auf die neueste Version und dem herausnehmen der Zeilen minSOC; target SOC und reset on disconnect bleiben diese Fehlermeldungen: 

failed configuring loadpoints
failed configuring loadpoint
1 error(s) decoding: * '' has invalid keys: vehicles

Anbei meine YAML- Datei

Was gibt es noch für Möglichkeiten! Danke für`s Weiterhelfen! MFG Alex

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 2c22bd34784be8dd0c8d11c6f122a90fc34570abb6629e85a888d8d8a8c2557e

interval: 10s # control cycle interval
sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJleHAiOjE3ODQ5MTA4ODAsImlhdCI6MTY5MDMwMjg4MCwiaXNzIjoiZXZjYy5pbyIsInN1YiI6IkFsZXhHYWJsIn0.dLKWiqEEvW7pts5cUkSdFuX0JzQoJ4HpkaIKhC5Sbco

meters:
- type: template
  template: solaredge-inverter 
  id: 1  
  host: 192.168.178.27  
  port: 1502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: solaredge-inverter 
  id: 1  
  host: 192.168.178.27  
  port: 1502  
  usage: minpv  
  modbus: tcpip  
  name: pv2

vehicles:
- type: template
  template: citroen 
  title: Auto  
  user: alex.xxxxxxxx@gmail.com  
  password: Axxxx!  
  capacity: 50  
  name: ev3

site:
  title: My home
  meters:
    grid: grid1
    pv:
    - pv2

chargers:
  - name: my_charger
    type: template
    template: go-e-v3
    host: 192.168.178.42  

loadpoints:
  - title: Garage
    charger: my_charger
    vehicles: 
    - ev3    
    phases: 3
    mincurrent: 6
    maxcurrent: 16
    mode: pv
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

### Version

0.123.5

0x3d13f, 2024-01-05:
Du musst noch die Zeile `vehicles` und die Folgezeile im Loadpoint raus nehmen.

AlexGabl, 2024-01-05:
Vielen Dank - funktioniert!

hätte ich das sehen können?

LG Alex

0x3d13f, 2024-01-05:
Na, die Fehlermeldung gibt schon einen Hinweis auf den key vehicles.

mdkeil, 2024-01-05:
> Du musst noch die Zeile `vehicles:` und die Folgezeile `- ev3` im Loadpoint raus nehmen.

nö.. `vehicles` unter loadpoint muss nur in `vehicle` umbenannt werden.. Wobei Du dass nur benötigst, wenn du ein Fahrzeug fest an einen lp zuordnen möchtest, was aber generell nicht notwendig ist.

0x3d13f, 2024-01-05:
Ah, Mist. Also dann `vehicle: ev3`, richtig?

AlexGabl, 2024-01-05:
Passt! Super! Vielen dank

closed 2024-01-06 by andig