evcc-io/evcc #11515: Easee: Charger Template Fehler beim Start

mrvodoo, 2024-01-05:
### Describe the bug

Hi zusammen,
ich betreibe evcc seit ca. einem Jahr problemlos in Docker auf meiner Synology NAS mit unveränderter config yaml. Nach dem Update heute auf Version 0.123.5 konnte ich den Container nicht mehr starten. Auch der Container mit der älteren Version lässt sich mit gleicher Fehlermeldung nicht mehr starten.

Bereits geprüft:
1. Login im  easee Portal erfolgreich
2. easee APIs scheinen alle online zu sein (https://status.easee.com/)
3. easee App funktioniert ohne Fehlermeldung

Wäre dankbar für euren Support :)
VG
Marcel

### Steps to reproduce

1. Versuch Docker Container mit evcc/latest (Version 0.123.5) zu starten
2. Start Abbruch mit Fehlermeldung

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: trace
#levels:
  #cache: error

# unique installation id
plant: ****
interval: 15s # control cycle interval

sponsortoken: ****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: discovergy 
  usage: grid  
  user: **** 
  password: ****  
  meter: ****  
  name: grid1
- type: template
  template: huawei-dongle-powersensor 
  id: 2  
  host: ***.***.***.***  
  port: 502  
  usage: pv  
  storageunit: 1  
  modbus: tcpip  
  name: pv1
- type: template
  template: huawei-dongle-powersensor 
  id: 2  
  host: ***.***.***.***  
  port: 502  
  usage: battery  
  storageunit: 1  
  modbus: tcpip  
  name: battery1

chargers:
- type: template
  template: easee 
  user: ****  
  password: ****  
  charger: ****  
  name: wallbox1

loadpoints:
- title: Carport
  charger: wallbox1
  mode: minpv
  phases: 1
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

vehicles:
- name: zoe
  title: Zoe
  type: template
  template: offline
  capacity: 41
  phases: 1
  mode: minpv
  identifiers:
  - ****
  - ****
  icon: car

site:
  title: Home
  meters:
    grid: grid1
    pvs:
    - pv1
    batteries:
    - battery1
  prioritySoc: 30 # Hausbatterie bekommt bis zum Soc 30% Priorität beim Laden
  residualPower: 100 # Batteriespeicher - Positiver Wert: Verbleibende Netzeinspeiseleistung
  bufferSoc: 50 # Hausbatterie wird oberhalb Soc 50% als Puffer genutzt

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.4202 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.093 # [currency]/kWh
```

### Log details

```text
[main  ] INFO 2024/01/05 21:52:12 evcc 0.123.5
[main  ] INFO 2024/01/05 21:52:12 using config file: /etc/evcc.yaml
[main  ] INFO 2024/01/05 21:52:12 starting ui and api at :7070
[db    ] INFO 2024/01/05 21:52:12 using sqlite database: /root/.evcc/evcc.db
[main  ] FATAL 2024/01/05 21:52:22 cannot create charger 'wallbox1': cannot create charger type 'template': cannot create charger type 'easee': Post "https://blackhole.easee.com": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[main  ] FATAL 2024/01/05 21:52:22 will attempt restart in: 5m0s
```

### What type of operating system are you running?

Docker container

### Version

0.123.5

0x3d13f, 2024-01-05:
Abgesehen von dem Easee Problem (da kann ich nicht helfen) wirst Du noch ein paar Anpassungen in der evcc.yaml vornehmen müssen, damit diese Version startet:
`resetondisconnect`, `buffersoc`, `prioritysoc`  (entfernen), `pvs` (ändern in `pv`) und `batteries` (ändern in `battery`).

mrvodoo, 2024-01-05:
> Abgesehen von dem Easee Problem (da kann ich nicht helfen) wirst Du noch ein paar Anpassungen in der evcc.yaml vornehmen müssen, damit diese Version startet: `resetondisconnect`, `buffersoc`, `prioritysoc` (entfernen), `pvs` (ändern in `pv`) und `batteries` (ändern in `battery`).

Danke dir für die schnelle Antwort und das Review der yaml. Ist angepasst. 
Sollte wohl öfter updaten und die Release Notes besser lesen ;)

Das evcc Problem besteht nach Anpassung weiterhin. Vermute ein API Problem aus easee Seite.

mrvodoo, 2024-01-05:
Hier noch die detaillierteren Traces:

[easee ] TRACE 2024/01/05 22:53:34 POST https://api.easee.com/api/accounts/login
[easee ] TRACE 2024/01/05 22:53:34 {"userName":"***","password":"***"}
[easee ] TRACE 2024/01/05 22:53:34 GET https://blackhole.easee.com
[cache ] TRACE 2024/01/05 22:53:44 file: /etc/evcc.yaml
[main  ] FATAL 2024/01/05 22:53:44 cannot create charger 'wallbox1': cannot create charger type 'template': cannot create charger type 'easee': Post "https://blackhole.easee.com": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[main  ] FATAL 2024/01/05 22:53:44 will attempt restart in: 5m0s

andig, 2024-01-06:
Easee blockt dich- Support kontaktieren.