evcc-io/evcc #11517: OpenWB Software 2.0: Warning for incorrect chagrger status

FelR, 2024-01-06:
### Describe the bug

After updating an OpenWB from software version 1.9 to 2.0, various warning messages are generated during charging. 

- If the charging process is activated via fast charging, the warning `charger logic error: disabled but charging` is displayed.
- If the vehicle ends the charging process (vehicle soc limit reached) while fast charging is still active in EVCC, the warning `charger out of sync: expected enabled, got disabled` is displayed.

### Steps to reproduce

See bug description.

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: https
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.abc.xyz
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken:

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
  #    # rtu: true
  #    # readonly: true

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
[... 249 lines cut ...]
[site  ] DEBUG 2024/01/06 09:45:34 battery power: 0W
[site  ] DEBUG 2024/01/06 09:45:34 grid power: 11510W
[site  ] DEBUG 2024/01/06 09:45:34 site power: 11510W
[lp-1  ] DEBUG 2024/01/06 09:45:34 charge currents: [16.1 16.2 16.1]A
[lp-1  ] DEBUG 2024/01/06 09:45:34 detected active phases: 3p
[lp-1  ] DEBUG 2024/01/06 09:45:34 charge total import: 764.341kWh
[lp-1  ] DEBUG 2024/01/06 09:45:34 charger status: C
[lp-1  ] DEBUG 2024/01/06 09:45:34 soc estimated: 79.19% (vehicle: 79.00%)
[lp-1  ] DEBUG 2024/01/06 09:45:34 vehicle soc: 79%
[lp-1  ] DEBUG 2024/01/06 09:45:34 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/01/06 09:45:34 vehicle range: 419km
[lp-1  ] WARN 2024/01/06 09:45:34 charger logic error: disabled but charging
[lp-1  ] DEBUG 2024/01/06 09:45:34 charger: guard elapse
[lp-1  ] DEBUG 2024/01/06 09:45:34 guard timer inactive
[lp-1  ] DEBUG 2024/01/06 09:45:34 set tesla plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
[site  ] DEBUG 2024/01/06 09:45:44 ----
```

### What type of operating system are you running?

Docker container

### Version

v0.123.5

andig, 2024-01-06:
@MarkusGH siehst du das auch? Ich dachte eigentlich, die hätten wir gelöst?

andig, 2024-01-06:
@FelR es braucht bitte mal ein trace log für `openwb-2.0`.

FelR, 2024-01-06:
Hier einmal das Trace-Log während des Ladens:
```
[site  ] DEBUG 2024/01/06 14:40:26 ----
[openwb-2.0] TRACE 2024/01/06 14:40:26 modbus: send 00 34 00 00 00 06 01 04 27 74 00 02
[openwb-2.0] TRACE 2024/01/06 14:40:26 modbus: recv 00 34 00 00 00 07 01 04 04 00 00 2c b0
[lp-1  ] DEBUG 2024/01/06 14:40:26 charge power: 11440W
[lp-2  ] DEBUG 2024/01/06 14:40:26 charge power: 0W
[site  ] DEBUG 2024/01/06 14:40:26 pv power: 178W
[site  ] DEBUG 2024/01/06 14:40:26 battery soc: 65%
[site  ] DEBUG 2024/01/06 14:40:26 battery power: -419W
[site  ] DEBUG 2024/01/06 14:40:26 grid power: 12981W
[site  ] DEBUG 2024/01/06 14:40:26 site power: 12562W
[openwb-2.0] TRACE 2024/01/06 14:40:26 modbus: send 00 35 00 00 00 06 01 04 27 7b 00 03
[openwb-2.0] TRACE 2024/01/06 14:40:26 modbus: recv 00 35 00 00 00 09 01 04 06 06 4a 06 54 06 51
[lp-1  ] DEBUG 2024/01/06 14:40:26 charge currents: [16.1 16.2 16.2]A
[openwb-2.0] TRACE 2024/01/06 14:40:26 modbus: send 00 36 00 00 00 06 01 04 27 76 00 02
[openwb-2.0] TRACE 2024/01/06 14:40:26 modbus: recv 00 36 00 00 00 07 01 04 04 00 0b ad 32
[openwb-2.0] TRACE 2024/01/06 14:40:26 modbus: send 00 37 00 00 00 06 01 04 27 76 00 02
[openwb-2.0] TRACE 2024/01/06 14:40:26 modbus: recv 00 37 00 00 00 07 01 04 04 00 0b ad 32
[lp-1  ] DEBUG 2024/01/06 14:40:26 charge total import: 765.234kWh
[openwb-2.0] TRACE 2024/01/06 14:40:26 modbus: send 00 38 00 00 00 06 01 04 27 83 00 01
[openwb-2.0] TRACE 2024/01/06 14:40:26 modbus: recv 00 38 00 00 00 05 01 04 02 00 01
[lp-1  ] DEBUG 2024/01/06 14:40:26 charger status: C
[openwb-2.0] TRACE 2024/01/06 14:40:26 modbus: send 00 39 00 00 00 06 01 04 27 b0 00 0a
[openwb-2.0] TRACE 2024/01/06 14:40:26 modbus: recv 00 39 00 00 00 17 01 04 14 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[lp-1  ] DEBUG 2024/01/06 14:40:26 soc estimated: 80.04% (vehicle: 80.00%)
[lp-1  ] DEBUG 2024/01/06 14:40:26 vehicle soc: 80%
[lp-1  ] DEBUG 2024/01/06 14:40:26 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/01/06 14:40:26 vehicle range: 420km
[openwb-2.0] TRACE 2024/01/06 14:40:26 modbus: send 00 3a 00 00 00 06 01 04 27 84 00 01
[openwb-2.0] TRACE 2024/01/06 14:40:26 modbus: recv 00 3a 00 00 00 05 01 04 02 06 40
[lp-1  ] WARN 2024/01/06 14:40:26 charger logic error: disabled but charging
[lp-1  ] DEBUG 2024/01/06 14:40:26 set tesla plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
[site  ] DEBUG 2024/01/06 14:40:36 ----
```

Und einmal nachdem das Fahrzeug den Ladevorgang beendet hat:

```
[site  ] DEBUG 2024/01/06 14:44:46 ----
[openwb-2.0] TRACE 2024/01/06 14:44:46 modbus: send 00 a5 00 00 00 06 01 04 27 74 00 02
[openwb-2.0] TRACE 2024/01/06 14:44:46 modbus: recv 00 a5 00 00 00 07 01 04 04 00 00 00 00
[lp-1  ] DEBUG 2024/01/06 14:44:46 charge power: 0W
[lp-2  ] DEBUG 2024/01/06 14:44:46 charge power: 0W
[site  ] DEBUG 2024/01/06 14:44:46 pv power: 461W
[site  ] DEBUG 2024/01/06 14:44:46 battery soc: 65%
[site  ] DEBUG 2024/01/06 14:44:46 battery power: 0W
[site  ] DEBUG 2024/01/06 14:44:46 grid power: 1408W
[site  ] DEBUG 2024/01/06 14:44:46 site power: 1408W
[openwb-2.0] TRACE 2024/01/06 14:44:46 modbus: send 00 a6 00 00 00 06 01 04 27 7b 00 03
[openwb-2.0] TRACE 2024/01/06 14:44:46 modbus: recv 00 a6 00 00 00 09 01 04 06 00 00 00 00 00 00
[lp-1  ] DEBUG 2024/01/06 14:44:46 charge currents: [0 0 0]A
[openwb-2.0] TRACE 2024/01/06 14:44:46 modbus: send 00 a7 00 00 00 06 01 04 27 76 00 02
[openwb-2.0] TRACE 2024/01/06 14:44:46 modbus: recv 00 a7 00 00 00 07 01 04 04 00 0b ad 47
[lp-1  ] DEBUG 2024/01/06 14:44:46 charge total import: 765.255kWh
[openwb-2.0] TRACE 2024/01/06 14:44:46 modbus: send 00 a8 00 00 00 06 01 04 27 83 00 01
[openwb-2.0] TRACE 2024/01/06 14:44:46 modbus: recv 00 a8 00 00 00 05 01 04 02 00 00
[openwb-2.0] TRACE 2024/01/06 14:44:46 modbus: send 00 a9 00 00 00 06 01 04 27 82 00 01
[openwb-2.0] TRACE 2024/01/06 14:44:46 modbus: recv 00 a9 00 00 00 05 01 04 02 00 01
[lp-1  ] DEBUG 2024/01/06 14:44:46 charger status: B
[openwb-2.0] TRACE 2024/01/06 14:44:46 modbus: send 00 aa 00 00 00 06 01 04 27 b0 00 0a
[openwb-2.0] TRACE 2024/01/06 14:44:46 modbus: recv 00 aa 00 00 00 17 01 04 14 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[openwb-2.0] TRACE 2024/01/06 14:44:46 modbus: send 00 ab 00 00 00 06 01 04 27 84 00 01
[openwb-2.0] TRACE 2024/01/06 14:44:46 modbus: recv 00 ab 00 00 00 05 01 04 02 06 40
[lp-1  ] WARN 2024/01/06 14:44:46 charger out of sync: expected enabled, got disabled
[lp-1  ] DEBUG 2024/01/06 14:44:46 charger: guard elapse
[lp-1  ] DEBUG 2024/01/06 14:44:46 guard timer inactive
[lp-1  ] DEBUG 2024/01/06 14:44:46 set tesla plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
[openwb-2.0] TRACE 2024/01/06 14:44:46 modbus: send 00 ac 00 00 00 06 01 06 27 bb 06 40
[openwb-2.0] TRACE 2024/01/06 14:44:46 modbus: recv 00 ac 00 00 00 06 01 06 27 bb 06 40
[lp-1  ] DEBUG 2024/01/06 14:44:46 charger enable
[lp-1  ] DEBUG 2024/01/06 14:44:46 wake-up timer: start
[site  ] DEBUG 2024/01/06 14:44:56 ----
```

MarkusGH, 2024-01-07:
> @MarkusGH siehst du das auch? Ich dachte eigentlich, die hätten wir gelöst?

Die Fehlermeldungen sind korrekt - die OpenWB 2.0 benimmt sich nicht richtig und der Loadpoint beklagt sich zu Recht.

Fix in https://github.com/evcc-io/evcc/pull/11543

andig, 2024-01-07:
@FelR hast Du bitte mal ein trace log wo Du von Sofortladen auf "aus" umschaltest? Nur das Stück, bis die Fehlermeldung kommt. DANKE!

closed 2024-01-07 by andig

FelR, 2024-01-07:
Habe nur am Wochenende Zugriff auf die Wallbox. Kann leider erst wieder in 5 Tagen weiter testen.