evcc-io/evcc #11527: ID.3 (offline) after update from 0.123.0 to 0.123.5

fizzgig242, 2024-01-06:
### Describe the bug

Hey guys,

I updated my evcc docker container on my Synology NAS from 0.123.0 to 0.123.5 and after the update evcc is no longer able to
communicate with my VW ID.3. It is always shown as offline.

I did not change the config file. Just updated the docker container via Portainer.

Any ideas?

Do you need more information?

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: aaa

interval: 10s # control cycle interval

sponsortoken: 12345

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: fronius-solarapi-v1 
  usage: grid  
  host: 192.168.0.108  
  name: grid1
- type: template
  template: fronius-solarapi-v1 
  usage: pv  
  host: 192.168.0.108  
  name: pv2
- type: template
  template: fronius-solarapi-v1 
  usage: battery  
  host: 192.168.0.108  
  name: battery3

chargers:
- type: template
  template: go-e-v3 
  host: 192.168.0.75  
  name: wallbox5

vehicles:
- type: template
  template: id 
  title: ID.3  
  user: xyz 
  password: abc  
  vin: 123  
  capacity: 58  
  timeout: 10s  
  name: ev4

loadpoints:
- title: Wallbox Garage
  charger: wallbox5
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16

site:
  title: Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.401 # [currency]/kWh

  feedin:
    type: fixed
    price: 0.08 # [currency]/kWh

  co2:
    type: grünstromindex
    zip: xxx # PLZ mit führender Null mit in "" setzen
```

### Log details

```text
-
```

### What type of operating system are you running?

Docker container

### Version

0.123.5

fizzgig242, 2024-01-06:
When trying to create a new configuration I get following error:

Teste die Konfiguration von Volkswagen We Connect ID ...
   Fehler: cannot create vehicle type 'vw': missing state

mdkeil, 2024-01-06:
> Do you need more information?

complete evcc log?

f-lehmann-gmx-de, 2024-01-06:
I have the same issue. Used 0123.1 the whole time and worked perfect. After upgrade to latest 0.123.5 the ID3 is offline. 
[evcc_log.csv](https://github.com/evcc-io/evcc/files/13850459/evcc_log.csv)

fizzgig242, 2024-01-06:
Here is my log:

```
[main  ] INFO 2024/01/06 17:41:05 evcc 0.123.5

[main  ] INFO 2024/01/06 17:41:05 using config file: /etc/evcc.yaml

[main  ] INFO 2024/01/06 17:41:05 starting ui and api at :7070

[db    ] INFO 2024/01/06 17:41:05 using sqlite database: /root/.evcc/evcc.db

[main  ] ERROR 2024/01/06 17:41:06 creating vehicle ev4 failed: cannot create vehicle type 'template': cannot create vehicle type 'vw': missing state

[lp-1  ] DEBUG 2024/01/06 17:41:07 charge total import: 166.800kWh

[site  ] WARN 2024/01/06 17:41:07 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower

[site  ] INFO 2024/01/06 17:41:07 site config:

[site  ] INFO 2024/01/06 17:41:07   meters:      grid ✓ pv ✓ battery ✓

[site  ] INFO 2024/01/06 17:41:07     grid:      power ✓ energy ✗ currents ✗

[site  ] INFO 2024/01/06 17:41:07     pv 1:      power ✓ energy ✗ currents ✗

[site  ] INFO 2024/01/06 17:41:07     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗

[site  ] INFO 2024/01/06 17:41:07   vehicles:

[site  ] INFO 2024/01/06 17:41:07     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗

[lp-1  ] INFO 2024/01/06 17:41:07 loadpoint 1:

[lp-1  ] INFO 2024/01/06 17:41:07   mode:        pv

[lp-1  ] INFO 2024/01/06 17:41:07   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗

[lp-1  ] INFO 2024/01/06 17:41:07   meters:      charge ✓

[lp-1  ] INFO 2024/01/06 17:41:07     charge:    power ✓ energy ✓ currents ✓

[site  ] DEBUG 2024/01/06 17:41:07 set buffer soc: 5

[site  ] DEBUG 2024/01/06 17:41:07 set buffer start soc: 0

[site  ] DEBUG 2024/01/06 17:41:07 set priority soc: 0

[lp-1  ] DEBUG 2024/01/06 17:41:07 phase timer inactive

[lp-1  ] DEBUG 2024/01/06 17:41:07 pv timer inactive

[lp-1  ] DEBUG 2024/01/06 17:41:07 guard timer inactive

[site  ] WARN 2024/01/06 17:41:07 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

[site  ] DEBUG 2024/01/06 17:41:07 ----

[lp-1  ] DEBUG 2024/01/06 17:41:07 charge power: 0W

[site  ] DEBUG 2024/01/06 17:41:07 pv power: 0W

[site  ] DEBUG 2024/01/06 17:41:07 battery soc: 5%

[site  ] DEBUG 2024/01/06 17:41:07 battery power: 0W

[site  ] DEBUG 2024/01/06 17:41:07 grid power: 471W

[site  ] DEBUG 2024/01/06 17:41:07 site power: 471W

[lp-1  ] DEBUG 2024/01/06 17:41:07 charge voltages: [229 227 229]V

[lp-1  ] DEBUG 2024/01/06 17:41:07 detected connected phases: 3p

[lp-1  ] DEBUG 2024/01/06 17:41:07 charge currents: [0 0 0]A

[lp-1  ] DEBUG 2024/01/06 17:41:07 charge total import: 166.800kWh

[lp-1  ] DEBUG 2024/01/06 17:41:07 charger status: B

[lp-1  ] INFO 2024/01/06 17:41:07 car connected

[lp-1  ] DEBUG 2024/01/06 17:41:07 vehicle api refresh

[lp-1  ] DEBUG 2024/01/06 17:41:07 pv timer elapse

[lp-1  ] DEBUG 2024/01/06 17:41:07 pv timer inactive

[lp-1  ] DEBUG 2024/01/06 17:41:07 charger: guard elapse

[lp-1  ] DEBUG 2024/01/06 17:41:07 guard timer inactive

[lp-1  ] DEBUG 2024/01/06 17:41:07 charge total import: 166.800kWh

[lp-1  ] DEBUG 2024/01/06 17:41:07 pv charge current: 0A = 0A + -0.683A (471W @ 3p)

[lp-1  ] DEBUG 2024/01/06 17:41:07 pv enable timer reset

[lp-1  ] DEBUG 2024/01/06 17:41:07 pv timer inactive

[site  ] DEBUG 2024/01/06 17:41:17 ----

[lp-1  ] DEBUG 2024/01/06 17:41:17 charge power: 0W

[site  ] DEBUG 2024/01/06 17:41:17 pv power: 0W

[site  ] DEBUG 2024/01/06 17:41:17 battery soc: 5%

[site  ] DEBUG 2024/01/06 17:41:17 battery power: 0W

[site  ] DEBUG 2024/01/06 17:41:17 grid power: 458W

[site  ] DEBUG 2024/01/06 17:41:17 site power: 458W

[lp-1  ] DEBUG 2024/01/06 17:41:17 charge voltages: [229 227 229]V

[lp-1  ] DEBUG 2024/01/06 17:41:17 detected connected phases: 3p

[lp-1  ] DEBUG 2024/01/06 17:41:17 charge currents: [0 0 0]A

[lp-1  ] DEBUG 2024/01/06 17:41:17 charge total import: 166.800kWh

[lp-1  ] DEBUG 2024/01/06 17:41:17 charger status: B

[lp-1  ] DEBUG 2024/01/06 17:41:17 pv charge current: 0A = 0A + -0.664A (458W @ 3p)
```

StefanSchoof, 2024-01-06:
Try to log on the VW Webpage (not the App) and see if there is an new agb or other things to approve.

Also ensure your car is not in the offline mode.

fizzgig242, 2024-01-06:
Working again. 

Perhaps a temporary problem at VWs side.

closed 2024-01-06 by fizzgig242

f-lehmann-gmx-de, 2024-01-07:
@fizzgig242 Did you any change on your config? For me the issue is still persistent. Did you restart the docker or did it work somewhen automatically?

fizzgig242, 2024-01-07:
I logged into myVolkswagen and found a message regarding ads to approve to. I approved it and restarted the container.

After that it worked again and is still working.

Because of the warning messages in my log:

```
[site  ] WARN 2024/01/06 17:41:07 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower

[site  ] WARN 2024/01/06 17:41:07 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
```

I changed those two settings too.