evcc-io/evcc #11531: evcc beroichtet von einem Eintrag "resetondisconnect", den ich jedoch gelöscht habe.

christian-moeller, 2024-01-06:
### Describe the bug

failed configuring loadpoints
failed configuring loadpoint
1 error(s) decoding: * '' has invalid keys: resetondisconnect
Bitte überprüfe deine Konfigurationsdatei. Sollte dir die Fehlermeldung nicht weiterhelfen, suche in unseren [GitHub Diskussionen](https://github.com/evcc-io/evcc/discussions) nach einer Lösung.

Hinweis: Ein weiterer Grund könnte ein fehlerhaftes Gerät (Wechselrichter, Zähler, …) sein. Überprüfe deine Netzwerkverbindungen.

Konfiguration
Verwendete Konfigurationsdatei: /etc/evcc.yaml
Behebe das Problem und starte den Server neu.

# open evcc at http://evcc.local:7070  
network:  
  schema: http  
  host: evcc.local # .local suffix announces the hostname on MDNS  
  port: 7070  
  
log: info  
levels:  
  cache: error  
  
interval: 10s # control cycle interval  
  
sponsortoken: *****  
  
tariffs:  
  currency: EUR  
  grid:  
    type: fixed  
    price: ##### 
  
  feedin:  
    type: fixed  
    price: #####  
  
  co2:  
    type: grünstromindex  
    zip: #####  
  
meters:  
- type: template  
  template: volkszaehler-importexport   
  usage: grid    
  host: 192.168.1.71    
  port: 80    
  importuuid: fd90fa00-####
  exportuuid: 2bfec8c0-####
  name: grid1  
  
- name: Kostal_Piko  
  type: template  
  template: kostal-piko-legacy  
  usage: pv  
  host: 192.168.####
  user: *****  
  password: *****  
  
chargers:  
- type: template  
  template: go-e-v3   
  host: 192.168.####
  name: go-e-wallbox  
  
loadpoints:  
- title: Carport  
  charger: go-e-wallbox  
  mode: pv  
  phases: 0  
  mincurrent: 6  
  maxcurrent: 16  
  resetOnDisconnect: true        <<<=== Hier steht die ominöse Zeile!
  vehicle: mokkae  
  soc:  
    poll:  
      mode: charging  
      interval: 60m  
    estimate: true  
  
vehicles:  
- name: mokkae  
  type: template  
  template: opel  
  title: MokkaE  
  user: *****  
  password: *****  
  vin: *****  
  capacity: 50  
#  mode: pv  
#  minSoC: 30  
#  targetSoC: 80  
  
site:  
  title: Möller-Loccum  
  meters:  
    grid: grid1  
    pv: Kostal_Piko

Wenn ich nun "cat /etc/evcc.yaml" eingebe, erhalte ich folgenden Output:

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 10s # control cycle interval

sponsortoken: #####
tariffs:
  currency: EUR
  grid:
    type: fixed
    price: #####

  feedin:
    type: fixed
    price: #####

  co2:
    type: grünstromindex
    zip: #####

meters:
- type: template
  template: volkszaehler-importexport
  usage: grid
  host: 192.168.#####
  port: 80
  importuuid: fd90fa00-#####
  exportuuid: 2bfec8c0-#####
  name: grid1

- name: Kostal_Piko
  type: template
  template: kostal-piko-legacy
  usage: pv
  host: 192.168.#####
  user: pvserver
  password: pvwr

chargers:
- type: template
  template: go-e-v3
  host: 192.168.#####
  name: go-e-wallbox

loadpoints:
- title: Carport
  charger: go-e-wallbox
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 16
                                                                 <<<=== Hier steht NICHTS von dem vorher Ausgegebenen!?!?!?!?
  vehicle: mokkae
  soc:
    poll:
      mode: charging
      interval: 60m
    estimate: true

vehicles:
- name: mokkae
  type: template
  template: opel
  title: MokkaE
  user: #####
  password: #####
  vin: #####
  capacity: 50
#  mode: pv
#  minSoC: 30
#  targetSoC: 80

site:
  title: Möller-Loccum
  meters:
    grid: grid1
    pv: Kostal_Piko

Wo also nimmt evcc die .yaml-Datei tatsächlich her, damit ich dem Service dieses Gehabe abgewöhnen kann?

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
s.o.
```

### Log details

```text
[main  ] INFO 2024/01/06 20:29:33 evcc 0.123.5
[main  ] INFO 2024/01/06 20:29:33 using config file: /home/pi/evcc.yaml
[main  ] INFO 2024/01/06 20:29:33 starting ui and api at :7070
[db    ] INFO 2024/01/06 20:29:33 using sqlite database: /home/pi/.evcc/evcc.db
[lp-1  ] DEBUG 2024/01/06 20:29:36 charge total import: 4598.761kWh
[site  ] INFO 2024/01/06 20:29:36 site config:
[site  ] INFO 2024/01/06 20:29:36   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2024/01/06 20:29:36     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/01/06 20:29:36     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/01/06 20:29:36   vehicles:
[site  ] INFO 2024/01/06 20:29:36     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
[lp-1  ] INFO 2024/01/06 20:29:36 loadpoint 1:
[lp-1  ] INFO 2024/01/06 20:29:36   mode:        pv
[lp-1  ] INFO 2024/01/06 20:29:36   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2024/01/06 20:29:36   meters:      charge ✓
[lp-1  ] INFO 2024/01/06 20:29:36     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2024/01/06 20:29:36 phase timer inactive
[lp-1  ] DEBUG 2024/01/06 20:29:36 pv timer inactive
[lp-1  ] DEBUG 2024/01/06 20:29:36 guard timer inactive
[lp-1  ] INFO 2024/01/06 20:29:36 vehicle updated: unknown -> MokkaE
[site  ] WARN 2024/01/06 20:29:36 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/01/06 20:29:36 ----
[lp-1  ] DEBUG 2024/01/06 20:29:36 charge power: 0W
[site  ] DEBUG 2024/01/06 20:29:36 pv power: 0W
[site  ] DEBUG 2024/01/06 20:29:36 grid power: 5887W
[site  ] DEBUG 2024/01/06 20:29:36 site power: 5887W
[lp-1  ] DEBUG 2024/01/06 20:29:36 vehicle odometer: 22627km
[lp-1  ] DEBUG 2024/01/06 20:29:36 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/01/06 20:29:36 charge total import: 4598.761kWh
[lp-1  ] DEBUG 2024/01/06 20:29:36 charger status: B
[lp-1  ] INFO 2024/01/06 20:29:36 car connected
[lp-1  ] DEBUG 2024/01/06 20:29:36 pv timer elapse
[lp-1  ] DEBUG 2024/01/06 20:29:36 pv timer inactive
[lp-1  ] DEBUG 2024/01/06 20:29:36 charger: guard elapse
[lp-1  ] DEBUG 2024/01/06 20:29:36 guard timer inactive
[lp-1  ] DEBUG 2024/01/06 20:29:36 charge total import: 4598.761kWh
[lp-1  ] DEBUG 2024/01/06 20:29:36 vehicle status: B
[lp-1  ] DEBUG 2024/01/06 20:29:36 vehicle soc: 45%
[lp-1  ] DEBUG 2024/01/06 20:29:36 vehicle range: 86km
[lp-1  ] DEBUG 2024/01/06 20:29:36 set mokkae plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
[lp-1  ] DEBUG 2024/01/06 20:29:36 available power -5887W < 4140W min 3p threshold
[lp-1  ] DEBUG 2024/01/06 20:29:36 phase scale1p in 0s
[lp-1  ] DEBUG 2024/01/06 20:29:36 switched phases: 1p
[lp-1  ] DEBUG 2024/01/06 20:29:36 phase timer inactive
[lp-1  ] DEBUG 2024/01/06 20:29:36 pv charge current: 0A = 0A + -25.6A (5887W @ 1p)
[lp-1  ] DEBUG 2024/01/06 20:29:36 pv enable timer reset
[lp-1  ] DEBUG 2024/01/06 20:29:36 pv timer inactive
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.123.5

mdkeil, 2024-01-06:
> Wenn ich nun "cat /etc/evcc.yaml" eingebe, erhalte ich folgenden Output:

```
[main  ] INFO 2024/01/06 20:29:33 evcc 0.123.5
[main  ] INFO 2024/01/06 20:29:33 using config file: /home/pi/evcc.yaml
```
außerdem müssen `minSOC` und `targetSoc` ebenfalls raus.

0x3d13f, 2024-01-06:
In Log steht, dass er nicht die evcc.yaml in /etc, sondern in /home/pi benutzt. Dann musst Du diese auch ändern.

closed 2024-01-06 by andig