evcc-io/evcc #11534: PSA API fails with: oauth2: "invalid_grant" "grant is invalid"

stan23, 2024-01-07:
### Describe the bug

After some days or weeks the connection to PSA API breaks.
As a result, the SoC is unknown and planned charging does not work anymore.
Restart of cvcc solves the problem.

Is there a way to catch this and force a kind of re-authentication like a cvcc restart does?

Can I add even more details to the log beyond "Peugeot: trace"?

### Steps to reproduce

No special steps, just wait and after few days it happens.

1. Car was connected with 21% SoC and initial query of the API seems to be fine.
2. 5 minutes later the PSA API seems to report a timeout.
3. Same logs continue for about 3 hours *without* actually changing to the configured minimum SoC of 50%.
4. 7.5 hours later there's a disconnect event without touching the car, but API error remains the same:

All logs are in attached file.
[logs.txt](https://github.com/evcc-io/evcc/files/13852589/logs.txt)

### Configuration details

```yaml
<details><summary>Konfiguration (/etc/evcc.yaml)</summary>

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error
  peugeot: trace

interval: 10s # control cycle interval

sponsortoken: *****

meters:
- name: myPV
  type: template
  template: fronius-solarapi-v1 
  host: fronius.f17a  
  usage: pv
- name: myGrid
  type: custom
  power:
    source: mqtt
    topic: meter/energy/gridPowerCombined
- name: myHK
  type: homematic
  uri: homematic-raspi.f17a:2001
  device: NEQ0115240
  meterchannel: 2
  cache: 30s

chargers:
- name: myWallbox
  type: template
  template: go-e-v3
  host: go-eCharger_112810.f17a  

vehicles:
- name: my208e
  type: template
  template: peugeot 
[... 125 lines cut ...]
[lp-1  ] DEBUG 2024/01/07 08:03:44 pv timer inactive
[lp-1  ] DEBUG 2024/01/07 08:03:44 charger: guard elapse
[lp-1  ] DEBUG 2024/01/07 08:03:44 guard timer inactive
[lp-1  ] DEBUG 2024/01/07 08:03:44 charge total import: 3772.565kWh
[lp-1  ] DEBUG 2024/01/07 08:03:44 vehicle status: B
[lp-1  ] DEBUG 2024/01/07 08:03:44 vehicle soc: 21%
[lp-1  ] DEBUG 2024/01/07 08:03:44 vehicle range: 46km
[lp-1  ] DEBUG 2024/01/07 08:03:44 set my208e plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
[lp-1  ] DEBUG 2024/01/07 08:03:44 available power -529W < 4140W min 3p threshold
[lp-1  ] DEBUG 2024/01/07 08:03:44 phase scale1p in 0s
[pushover] DEBUG 2024/01/07 08:03:44 sending to ***
[lp-1  ] DEBUG 2024/01/07 08:03:44 switched phases: 1p
[lp-1  ] DEBUG 2024/01/07 08:03:44 phase timer inactive
[lp-1  ] DEBUG 2024/01/07 08:03:44 pv charge current: 0A = 0A + -2.3A (529W @ 1p)
[lp-1  ] DEBUG 2024/01/07 08:03:44 pv enable timer reset
[lp-1  ] DEBUG 2024/01/07 08:03:44 pv timer inactive
```

### What type of operating system are you running?

Linux

### Version

0.123.5

andig, 2024-01-07:
Bitte trace log.

andig, 2024-01-07:
I don't think there's anything to do. We have a valid refresh token. Refresh fails with unspecified server timeout. Afterwards refresh totally fails with previously valid refresh token. My guess: PSA had a infrastructure hick-up and invalidated all tokens. Since we're not seeing this regularly wontfix.

closed 2024-01-07 by andig

stan23, 2024-01-07:
Okay, ich schalte alles auf `trace` und warte ab bis es wieder auftritt.

andig, 2024-01-07:
Could you try finding a log that does not have a timeout error before? Or does it always happen together with a timeout error?

stan23, 2024-01-07:
I have only the last 3 occurrences due to log rotate, and it is always after that same timeout error.

But I'll keep an eye on it in the future once it happens again.

andig, 2024-01-07:
Does it ever refresh successfully? Think it did in the original log. So not sure what is going on at PSA...

stan23, 2024-01-08:
It did refresh a number of times until it broke after the timeout at 18:48.

```
164:Dec 31 16:09:25 evccCT evcc[581451]: [peugeot] TRACE 2023/12/31 16:09:25 grant_type=refresh_token&refresh_token=e33aab16-9fa8-4ce3-a541-66d2347c94f2
513:Jan  2 08:03:24 evccCT evcc[581451]: [peugeot] TRACE 2024/01/02 08:03:24 grant_type=refresh_token&refresh_token=75b2ed6d-c7bb-4cf2-abaa-e672408ad944
569:Jan  2 12:25:34 evccCT evcc[581451]: [peugeot] TRACE 2024/01/02 12:25:34 grant_type=refresh_token&refresh_token=938db208-5f60-4e5e-9a0b-d759c92ec5d5
589:Jan  2 13:26:14 evccCT evcc[581451]: [peugeot] TRACE 2024/01/02 13:26:14 grant_type=refresh_token&refresh_token=a768f870-794d-48b9-b72e-28b222e97c7a
611:Jan  2 14:26:54 evccCT evcc[581451]: [peugeot] TRACE 2024/01/02 14:26:54 grant_type=refresh_token&refresh_token=f8f4509d-c08a-44e8-bad1-f7ab3cd686fd
630:Jan  2 15:27:34 evccCT evcc[581451]: [peugeot] TRACE 2024/01/02 15:27:34 grant_type=refresh_token&refresh_token=ede69ae1-d561-4bc2-b812-196458426492
707:Jan  2 21:08:54 evccCT evcc[581451]: [peugeot] TRACE 2024/01/02 21:08:54 grant_type=refresh_token&refresh_token=114ddf61-dfa2-48fb-9388-44cdcbc22ae1
823:Jan  3 10:08:25 evccCT evcc[581451]: [peugeot] TRACE 2024/01/03 10:08:25 grant_type=refresh_token&refresh_token=614db5e4-f136-4bd3-8dba-89cb69c4e4a7
1171:Jan  4 07:36:26 evccCT evcc[593399]: [peugeot] TRACE 2024/01/04 07:36:26 grant_type=refresh_token&refresh_token=f913cd59-3f27-4b6e-b8b2-7a29cf6a57d9
1190:Jan  4 08:44:56 evccCT evcc[593399]: [peugeot] TRACE 2024/01/04 08:44:56 grant_type=refresh_token&refresh_token=f52be6f8-df9f-4b79-ac63-8704e90f76fe
1214:Jan  4 09:45:26 evccCT evcc[593399]: [peugeot] TRACE 2024/01/04 09:45:26 grant_type=refresh_token&refresh_token=fc524313-c258-40fc-a2e5-717145083d6f
1294:Jan  4 17:45:36 evccCT evcc[593399]: [peugeot] TRACE 2024/01/04 17:45:36 grant_type=refresh_token&refresh_token=5f825017-74d1-4fce-b414-6b8fccb15a0f
1318:Jan  4 18:48:25 evccCT evcc[593399]: [peugeot] TRACE 2024/01/04 18:48:25 grant_type=refresh_token&refresh_token=f1a7f686-f46a-4e4a-832c-a18d805b9f92
1324:Jan  4 18:48:35 evccCT evcc[593399]: [peugeot] TRACE 2024/01/04 18:48:35 grant_type=refresh_token&refresh_token=f1a7f686-f46a-4e4a-832c-a18d805b9f92
1332:Jan  4 18:48:56 evccCT evcc[593399]: [peugeot] TRACE 2024/01/04 18:48:56 grant_type=refresh_token&refresh_token=f1a7f686-f46a-4e4a-832c-a18d805b9f92
1342:Jan  4 18:49:25 evccCT evcc[593399]: [peugeot] TRACE 2024/01/04 18:49:25 grant_type=refresh_token&refresh_token=f1a7f686-f46a-4e4a-832c-a18d805b9f92```

andig, 2024-01-08:
This seems to indicate that refresh is working until PSA has a hickup. If this is a new problem let's monitor it for some time. If you're running on docker you could for example restart the container each night to trigger new login. If it persist we'll need to think about logging in again when refresh fails.

stan23, 2024-01-08:
Good idea but this is a Proxmox LXC.
I will wait until it happens again and then look at the evcc API to see if anything indicates the problem. If so, I'll send a push message to myself to restart evcc.

If I figure out anything I'll update this ticket.
Thanks for your time looking into it 👍

stan23, 2026-09-04:
The PSA integration in HomeAssistant breaks as well, and so does the native app. But they all break on different days, not related to each other.

For all successful refreshes, the response comes within 3 ms according to the syslog timestamp. But the error is shown after 2 ms. Are the syslog timestamps not reliable due to buffering?

I wonder whether the response comes later than the timeout, or does never reach evcc at all.

```
2026-09-04T16:03:38.349509+02:00 evccCT evcc[83272]: [peugeot] TRACE 2026/09/04 16:03:38 POST https://idpcvs.peugeot.com/am/oauth2/access_token

2026-09-04T16:03:38.365609+02:00 evccCT evcc[83272]: [peugeot] TRACE 2026/09/04 16:03:38 grant_type=refresh_token&refresh_token=c05e2271-8250-4e17-a5ed-1e80aa6fc8c5

2026-09-04T16:03:38.367512+02:00 evccCT evcc[83272]: [lp-1  ] ERROR 2026/09/04 16:03:38 charger soc: Get "https://api.groupe-psa.com/connectedcar/v4/user/vehicles/17f6bb30<cut>ba1b6783/status?client_id=11111111-2222-3333-4444-555555555555": Post "https://idpcvs.peugeot.com/am/oauth2/access_token": net/http: request canceled (Client.Timeout exceeded while awaiting headers) (Client.Timeout exceeded while awaiting headers)

2026-09-04T16:03:38.367833+02:00 evccCT evcc[83272]: [lp-1  ] ERROR 2026/09/04 16:03:38 vehicle range: Get "https://api.groupe-psa.com/connectedcar/v4/user/vehicles/17f6bb30<cut>ba1b6783/status?client_id=11111111-2222-3333-4444-555555555555": Post "https://idpcvs.peugeot.com/am/oauth2/access_token": net/http: request canceled (Client.Timeout exceeded while awaiting headers) (Client.Timeout exceeded while awaiting headers)

2026-09-04T16:03:58.579495+02:00 evccCT evcc[83272]: [peugeot] TRACE 2026/09/04 16:03:58 POST https://idpcvs.peugeot.com/am/oauth2/access_token

2026-09-04T16:03:58.579699+02:00 evccCT evcc[83272]: [peugeot] TRACE 2026/09/04 16:03:58 grant_type=refresh_token&refresh_token=c05e2271-8250-4e17-a5ed-1e80aa6fc8c5

2026-09-04T16:03:58.579767+02:00 evccCT evcc[83272]: --

2026-09-04T16:03:58.579828+02:00 evccCT evcc[83272]: {"error_description":"grant is invalid","error":"invalid_grant"}

2026-09-04T16:03:58.617211+02:00 evccCT evcc[83272]: [lp-1  ] ERROR 2026/09/04 16:03:58 charger soc: Get "https://api.groupe-psa.com/connectedcar/v4/user/vehicles/17f6bb30<cut>ba1b6783/status?client_id=11111111-2222-3333-4444-555555555555": oauth2: "invalid_grant" "grant is invalid"

2026-09-04T16:03:58.617373+02:00 evccCT evcc[83272]: [lp-1  ] ERROR 2026/09/04 16:03:58 vehicle range: Get "https://api.groupe-psa.com/connectedcar/v4/user/vehicles/17f6bb30<cut>ba1b6783/status?client_id=11111111-2222-3333-4444-555555555555": oauth2: "invalid_grant" "grant is invalid"
```