evcc-io/evcc #11535: Energieflussdiagramm: Farbliche Unterscheidung Solar- und Batterieleistung

naltatis, 2024-01-07:
Ich bin dem nicht ganz abgeneigt. Wir haben uns damals bewusst dafür entschieden, grün als Farbe für "Selbstproduzierte Energie" (Sonne & Batterie) zu verwenden, um das Diagramm simpel zu halten.

Die Annahme, dass die Batterie immer nur eigenen PV Strom enthält, wird zunehmend nicht mehr korrekt, wenn man den Hausspeicher auch als Netzpuffer (günstig aus Netz Aufladen / Tibber) verwendet. Das wäre für mich auch ein gutes Argument, die Batterieleistung farblich anders darzustellen.

Ich mach mir da mal Gedanke zu. Allerdings würde ich das Thema auf unserer Prioliste eher etwas weiter unten sehen.

_Originally posted by @naltatis in https://github.com/evcc-io/evcc/discussions/11506#discussioncomment-8035514_

SolarPowerEV, 2024-01-08:
So, da ich heute dank Sonnenschein nicht nur zum erstem Mal erfolgreich dank evcc PV-Überschuss in mein E-Auto geladen habe (und zu meiner Freude und Überraschung festgestellt habe, dass evcc meine Alfen Wallbox sogar in 100 W-Schritten hoch- und runterreguliert hat - ich hatte naiverweise angenommen, dass nur ganze Ampere-Werte regulierbar wären und (somit einphasig immer mindestens 230 W und dreiphasig sogar immer 690 W große Schritte geschehen würde) - hatte ich auch noch etwas Strom im Hausspeicher. Dadurch konnte ich auch endlich mal einen Screenshot von der Situation machen, die mich auf den ersten Blick immer hatte glauben lassen, die Anlage würde immer noch Sonnenstrom produzieren, wenn ich evcc auf meinem Handy aufgerufen habe:

[image]

Man sieht halt erstmal nur den grünen Balken und muss - gerade auf dem Handy - dann halt schon genau hinschauen, um zu sehen, dass da ein Batteriesymbol und kein Sonnensymbol über dem Balken ist. Klar, wenn man es weiß und drauf achtet, ist es eindeutig. Keine Frage. Aber selbst zwei Grüntöne würden halt wie gesagt helfen, auf den ersten Blick zu sehen, ob man gerade noch PV-Strom hat oder doch schon seinen Speicher entlädt.

Also nochmal danke fürs Aufnehmen meiner Anregung. Aber den Screenshot, der Situation die mich damals zu dem Vorschlag bewogen hat, wollte ich noch für alle Fälle nachliefern.

andig, 2024-01-19:
Brauchen wir das denn? Die Gliederung ist über die Symbole vorgegeben. Spätestens wenn Netzladen umgesetzt würde wäre der Batteriestrom auch nicht mehr zwingend grün sondern auch tw. grau.

naltatis, 2024-01-19:
> Spätestens wenn Netzladen umgesetzt würde wäre der Batteriestrom auch nicht mehr zwingend grün sondern auch tw. grau.

Genau das ist für mich das Argument, warum ich sehe, dass wir die Batterie-Energie abheben sollten. Momentan haben wir (nicht nur in der Visualisierung, auch in den Preisberechnungen) die Annahme, dass die Batterie nur eigenen Sonnenstrom enthält. Diese Annahme ist nicht mehr immer richtig.

VolkerK62, 2024-01-19:
>die Annahme, dass die Batterie nur eigenen Sonnenstrom enthält. Diese Annahme ist nicht mehr immer richtig.

Willst du das in Zukunft anders berücksichtigen? Also der gespeicherten Energie auf Basis der "Herkunft" einen Preis vepassen?

umrath, 2024-03-06:
Grundsätzlich ist die Idee Klasse.
Es gibt noch einen anderen Aspekt, den @SolarPowerEV schon indirekt angeschnitten hat: Kosten

Batteriestrom ist immer teurer als PV-Strom.
Ggfs. lässt sich also auch darüber eine Lösung formulieren (oder in Zukunft möglicherweise sogar die gesamte Steuerung darüber abbilden).

naltatis, 2024-03-08:
> Batteriestrom ist immer teurer als PV-Strom.

Ja, haben wir bislang bewusst ausgeklammert, weil das hochgradig individuell ist und von den verwendeten Systemen, deren Wirkungsgraden und(!) dem Nutzungsverhalten (kurze hohe Last, lange niedrige Last ...) abhängt.

PR #11733 (2024-03-08): Add battery color to visualization
Introduce dedicated battery color for self consumption.

fixes #11535

[image]

**Todos:**

- [x] Decide on color (cc @markusrem)
- [x] Test edge cases
- [x] Adjust legend (same label + split color badge vs. third center label)
closed 2024-03-08 by andig