evcc-io/evcc #11536: Einstellungsmenü nicht durchgängig identisch

VolkerK62, 2024-01-07:
Die neuen Einstellungen für "Batterie" und "smartes Netzladen" sind nicht durchgängig verfügbar.
Sie fehlen bei den Ansichten "Ladevorgänge" und "device settings".

[image]

[image]

naltatis, 2024-01-07:
Jo, auf diesen Seiten fehlt die Info ob es Batterie oder dyn. Netzinfos gibt. Ich schau da rein ...

closed 2024-01-08 by naltatis

StevieC121176, 2024-01-10:
Ich denke mal es hängt mit den Änderungen zu diesem Issue zusammen. Mit Version `0.123.7 (68da7ac9)` kann ich den Menüpunkt `Ladevorgänge` und `Device Configuration` nicht mehr aufrufen.

PR #11542 (2024-01-08): Fix: battery/grid settings links on sessons and config page
fixes #11536